warpme/minimyth2 #2: no build

Shaggy013, 2021-05-13:
libaries missing in the setup 

libc-dev:i386

make garchive crashes on several packages like imagemagick ,nvidia , perl packages

and the build crashes on

libtool

ow and add patch for sd/emmc problem rk3399

https://github.com/armbian/build/blob/master/patch/u-boot/u-boot-rockchip64-mainline/general-prioritize-sd.patch

and a question how you make the checksum files ? to add patches ?

i now changed the rockpi to orangepi-4 to make a image for orangepi-4

warpme, 2021-05-13:
Shaggy013

- re libaries missing in the setup: libc-dev:i386

What distro are you using for build?

- re make garchive issue: 

yes. this is know issue and needs some cleanup as garchive needs to be arch dependent (as some packages are not aval. for some arch) and currently it is not. pls simply skip garchive and go directly to make build. I already updated build instructions to skip this stage. 
  
- re libtool crash: 

i tested full build of current master of mm2 on vm with vanilla: ubuntu21, fedora34 and archlinux current. no issues observed with full build from vanilla os. 
i suspect there is interference with sw. on your host machine. May you try with temporally uninstalled libtool on your host and see is there difference?
 
- re sd/mms on rk3399:

i'll test it on my rockpi4 and if all wilbe ok - i'll add this patch. thx for this!

- re make the checksum files:

go to package and run "make makesums"

- re to add patches:

-create "files" dir in package dir
-put patch in "files" dir
-add "PATCHFILES   = <patch filename>" in make file
-run  "make makesums" in package dir

Shaggy013, 2021-05-13:
Yeah you putting much work in it seeing at the commits .

i came across your project for the search hvec in 5.12  for rk3399  i just like try things.
and your way is diffrent as the more french sounding kodi buildroot 
and mythtv was one of the first vdrtools i used back in the days. so i also suprised that it is still active keep it up .
 
there is not much diference for orangepi-4 
the firmware blob for the wifi/bt is only wich you need for orangepi 4 to get wifi/bt

[5000-Boards-add-Orangepi4.txt](https://github.com/warpme/minimyth2/files/6473192/5000-Boards-add-Orangepi4.txt)
[add-board-orangepi-4-uboot.txt](https://github.com/warpme/minimyth2/files/6473270/add-board-orangepi-4-uboot.txt)
[general-prioritize-sd.txt](https://github.com/warpme/minimyth2/files/6473271/general-prioritize-sd.txt)
the rest is already there already for the 3328 and 3399 
and maybey sound but it works with the rockpi setting to test 
only thing that is missing is is the dts files for the device tree in uboot and kernel.

the mmc/sd patch is for boards with emmc and want to boot form sd . you could even adapt it to boot from usb to put the usb above them 

i try to compile it in a fresh ubunntu install with office and thunderbird removed and just apt updated.
did apt get for your needed packages and did install the needed packages for armbian.
i normaly run Gentoo.

but i gonna try your information to getting it build .

this are just the files i added and changed +build the boot scr files.

warpme, 2021-05-14:
I added OrangePI-4 patches to kernel and create new board for this device.

Here is SD card image for Orange-4 device.
https://github.com/warpme/minimyth2/releases/tag/v11.19.1-v32-Pre-2751

All code should be ok for this device except bootloader: currently it has included rksd_loader.img from rk3399 RockPI-4. 
Maybe this will be enough to start OrangePI-4 board to Linux kernel?
Pls try - best with attached serial to see how boot process goes?

If boot will fail - we my try with armbian bootloader DD to SD card?

In long-term I have plan to go with mainline u-boot on rk3399 devices - but atm thing is tricky in ATF: building ATF needs gcc toolchain for CortexM0 (dram init in ATF is for this target).   

let me know how it goes!

Shaggy013, 2021-05-14:
great thnx i go check it out

i still stuck at libtool compile it crashes at config 

atf or the other loader makes not much diffrence for the orangepi mayby ram speed i thought 

did you ever looked at https://github.com/150balbes
it is for biggest part just a dts file change to get a lot of boards working with 1 kernel .

but again thnx

warpme, 2021-05-14:
- re: "i still stuck at libtool compile it crashes at config"

just to be sure: libtool in not needed by minimyth2 except functionality to build minimal static gcc compiler.

- re: atf or the other loader makes not much diffrence for the orangepi mayby ram speed i thought

may you pls try image i build on your hw?
I don't have orangepi-4 hw so checking how far it boots is valuable for me.

- re: did you ever looked at https://github.com/150balbes

yes. many times i exchanged with Oleg (150balbes) about various things :-)

btw: i already got mainline uboot for rockpi4 compiled and now testing image for how it goes on hw i have.
if all will go ok - we are in home with orangepi-4...

Shaggy013, 2021-05-14:
well you missed the mmc patch in uboot  so booting with a sd with mmc onboard is difficult

U-Boot 2017.09-armbian (Feb 17 2020 - 10:01:10 +0100)

Model: RockPi-4B
DRAM:  3.9 GiB
DCDC_REG1@vdd_center: ; enabling
DCDC_REG2@vdd_cpu_l: ; enabling
DCDC_REG3@vcc_ddr: ; enabling (ret: -38)
DCDC_REG4@vcc_1v8: set 1800000 uV; enabling
LDO_REG1@vcc1v8_dvp: set 1800000 uV; enabling
LDO_REG2@vcc3v0_touch: set 3000000 uV; enabling
LDO_REG3@vcc1v8_pmu: set 1800000 uV; enabling
LDO_REG4@vcc_sd: set 3300000 uV; enabling
LDO_REG5@vcca3v0_codec: set 3000000 uV; enabling
LDO_REG6@vcc_1v5: set 1500000 uV; enabling
LDO_REG7@vcca1v8_codec: set 1800000 uV; enabling
LDO_REG8@vcc_3v0: set 3000000 uV; enabling
SWITCH_REG1@vcc3v3_s3: ; enabling (ret: -38)
SWITCH_REG2@vcc3v3_s0: ; enabling (ret: -38)
vcc1v8-s0@vcc1v8_s0: set 1800000 uV; enabling (ret: -38)
dc-12v@dc_12v: set 12000000 uV; enabling (ret: -38)
vcc-sys@vcc_sys: set 5000000 uV; enabling (ret: -38)
vcc3v3-sys@vcc3v3_sys: set 3300000 uV; enabling (ret: -38)
vcc-phy-regulator@vcc_phy: ; enabling (ret: -38)
vdd-log@vdd_log: ; enabling (ret: -38)
MMC:   sdhci@fe330000: 0, dwmmc@fe320000: 1
SF: unrecognized JEDEC id bytes: 49, 00, 00
*** Warning - spi_flash_probe_bus_cs() failed, using default environment

In:    serial@ff1a0000
Out:   serial@ff1a0000
Err:   serial@ff1a0000
Model: RockPi-4B
Net:   eth0: ethernet@fe300000
Hit any key to stop autoboot:  0
Card did not respond to voltage select!
mmc_init: -95, time 20
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
reading /boot.scr
1682 bytes read in 5 ms (328.1 KiB/s)
## Executing script at 00500000
reading /Image
30571008 bytes read in 3243 ms (9 MiB/s)
reading /rk3399_dtb
** Unable to read file /rk3399_dtb **
reading /dtbs/rockchip/rk3399-orangepi-4.dtb
62396 bytes read in 14 ms (4.2 MiB/s)
libfdt fdt_path_offset() returned FDT_ERR_NOTFOUND
reading /initramfs-linux.img
** Unable to read file /initramfs-linux.img **
## Flattened Device Tree blob at 01f00000
   Booting using the fdt blob at 0x1f00000
   reserving fdt memory region: addr=1f00000 size=10000
   Loading Device Tree to 00000000f5ef0000, end 00000000f5f02fff ... OK

Starting kernel ...

https://github.com/150balbes/build/blob/armbian-tv/patch/u-boot/u-boot-rockchip64-mainline/u-boot-rk-rk3399-usb-start.patch
this makes it for all rk3399 boards easy

warpme, 2021-05-14:
oh well - it is not me :-p
image i provide you is with rksd_loader.img blob from 1-year old Armbian.
but it looks like we are close: rockpi4 boots nicely with mainline uboot for me.
switching orangepi-4 to mainline uboot is now trivial :-)

Shaggy013, 2021-05-14:
well i disabled spi in favor of tft testing

but other then that sd i sure i dont have a rockpi uboot 

this is mine on mmc

# DDR Version 1.24 20191016
In
channel 0
CS = 0
MR0=0x18
MR4=0x1
MR5=0x6
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0xFF
CS = 1
MR0=0x18
MR4=0x1
MR5=0x6
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0xFF
channel 1
CS = 0
MR0=0x18
MR4=0x1
MR5=0x6
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0xFF
CS = 1
MR0=0x18
MR4=0x1
MR5=0x6
MR8=0x8
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0xFF
channel 0 training pass!
channel 1 training pass!
change freq to 416MHz 0,1
Channel 0: LPDDR4,416MHz
Bus Width=32 Col=10 Bank=8 Row=15/15 CS=2 Die Bus-Width=16 Size=2048MB
Channel 1: LPDDR4,416MHz
Bus Width=32 Col=10 Bank=8 Row=15/15 CS=2 Die Bus-Width=16 Size=2048MB
256B stride
channel 0
CS = 0
MR0=0x18
MR4=0x1
MR5=0x6
MR8=0x8
MR12=0x72
MR14=0x72
[... 389 lines cut ...]
mmc_init: -95, time 20
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
reading /boot.scr
1682 bytes read in 4 ms (410.2 KiB/s)
## Executing script at 00500000
reading /Image
30571008 bytes read in 3243 ms (9 MiB/s)
reading /rk3399_dtb
** Unable to read file /rk3399_dtb **
reading /dtbs/rockchip/rk3399-orangepi-4.dtb
62396 bytes read in 14 ms (4.2 MiB/s)
libfdt fdt_path_offset() returned FDT_ERR_NOTFOUND
reading /initramfs-linux.img
** Unable to read file /initramfs-linux.img **
## Flattened Device Tree blob at 01f00000
   Booting using the fdt blob at 0x1f00000
   reserving fdt memory region: addr=1f00000 size=10000
   Loading Device Tree to 00000000f5ef0000, end 00000000f5f02fff ... OK

will zero it tommorrow to see if it not a offset mixup

and i have a gentoo kernel with a intiram with your patches so also can boot the /

warpme, 2021-05-14:
i quickly build orangepi-4 with mainline uboot
https://github.com/warpme/minimyth2/releases/tag/v11.19.1-v32-Pre-2751
pls try
(it was quick  code changes so i may miss something but maybe...)

Shaggy013, 2021-05-14:
https://github.com/Depau/alarm-orangepi4

depau has both for example 

ok but then i get some sleep :)

Shaggy013, 2021-05-14:
U-Boot TPL 2021.04-rc4 (May 15 2021 - 00:25:09)
Channel 0: LPDDR4, 50MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
Channel 1: LPDDR4, 50MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
256B stride
lpddr4_set_rate: change freq to 400000000 mhz 0, 1
lpddr4_set_rate: change freq to 800000000 mhz 1, 0
Trying to boot from BOOTROM
Returning to boot ROM...

U-Boot SPL 2021.04-rc4 (May 15 2021 - 00:25:09 +0200)
Trying to boot from MMC1
NOTICE:  BL31: v2.4(release):v2.4-2-gb623c807e
NOTICE:  BL31: Built : 22:21:47, Nov 28 2020

U-Boot 2021.04-rc4 (May 15 2021 - 00:25:09 +0200)

Model: OrangePi 4 AI board
DRAM:  3.9 GiB
PMIC:  RK808
MMC:   mmc@fe320000: 1, sdhci@fe330000: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Model: OrangePi 4 AI board
Net:   eth0: ethernet@fe300000
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
1682 bytes read in 4 ms (410.2 KiB/s)
## Executing script at 00500000
30571008 bytes read in 1311 ms (22.2 MiB/s)
Failed to load '/rk3399_dtb'
62396 bytes read in 9 ms (6.6 MiB/s)
libfdt fdt_path_offset() returned FDT_ERR_NOTFOUND
Failed to load '/initramfs-linux.img'
Moving Image from 0x2080000 to 0x2200000, end=3fc0000
## Flattened Device Tree blob at 01f00000
   Booting using the fdt blob at 0x1f00000
   Loading Device Tree to 00000000f5f14000, end 00000000f5f26fff ... OK

Starting kernel ...

Shaggy013, 2021-05-15:
yeah no boot

Shaggy013, 2021-05-15:
the firmware is missing the dp firmware for rockchip
rockchip/dptx.bin

why all that mess of copy to a rootfs and no intramfs or a initrd ?
doesnt sbin/init look for a etc/.... file in / an till rootfs or rootfs-ro is not / how does sbin/init see the file ?

warpme, 2021-05-16:
Ok - i added missing firmware + update some wifi/bt firmware as well.
pls try.
for any issues - pls provide log/dmesg

Shaggy013, 2021-05-16:
U-Boot TPL 2021.04-rc4 (May 16 2021 - 14:49:37)
Channel 0: LPDDR4, 50MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
Channel 1: LPDDR4, 50MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
256B stride
lpddr4_set_rate: change freq to 400000000 mhz 0, 1
lpddr4_set_rate: change freq to 800000000 mhz 1, 0
Trying to boot from BOOTROM
Returning to boot ROM...

U-Boot SPL 2021.04-rc4 (May 16 2021 - 14:49:37 +0200)
Trying to boot from MMC1
NOTICE:  BL31: v2.4(release):v2.4-2-gb623c807e
NOTICE:  BL31: Built : 22:21:47, Nov 28 2020

U-Boot 2021.04-rc4 (May 16 2021 - 14:49:37 +0200)

Model: OrangePi 4 AI board
DRAM:  3.9 GiB
PMIC:  RK808
MMC:   mmc@fe320000: 1, sdhci@fe330000: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Model: OrangePi 4 AI board
Net:   eth0: ethernet@fe300000
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found U-Boot script /boot.scr
1682 bytes read in 4 ms (410.2 KiB/s)
## Executing script at 00500000
30571008 bytes read in 1311 ms (22.2 MiB/s)
Failed to load '/rk3399_dtb'
62396 bytes read in 9 ms (6.6 MiB/s)
libfdt fdt_path_offset() returned FDT_ERR_NOTFOUND
Failed to load '/initramfs-linux.img'
Moving Image from 0x2080000 to 0x2200000, end=3fc0000
## Flattened Device Tree blob at 01f00000
   Booting using the fdt blob at 0x1f00000
   Loading Device Tree to 00000000f5f14000, end 00000000f5f26fff ... OK

Starting kernel ...

warpme, 2021-05-16:
hmm - so it looks like something in kernel not going well.
sd card fat partition has boot-debug.scr script which has added debug in kernel cmd.line.
Pls try rename it to debug.scr. This should give kernel boot verbosity - maybe we will see where issue is?

btw: in my rockpi4 adding 'debug' to kernel cmd.line make boot hanging!
i don't know why but pls be aware this also may happen on orangepi4. 
I think this is kind of kernel bug?

Shaggy013, 2021-05-16:
i check it ;)
added a echo 
Boot partuuid is 346d1a6a-01
Root partuuid is 346d1a6a-02

well that goes well from uboot 

i adding debug now to kernel

Shaggy013, 2021-05-16:
well the kernel boots but no serial or screen

warpme, 2021-05-16:
grrr - so the same like on my rockpi4
so it looks i need first to fix kernel verbosity on rk3399....
as soon i'll fix this - is let you know!

Shaggy013, 2021-05-16:
i know whats wrong 

disable dp

Shaggy013, 2021-05-16:
you can do it in dts or just leave the module from kernel

Shaggy013, 2021-05-16:
or disable hdmi and videoconsole in uboot
same as on armbian

warpme, 2021-05-16:
"leave the module from kernel" - i understand we want kernel will NOT load dp module.
If yes - then we can remove this module from sd card, go back with boot.scr without 'debug' and this may work?
To remove module - pls mount 2nd sdcard partition and remove required module from /rootfs-ro/lib/modules/5.12.3/kernel/....

Shaggy013, 2021-05-16:
rk3399 debug serial is 1500000 on tty2
and 
# CONFIG_ROCKCHIP_CDN_DP is not set
or
&cdn_dp {
	status = "disabled";
in dts

warpme, 2021-05-16:
excellent :-) 
rebuilding 5.12.4 with "CONFIG_ROCKCHIP_CDN_DP is not set" + added wifi nvram for orangepi-4

Shaggy013, 2021-05-16:
well that is hard to do with a builtin 
and the dts does the trick also

Shaggy013, 2021-05-16:
[image]

warpme, 2021-05-16:
nice :-)
pls test also peripherals: i want to know what works and what not

Shaggy013, 2021-05-16:
yeh yeh 
is +10  years ago i used mythttv

well search crashedwith debug on 
 but i will check it out

warpme, 2021-05-17:
I updated & uploaded image with CONFIG_ROCKCHIP_CDN_DP is not set" + added wifi nvram for orangepi-4.
Interesting that i still have kernel boot hang with "debug" on rockpi4.
DTS for  rockpi4 seems to not enabling cdn_dp - so i think playing  with CONFIG_ROCKCHIP_CDN_DP is not root cause solution and root issue is elsewhere.

Shaggy013, 2021-05-18:
indeed the cause is that uboot and kernel are not 100% in sync 

and the dp is a extra fb for a screen so that somehow get mixed up or not reset properly before kernel starts

you can disable hdmi and dp in uboot so kernel can fresh initialize them or dp in kernel dts with some warnings.

and thought i saw a try of patch last week on patchwork .

and if you look how a pinebook and a rockpro64 is brought up in uboot you see they fix some powerails due a extra reset. and playing with powerrails in not funny killed almost my board :)

for me removing the extcon in dts for dp  fixes it for me to boot uboot and kernel with hdmi support and dp module on.

maybye this fix it ?

https://patchwork.kernel.org/project/linux-rockchip/patch/20210516074833.451643-1-javierm@redhat.com/

warpme, 2021-05-19:
thx.
i tried above patch on 5.12 and it looks it depends on future simple-framebuffer infra.
looking on https://lore.kernel.org/virtualization/20210318102921.21536-1-tzimmermann@suse.de/ i see it is interesting.
Unfortunately attempt to add it to 5.12 gives me cascade of missing dependencies like https://patchwork.kernel.org/project/dri-devel/cover/20210430105840.30515-1-tzimmermann@suse.de/ then https://github.com/gregkh/linux/commit/6dd7b6ce43acd70f75ae285fee8eeb2dd4933dce and much more...

Shaggy013, 2021-05-19:
yeah then it becomes messy indeed

media kernel it patches against

this looks also a fix for it 

https://patchwork.kernel.org/project/linux-rockchip/patch/20210519134928.2696617-1-yangyingliang@huawei.com/

warpme, 2021-05-19:
ok good :)

I built 11.19.2 with this patch applied + enabled back cdn_dp.

If you have running mm2 on Your OrangePI with Internet access You may try on-line updates.
Easiest will be via GUI
Also You can to from telenet session to board.

In telnet session you can type :

`mm_do_online_update`

this will show what will be updated

To update:

`mm_do_online_update doupdate`

`sync`

`reboot`

Updates will be active after board reboot....

Shaggy013, 2021-05-24:
thc cdn patch works 

only your buildroot stops now at uboot-tools

make[12]: sdl2-config: No such file or directory
cc1: error: /opt/powerpc/include: Permission denied
make[13]: *** [scripts/Makefile.host:114: tools/gdb/gdbsend.o] Error 1
make[12]: *** [scripts/Makefile.build:432: tools/gdb] Error 2
make[11]: *** [Makefile:1728: tools] Error 2

warpme, 2021-05-24:
hmm...
may you pls:
go to script/bootloaders/u-boot.tools
```
make clean 
make build DESTIMG="build"
```
and provide full log for above?

warpme, 2021-05-25:
interesting line is:
cc1: error: /opt/powerpc/include: Permission denied
isn't that user under which you are compiling don't have any access rights to /opt/powerpc/include dir?
I don't see such error under:
archlinux 2020.10
archlinux 2021.03
fedora 33
fedora 34
ubuntu 20.04
ubuntu 21.04
what distro are you using as host?
it will be good to enable verbose building of u-boot tools to see what is compiler invocation causing failure.
Also you make turn-off parallel build (look at end of minimyth.conf.mk) to better correlate invocation with error.

Shaggy013, 2021-05-25:
That I checked. It just a clean xubuntu focal install Sent from my Huawei phone-------- Original message --------From: Piotr Oniszczuk ***@***.***>Date: Tue, 25 May 2021, 21:14To: warpme/minimyth2 ***@***.***>Cc: Shaggy013 ***@***.***>, Author ***@***.***>Subject: Re: [warpme/minimyth2] no build (#2)
interesting line is:
cc1: error: /opt/powerpc/include: Permission denied
isn't that user under which you are compiling don't have any access rights to /opt/powerpc/include dir?
I don't see such error under:
archlinux 2020.10
archlinux 2021.03
fedora 33
fedora 34
ubuntu 20.04
ubuntu 21.04
what distro are you using as host?
it will be good to enable verbose building of u-boot tools to see what is compiler invocation causing failure.
Also you make turn-off parallel build (look at end of minimyth.conf.mk) to better correlate invocation with error.

warpme, 2021-05-27:
hmm,
i can't reproduce your issue.

what i done:

- download https://torrent.ubuntu.com/xubuntu/releases/focal/release/desktop/xubuntu-20.04.2.0-desktop-amd64.iso.torrent
- install xbuntu under vmware
- run full system update
- do all mm2 wiki commands to build mm2 under ubuntu
- after 17h i got built mm2 SD image
- insert SD image into tanixTX6
- TX6 fully works as expected

mm2 build was uninterrupted - with single expected break: QtWebKit build exits due out of available memory (my vm has 8 cpu and 8GB ram. simple resume allows to finish mm2 build without any other interruptions).

did you install any additional software on yours xbuntu?

anyway: as i can't reproduce your's issue - only extra build logs my help us....

Shaggy013, 2021-05-29:
it is already strange that opt may not have write options for a user with that gar 

i only have a build repo in it from armbian 

user rights disabled for opt

closed 2022-03-03 by warpme