warpme/minimyth2 #10: Tanix TX6s - using latest head

heitbaum, 2021-06-25:
```
U-Boot SPL 2021.04 (Jun 25 2021 - 13:42:25 +0000)
DRAM: 4096 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.4(debug):11.19.1-v32-Pre-2762-gc8c226c-70-g724b08c3
NOTICE:  BL31: Built : 13:33:18, Jun 25 2021
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a085aa8, model: Tanix TX6s
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
WARNING: PMIC: No regulators DT node, skipping setup
INFO:    Changed devicetree to reserve BL31 memory.
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9

U-Boot 2021.04 (Jun 25 2021 - 13:42:25 +0000) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: Tanix TX6s
I2C:   ready
DRAM:  3 GiB
MMC:   mmc@4020000: 0, mmc@4022000: 1
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1... In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   No ethernet found.
starting USB...
No working controllers found
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1558 bytes read in 1 ms (1.5 MiB/s)
## Executing script at 4fc00000
30988800 bytes read in 2561 ms (11.5 MiB/s)
Failed to load '/h616_dtb'
24422 bytes read in 5 ms (4.7 MiB/s)
Failed to load '/initramfs-linux.img'
Moving Image from 0x40080000 to 0x40200000, end=42020000
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
   Loading Device Tree to 0000000049ff7000, end 0000000049ffffff ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.13.0-rc7 (docker@dfcc6e7f24af) (aarch64-minimyth-linux-gnu-gcc (GCC) 10.3.0, GNU ld (GNU Binutils) 2.36) #1 SMP PREEMPT Fri Jun 25 13:07:18 UTC 2021
[    0.000000] Machine model: Tanix TX6s
[    0.000000] efi: UEFI not found.
[    0.000000] earlycon: uart0 at MMIO32 0x0000000005000000 (options '115200n8')
[    0.000000] printk: bootconsole [uart0] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000004007ffff]
[    0.000000]   node   0: [mem 0x0000000040080000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000] On node 0 totalpages: 786432
[    0.000000]   DMA zone: 12288 pages used for memmap
[... 380 lines cut ...]
[    5.864450] Bluetooth: HCI UART protocol H4 registered
[    5.875779] input: sunxi-ir as /devices/platform/soc/7040000.ir/rc/rc0/input3
[    5.877400] Bluetooth: HCI UART protocol BCSP registered
[    5.885261] sunxi-ir 7040000.ir: initialized sunXi IR driver
[    5.889306] Bluetooth: HCI UART protocol LL registered
[    5.900321] Bluetooth: HCI UART protocol ATH3K registered
[    5.906059] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    5.913050] Bluetooth: HCI UART protocol Intel registered
[    5.919081] Bluetooth: HCI UART protocol Broadcom registered
[    5.925023] Bluetooth: HCI UART protocol QCA registered
[    5.930491] Bluetooth: HCI UART protocol Marvell registered

[    9.262734] loop: module loaded
[    9.331509] urandom_read: 2 callbacks suppressed
[    9.331526] random: udevd: uninitialized urandom read (16 bytes read)
[    9.342771] random: udevd: uninitialized urandom read (16 bytes read)

[   11.462955] dwmac-sun8i 5030000.ethernet eth0: PHY [stmmac-0:01] driver [Generic PHY] (irq=POLL)
[   11.473208] dwmac-sun8i 5030000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   11.481330] dwmac-sun8i 5030000.ethernet eth0: No Safety Features support found
[   11.488685] dwmac-sun8i 5030000.ethernet eth0: No MAC Management Counters available
[   11.496348] dwmac-sun8i 5030000.ethernet eth0: PTP not supported by HW
[   11.503338] dwmac-sun8i 5030000.ethernet eth0: configuring for phy/rmii link mode
[   11.511866] dwmac-sun8i 5030000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
```

heitbaum, 2021-06-25:
[image]

heitbaum, 2021-06-26:
did get a panic overnight:

```
[31422.206073] ------------[ cut here ]------------
[31422.210714] kernel BUG at kernel/irq_work.c:170!
[31422.215327] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP
[31422.220805] Modules linked in: brcmsmac mac80211 libarc4 bcma cordic brcmfmac brcmutil cfg80211 ipv6 sun50i_cpufreq_nvmem nls_ascii nls_cp437 vfat fat uinput loop hci_uart btintel btbcm btqca ir_nec_decoder rc_beelink_gs1 crct10dif_ce sunxi_cir sunxi_cedrus(C) rc_core sun50i_di dw_hdmi_cec joydev bluetooth v4l2_mem2mem videobuf2_v4l2 ecdh_generic ecc videobuf2_dma_contig videobuf2_memops rfkill videobuf2_common dwmac_sun8i sun8i_thermal sunxi_wdt panfrost videodev gpu_sched mc overlay
[31422.263753] CPU: 3 PID: 0 Comm: swapper/3 Tainted: G         C        5.13.0-rc7 #1
[31422.271398] Hardware name: Tanix TX6s (DT)
[31422.275487] pstate: 00000005 (nzcv daif -PAN -UAO -TCO BTYPE=--)
[31422.281484] pc : irq_work_run_list+0x68/0x6c
[31422.285760] lr : irq_work_tick+0x44/0xb0
[31422.289678] sp : ffff800011e33d50
[31422.292985] x29: ffff800011e33d50 x28: ffff0000bf9ac238 x27: ffff0000bf9ac278
[31422.300114] x26: ffff0000bf9ac2b8 x25: ffff0000bf9ac18c x24: ffff0000bf9ac180
[31422.307242] x23: ffff800011e33e90 x22: ffff800010151c90 x21: ffff800011a5a5f8
[31422.314370] x20: ffff80001159e9f8 x19: ffff80001159e9f0 x18: 0000000000000000
[31422.321498] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[31422.328625] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000040
[31422.335753] x11: ffff00000354b478 x10: ffff00000354b47a x9 : ffff80001009d4f8
[31422.342881] x8 : ffff000002800700 x7 : 0000000000000000 x6 : 000000af9ed1d2fd
[31422.350008] x5 : 00000001008e7724 x4 : 0000000000029599 x3 : ffff80001158c000
[31422.357136] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff0000bf9b59f0
[31422.364264] Call trace:
[31422.366706]  irq_work_run_list+0x68/0x6c
[31422.370624]  update_process_times+0xdc/0x100
[31422.374891]  tick_sched_handle+0x38/0x74
[31422.378811]  tick_sched_timer+0x80/0x140
[31422.382728]  __run_hrtimer+0x84/0x2c0
[31422.386385]  __hrtimer_run_queues+0xa4/0x120
[31422.390649]  hrtimer_interrupt+0xfc/0x2d0
[31422.394652]  arch_timer_handler_phys+0x3c/0x50
[31422.399094]  handle_percpu_devid_irq+0xac/0x280
[31422.403620]  __handle_domain_irq+0xb8/0x130
[31422.407796]  gic_handle_irq+0x58/0xe0
[31422.411458]  el1_irq+0xbc/0x154
[31422.414595]  arch_cpu_idle+0x18/0x3c
[31422.418165]  default_idle_call+0x30/0x1b8
[31422.422169]  cpuidle_idle_call+0x174/0x1f0
[31422.426263]  do_idle+0xac/0x100
[31422.429400]  cpu_startup_entry+0x2c/0x6c
[31422.433317]  secondary_start_kernel+0x104/0x180
[31422.437848] Code: a8c27bfd d50323bf d65f03c0 d65f03c0 (d4210000)
[31422.443934] ---[ end trace 842cb1862742e73c ]---
[31422.448544] Kernel panic - not syncing: Oops - BUG: Fatal exception in interrupt
[31422.455930] SMP: stopping secondary CPUs
[31422.459850] Kernel Offset: disabled
[31422.463332] CPU features: 0x00000001,00000846
[31422.467682] Memory Limit: none
[31422.470736] ---[ end Kernel panic - not syncing: Oops - BUG: Fatal exception in interrupt ]---
```

heitbaum, 2021-06-26:
mmc in u-boot reports correctly:
```
=> mmc list
mmc@4020000: 0 (SD)
mmc@4022000: 1
```
mmc@4022000 boot still not running - will cross reference u-boot and kernel mmc
```
=> devnum=1; run mmc_boot
Card did not respond to voltage select! : -110
=> devnum=0; run mmc_boot
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1558 bytes read in 1 ms (1.5 MiB/s)
## Executing script at 4fc00000
30988800 bytes read in 2560 ms (11.5 MiB/s)
Failed to load '/h616_dtb'
24422 bytes read in 6 ms (3.9 MiB/s)
Failed to load '/initramfs-linux.img'
Moving Image from 0x40080000 to 0x40200000, end=42020000
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
   Loading Device Tree to 0000000049ff7000, end 0000000049ffffff ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
```
kernel and os recognise eMMC
```
[    3.871736] sunxi-mmc 4022000.mmc: initialized, max. request size: 2048 KB, uses new timings mode
[    4.060429] mmc2: new DDR MMC card at address 0001
[    4.065999] mmcblk2: mmc2:0001 SCA64G 57.7 GiB
[    4.070818] mmcblk2boot0: mmc2:0001 SCA64G 4.00 MiB
[    4.076065] mmcblk2boot1: mmc2:0001 SCA64G 4.00 MiB
[    4.082826]  mmcblk2: p1 p2
[  177.370692] FAT-fs (mmcblk2p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[  196.983296] EXT4-fs (mmcblk2p2): mounted filesystem with ordered data mode. Opts: (null). Quota mode: disabled.
/dev/mmcblk2p1          524008    136600    387408  26% /mnt
/dev/mmcblk2p2        58028230      5304  58018830   0% /mnt2
```

warpme, 2021-06-26:
Rudi,
Only what comes to my mind is to ask apritzel on sunxi irc. 
For me this seems to be like u-boot issue.
Did you try to update bootloaders/u-boot-h616 to current master?

btw: do you know what wifi chip has your tx-6s box?

heitbaum, 2021-06-26:
Image of my mainboard is here:
https://linux-sunxi.org/images/5/5b/Tanix_TX6s_pcb_top.jpg
Chip is labelled as a SP6330-X // ZX191106

These are the details from the Android image:
**Android is using Dongle Host Driver, version 1.579.77.41.11**
- NVRAM version: AP6330_NVRAM_V1.0_20121130
- nvram_ap6330.txt
- fw_bcm40183b2_ag.bin
- config_ap6330.txt
- clm.blob
- F1 signature read @0x18000000=0x16044330
- F1 signature OK, socitype:0x1 chip:0x4330 rev:0x4 pkg:0x0
- Firmware: wl0: Jan 6 2014 15:11:29 version 5.90.195.89.13 FWID 01-72f124c5

warpme, 2021-06-26:
ah - nice.  your box seems to have wifi which should be supported by brcmfmac :-)

Unfortunately i can't find in your dmesg entry like:
`mmc0: new high speed SDIO card at address 0001`

Current mm2 master has 5.13 kernel patch for dt to correctly use gpio to enable power on wifi ( https://github.com/warpme/minimyth2/commit/85d91b26563426c345238282c4a7daa8617a92fd)

In my two h616 devices this commit results I started to see `mmc0: new high speed SDIO card at address 0001` entry in dmesg.
 
Latest mm2 git (jun24 or newer) has added preliminary ap6330 support in kernel, so theoretically it should (after discovering sdio on mmc0), load brcmfmac kernel module & nest wifi firmware (by https://github.com/warpme/minimyth2/commit/d81662d2979516908c08692ec194508b1ee78a17 and https://github.com/warpme/minimyth2/commit/65135208575eca540f2b8981f2b5bc26f9a05dab ).

is manual `modprobe brcmfmac` says something?

heitbaum, 2021-06-26:
```
[   20.781253] sun50i-cpufreq-nvmem: probe of sun50i-cpufreq-nvmem failed with error -2
[   30.079300] random: crng init done
[   30.082732] random: 2 urandom warning(s) missed due to ratelimiting
[   35.605995] cldo1: disabling
[   43.695513] input: eventlircd as /devices/virtual/input/input4
[   55.520953] NET: Registered protocol family 10
[   55.527448] Segment Routing with IPv6
[  923.446097] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  923.461333] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[ 1477.525689] usbcore: registered new interface driver brcmfmac
```
I seem to be running the your patches in the kernel
```
536e646a (HEAD -> addmmc, origin/addmmc, master) add-h616-tanix-tx6s-defconfig.patch: add mmc startup lines [heitbaum]
85024abb (upstream/master) kernel: sync h616 T95 and X96 dt to Tanix-TX6s [Piotr Oniszczuk]
4ad6cf1c kernel: fix h616 cpu opp for dvfs; disable cpufreq-nvmem as it still not works ok [Piotr Oniszczuk]
7898f5b6 (origin/dev, dev) kernel: wip: add di node in h616 dtsi (di not tested yet) [Piotr Oniszczuk]
3a8d2aa1 kernel: add bt and wifi nodes in OrangePi-Zero2 DT [Piotr Oniszczuk]
65135208 firmware: add ap6330 firmware [Piotr Oniszczuk]
d81662d2 kernel: add ap6330 support in 5.13 kernel. [Piotr Oniszczuk]
85d91b26 kernel: wip on tanix-tx6s wifi [Piotr Oniszczuk]
78f3e84f mesa-git: update to current git [Piotr Oniszczuk]
c493c9bc kenrel: update 5.13 kernel to 5.13.-rc7 [Piotr Oniszczuk]
```
the firmware is there:
```
root@tx6s:/lib/firmware # find . | grep 6330
./brcm/brcmfmac-ap6330-sdio.txt
./brcm/brcmfmac-ap6330-sdio.bin
```
I cant find anything special in the Android FDT https://gist.github.com/heitbaum/30676739ea278a4384ceb2f3486ce232#file-tanix-tx6s-fdt-devicetree-from-android-txt-L4458-L4473

### wlan firmware from the Android image
[from_android_image.zip](https://github.com/warpme/minimyth2/files/6720128/from_android_image.zip)

warpme, 2021-06-26:
hmm,
In Yours's android DT for wlan:

```
wlan_regon = <0x53 0x6 0x12 0x1 0xffffffff 0xffffffff 0x0>;
wlan_hostwake = <0x53 0x6 0xf 0x6 0xffffffff 0xffffffff 0x0>;
```
For Opi-zero2
xulong opi-zero2 (where i have schematics) https://github.com/orangepi-xunlong/linux-orangepi/blob/06f40bef0d9a08c233b4b2ed6f9064c3abe3ecb6/arch/arm64/boot/dts/sunxi/sun50i-h616-orangepi-zero2.dts#L4286

```
wlan_regon = <0x53 0x6 0x12 0x1 0xffffffff 0xffffffff 0x0>;
wlan_hostwake = <0x53 0x6 0xf 0x6 0xffffffff 0xffffffff 0x0>;
```
seems the same. 
From schematics of opi-zero2 wlan_reg_on is PG18. 
And this I added to DT and after this OPI-zero2 & my TX6s started to report SDIO dev on mmc0.
If your box reports SDIO on mmc0 - then issue is probably in brcmfmac (maybe chip ID is not matching).
If dmesg not shows any SDIO device - then issue is in gpio controlling WLAN_REG_ON (as this line turns-on wifi chip internal power block supplying wifi block)...
For sure 1st thing is to have SDIO reported in dmesg on mmc0.

btw: Are bootloaders/u-boot-h616 & bootloaders/arm-trusted-firmware-h616 are build exactly like in my repo?

warpme, 2021-06-27:
Rudi,
By curiosity: may You try https://github.com/warpme/minimyth2/releases/download/11.20.0-v32-Pre-2762-gc8c226c/MiniMyth2-armv8-master-11.20.0.r2762-board-h616.tanix_tx6s-SD-Image.img.xz and see dmesg about mmc sdio entries?
this image detects  correctly mmc sdio (and tries load xr819 and aw859a divers for me on my h616 devices).

heitbaum, 2021-07-01:
hmmm - your isnt working either - gues is good that it is behaving the same on both yyyyyyyyyyyyyyyyyyyuand my .img.xz
[dmesg.txt](https://github.com/warpme/minimyth2/files/6744950/dmesg.txt)

warpme, 2022-03-09:
Recent mm2 versions have included Jernej's cw1200 based support for xr819, so I'm closing this issue.

closed 2022-03-09 by warpme

arg7, 2023-03-02:
Hi Warpme,
I have the same issue with the latest MiniArch on Tanix TX6S with SP6330 wifi chip.

SDIO doesn't show up in dmesg, modprobe brcmfmac does nothing.

Another strange thing is that the board has 4GB of RAM, but linux sees only 3GB.
u-boot SPL shows 4GB, and next u-boot 3GB.

Here is log:
U-Boot SPL 2021.04 (Feb 25 2023 - 15:19:32 +0100)
**DRAM: 4096 MiB**
Failed to set core voltage! Can't set CPU frequency
Trying to boot from MMC1
NOTICE:  BL31: v2.8(debug):v12.13.0-v33-Pre-1175-g91e3da68e6-5-g0518a270
NOTICE:  BL31: Built : 15:19:28, Feb 25 2023
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a085a28, model: Tanix TX6s
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
WARNING: PMIC: No regulators DT node, skipping setup
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.

U-Boot 2021.04 (Feb 25 2023 - 15:19:32 +0100) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: Tanix TX6s
I2C:   ready
**DRAM:  3 GiB**
MMC:   mmc@4020000: 0, mmc@4022000: 1
Loading Environment from FAT... Card did not respond to voltage select! : -110
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   No ethernet found.
starting USB...
No working controllers found
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1501 bytes read in 2 ms (732.4 KiB/s)
## Executing script at 4fc00000
32115200 bytes read in 2665 ms (11.5 MiB/s)
Failed to load '/h616_dtb'
27993 bytes read in 9 ms (3 MiB/s)
7630304 bytes read in 636 ms (11.4 MiB/s)
Moving Image from 0x40080000 to 0x40200000, end=42140000
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
   Loading Ramdisk to 498b9000, end 49fffde0 ... OK
   Loading Device Tree to 00000000498af000, end 00000000498b8fff ... OK

Starting kernel ...

:: running early hook [udev]
Starting systemd-udevd version 253-1-arch
:: running hook [udev]
:: Triggering uevents...
:: running hook [keymap]
:: Loading keymap...kbd_mode: KDSKBMODE: Inappropriate ioctl for device
[... 91 lines cut ...]
[  OK  ] Reached target Host and Network Name Lookups.
[  OK  ] Reached target System Initialization.
[  OK  ] Started Refresh existing P…f archlinux-keyring regularly.
[  OK  ] Started Daily verification of password and group files.
[  OK  ] Started Daily Cleanup of Temporary Directories.
[  OK  ] Reached target Timer Units.
[  OK  ] Listening on D-Bus System Message Bus Socket.
[  OK  ] Reached target Socket Units.
[  OK  ] Reached target Basic System.
         Starting D-Bus System Message Bus...
[  OK  ] Started OpenSSH Daemon.
         Starting User Login Management...
         Starting Permit User Sessions...
[  OK  ] Finished Permit User Sessions.
[  OK  ] Started Getty on tty1.
[  OK  ] Started Serial Getty on ttyS0.
[  OK  ] Reached target Login Prompts.
[  OK  ] Started User Login Management.
[  OK  ] Started D-Bus System Message Bus.
[  OK  ] Reached target Multi-User System.
[  OK  ] Reached target Graphical Interface.

Arch Linux 6.2.0 (ttyS0)

alarm login: