warpme/minimyth2 #16: compile kernel and uboot only

tm4dev, 2021-11-08:
Hello,
I've just found your image MiniMyth2-armv8-master-11.23.0.r2941-board-h616.tanix_tx6s-SD-Image.img.xz could be bootable on my tvbox - x96 mate (h616), no error on serial console.

I want to compile kernel and uboot to run ubuntu (or armbian) on it, could you give me some hints? I know this is off-topic and taking your time, but I'm new on compiling kernel and really need your help, please :smile:

tm4dev, 2021-11-08:
I've tried uboot trees from https://github.com/u-boot/u-boot and https://github.com/jernejsk/linux-1/tree/h616-full, but it stuck, here is log

U-Boot SPL 2021.01-g4de1753d12-dirty (Nov 07 2021 - 17:41:37 +0700)           
DRAM: 4096 MiB                                                                
Trying to boot from MMC1                                                      
NOTICE:  BL31: v2.4(debug):v1.2-7312-gbdcbf04f3                               
NOTICE:  BL31: Built : 17:16:28, Nov  7 2021                                  
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)                             
NOTICE:  BL31: Found U-Boot DTB at 0x4a08bc48, model: X96 Mate                
INFO:    ARM GICv2 driver initialized                                         
INFO:    Configuring SPC Controller                                           
INFO:    PMIC: Probing AXP305 on I2C                                          
INFO:    PMIC: aldo1 voltage: 3.300V                                          
INFO:    PMIC: aldo2 voltage: 3.300V                                          
INFO:    PMIC: aldo3 voltage: 3.300V                                          
INFO:    PMIC: bldo1 voltage: 1.800V                                          
INFO:    PMIC: bldo2 voltage: 1.800V                                          
INFO:    PMIC: dcdcd voltage: 1.360V                                          
INFO:    PMIC: dcdce voltage: 3.300V                                          
INFO:    Changed devicetree to reserve BL31 memory.                           
INFO:    BL31: Platform setup done                                            
INFO:    BL31: Initializing runtime services                                  
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied              
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied             
NOTICE:  PSCI: System suspend is unavailable                                  
INFO:    BL31: Preparing for EL3 exit to normal world                         
INFO:    Entry point address = 0x4a000000                                     
INFO:    SPSR = 0x3c9                                                         
                                                                              
                                                                              
U-Boot 2021.01-g4de1753d12-dirty (Nov 07 2021 - 17:41:37 +0700) Allwinner Tecy
                                                                              
CPU:   Allwinner H616 (SUN50I)                                                
Model: X96 Mate                                                               
I2C:   ready                                                                  
DRAM:  4 GiB

warpme, 2021-11-08:
Hi,
This might be a bit more tricky....
1.kernel:

- grab all my 5.14 kernel patches https://github.com/warpme/minimyth2/tree/master/script/kernel/linux-5.14/files

- apply all them in order like in https://github.com/warpme/minimyth2/blob/a0514fc3e155db2ffe61040a5d2dc73d1e288a40/script/kernel/linux-5.14/Makefile line 45 to line 208 on vanilla 5.14.15 kernel sources
- use https://github.com/warpme/minimyth2/blob/master/script/kernel/linux-5.14/files/linux-5.14-arm64-armv8.config as kernel config
- compile your kernel for debian/armbian
- install kernel on SD card you want to use on TX6s

2.bootloader:
Generally You need compile ATF & Uboot then install in correct way on SD card.
Probably easier will be do DD bootloader from MiniMyth2 sdcard image to yours SD card + ask uboot to use DTB for x96 Mate

- to read bootloader form mm2 SD card:

`sudo dd if=/dev/<SD_card> of=tanix-tx6s-bootloader.img bs=1048576 count=16 `

- to write to Your SD card:
```
dd if=tanix-tx6s-bootloader.img of=/dev/<SD_card> conv=fsync bs=1 count=442
dd if=tanix-tx6s-bootloader.img of=/dev/<SD_card> conv=fsync bs=512 skip=1 seek=1
```
- Asking uboot to use DTB for x96 Mate:
use dtbs/allwinner/sun50i-h616-x96-mate.dtb in debian/armbian uboot script

tm4dev, 2021-11-08:
omg, thank you so much! I don't think you reply :smile: 
I'll follow your guide, still need your help then :smiley: thanks!!

warpme, 2021-11-08:
btw: i'm fan of archlinux & https://endeavouros.com
you may look on https://github.com/warpme/archlinuxarm

tm4dev, 2021-11-08:
hmm, if I go for archlinux with your "ArchLinux ARM TanixTX6 image, 5.11 kernel" on my tvbox, what I need are:
1. write ArchLinux-TanixTX6.img to SD card
2. put x96mate dtb on boot partition
3. edit boot.scr to load x96mate dtb
4. compile 5.14 (or 5.15) kernel then replace those binary onto sdcard

Am I missed anything? so what about uboot?

warpme, 2021-11-08:
Oh - i put links to archlinux etc just FYI.
For h616 tx6s you need build archlinux kernel with current patches.
You need any aarch64 box with running archlinuxarm to compile 5.14 kernel with my patches...

tm4dev, 2021-11-08:
bro, thank you a lot! 
I followed your guide and this https://github.com/hexdump0815/linux-mainline-and-mali-allwinner-h6-kernel 
with this tree https://github.com/jernejsk/linux-1
Compiled 5.15.0 kernel and it just work :smile: 
thank you for the very detailed guide :smile:

tm4dev, 2021-11-08:
here the output from serial, now I'm looking for installing os onto internal mmc, any hint please?

U-Boot SPL 2021.04 (Nov 01 2021 - 13:30:22 +0100)
DRAM: 4096 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.5(debug):v11.23.0-5-g02a022c4-dirty
NOTICE:  BL31: Built : 13:30:14, Nov  1 2021
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a085a08, model: Tanix TX6s
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
WARNING: PMIC: No regulators DT node, skipping setup
INFO:    Changed devicetree to reserve BL31 memory.
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9

U-Boot 2021.04 (Nov 01 2021 - 13:30:22 +0100) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: Tanix TX6s
I2C:   ready
DRAM:  3 GiB
MMC:   mmc@4020000: 0, mmc@4022000: 1
Loading Environment from FAT... Card did not respond to voltage select! : -110
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   No ethernet found.
starting USB...
No working controllers found
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1467 bytes read in 3 ms (477.5 KiB/s)
## Executing script at 4fc00000
30403072 bytes read in 5042 ms (5.7 MiB/s)
24596 bytes read in 8 ms (2.9 MiB/s)
Failed to load '/initramfs-linux.img'
Moving Image from 0x40080000 to 0x40200000, end=41f90000
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
   Loading Device Tree to 0000000049ff6000, end 0000000049ffffff ... OK

Starting kernel ...

Welcome to Armbian 21.11.0-trunk.53 Impish!

[  OK  ] Created slice system-modprobe.slice.
[  OK  ] Created slice system-serial\x2dgetty.slice.
[  OK  ] Created slice User and Session Slice.
[  OK  ] Started Forward Password R�…uests to Wall Directory Watch.
[UNSUPP] Starting of Arbitrary Exec�…Automount Point not supported.
[  OK  ] Reached target Local Encrypted Volumes.
[  OK  ] Reached target Slices.
[  OK  ] Reached target Swap.
[  OK  ] Reached target System Time Set.
[  OK  ] Reached target Local Verity Integrity Protected Volumes.
[  OK  ] Listening on RPCbind Server Activation Socket.
[  OK  ] Listening on Syslog Socket.
[  OK  ] Listening on fsck to fsckd communication Socket.
[  OK  ] Listening on initctl Compatibility Named Pipe.
[... 196 lines cut ...]
New root password: ********
Repeat password: *****************
Rejected - passwords do not match. Try again.
New root password: ********
Repeat password: ********

Choose default system command shell:

1) bash
2) zsh

Shell: BASH

Creating a new user account. Press <Ctrl-C> to abort

Please provide a username (eg. your forename): ewww
Create password: ********
Repeat password: ********

Please provide your real name: Ewww

Dear Ewww, your account ewww has been created and is sudo enabled.
Please use this account for your daily work from now on.

root@orangepizero2:~#

closed 2022-03-03 by warpme