warpme/minimyth2 #17: Bug in reading custom evmap for remote

DagNygren, 2022-02-14:
There is a bug in the reader part for using a custom evmap.
The file /etc/rc.d/init.d/conf.d/S_LIRC contains the following code:

```
    if [ "x${MM_LIRC_FETCH_EVMAP_CONF}" = "xyes" ] ; then                                 
                                                                                                   
        /bin/rm -f /etc/eventlircd.d/${MM_LIRC_FETCH_EVMAP_CONF}                                     
                                                                                                     
        mm_confro_get /${MM_LIRC_FETCH_EVMAP_CONF} /etc/eventlircd.d/${MM_LIRC_FETCH_EVMAP_CONF}     
                                                                                                       
        if [ ! -e /etc/eventlircd.d/${MM_LIRC_FETCH_EVMAP_CONF} ] ; then                               
            mm_message_output err "error: failed to fetch ${MM_LIRC_FETCH_EVMAP_CONF} file."             
        fi                                                                                               
                                                                                                         
    fi 

```

And this will only read the file if MM_LIRC_FETCH_EVMAP_CONF equals yes. Now "yes" is not an accepted name for a *.evmap file as it has to be named after the USB id. What follows is that the file will never be read.

Suggest replacing the 

`if [ "x${MM_LIRC_FETCH_EVMAP_CONF}" = "xyes" ] `

with

`if [ "x${MM_LIRC_FETCH_EVMAP_CONF}" != "xno" ] `

warpme, 2022-02-14:
Dag,
Many thx!
Hope it is fixed now in adaa1e1528

DagNygren, 2022-02-16:
NP. Also extracted the squashfs, changed it and repacked. And now the reading of the evmap works for me. One thing though: My fix will not work if the variable is empty. Will there be a default "no" ?

warpme, 2022-02-16:
:-)
Sometime ago i changed config paradigm from: yes/no to: var/<empty>.
reason was to simplify scripts for faster boot

closed 2022-03-03 by warpme