warpme/minimyth2 #19: X96_X6 Blinking with samsung TV

sickosarm, 2022-03-16:
Hi @warpme, nice work, many thanks.

With your image for this TV box i dont get image with my Samsung TV. I have forced 1920x1080 with video-HDMI-A-1:1920X1080@60 in extlinux and get image but turns off every 5 seconds and then on again. With linux kernel 4.19 there is no trouble.

warpme, 2022-03-16:
Hi!

May you pls try:
-50Hz refresh
-different hdmi cable
-lower video mode (i.e. 1280x720)

When you are saying "With linux kernel 4.19" - is this BSP kernel?

Also  -  when you have "turns off every 5 seconds and then on again" - have you any relevant entries in demsg and/or mythtv log?

You may also try with enabling drm debugging: 

`echo 0x16 > /sys/module/drm/parameters/debug`

and inspect dmesg

sickosarm, 2022-03-17:
No chance trying your suggestions.
Do you have hdmi output with your box? 
Linux 4.19 tried is not BSP and have hdmi output with another booting. Maybe in future kernels.....

warpme, 2022-03-17:
Ok thx!
Linux rk356x hdmi support is very initial.
Minimyth2 is based on this and i propose to wait till rk356x hdmi will be mainlined.
After that we can return to issue report - and if issue will be still present - we can issue official bug report in kernel bugzilla.
So now - i propose close this issue.

closed 2022-03-17 by warpme