warpme/minimyth2 #20: Crashes when skipping forward or backward on v31

kjeacle, 2022-03-20:
Posted in the forum a few months ago about this: https://www.tapatalk.com/groups/minimyth2/crashes-when-skipping-forward-or-backward-t160.html

I'm running ram-x86_64-minimyth2-31-11.10.0.r118 and everything works well.

Each new v31 release since then has not worked reliably for me, including ram-x86_64-minimyth2-31-11.23.0.r166.

Skipping forwards or backwards quickly a few times during playback causes a crash. Same thing happens on a few different ION/2 frontends.

There are some logs attached in the forum post above. Thanks!

warpme, 2022-03-21:
Hi Karl,

I see you also started this in MiniMyth2 forums
(https://www.tapatalk.com/groups/minimyth2/crashes-when-skipping-forward-or-backward-t160.html#p631)

Lets continue discussion here...

Looking on Yours logs i see in mythfrontend log:

`UI Screen Resolution: 1830 x 1030`

Are you using specific gui resolution on this frontend?
If yes - may you try with default as much as possible config?
(no custom resolution; use gui resolution for playback; etc)
i.e. change temporarily IP+hostname for frontend to get new settings in DB + use minimal minimyth.conf + set vdpau playback profile with minimal possible customisations?

(in logs i see issue with OpenGL interop layer. In my multiple ion2 frontends i don't have nay issues like yours - and i suspect your issue might be caused by non-standard playback profile settings + nvidia opnegl close-sources drivers)
btw: v31 is quite old, Maybe you can consider to upgrade to v32?

kjeacle, 2022-03-21:
Hi Piotr,

I have removed all my settings and booted with new hostname using only MM2 defaults.

I used IgnoreSchemaVerMismatch + IgnoreProtoVerMismatch to test v32 client with my v31 server.

In summary:

31-11.10.0.r118 - works perfectly
31-11.23.0.r166 - broken
32-11.28.3.r6   - works perfectly

"Broken" means that when I skip back 5 seconds repeatedly, mythfrontend crashes.

I tested three types of recordings:

MPEG-2 704x576
H.264 1920x1080
H.264 1440x1080

2x VDPAU Temporal was automatically selected for playback.

For 704x576, skipping back usually works.
For 1920x1080, it works sometimes but crashes happen.
For 1440x1080, it crashes very easily with a few presses.

I noticed that the error for 1920x1080 was "Waited XXXms for VDPAU context creation" while the error for 1440x1080 was "Waited XXXms for Destroy OpenGL interop".

My server is running FreeBSD and is using pre-compiled v31 packages. I can look into building from source if v32 packages don't appear, but moving to v32 doesn't seem necessary at present.

As 31-11.10.0.r118 works now, and 32-11.28.3.r6 will work in future for me, this will not affect my daily use, so this bug report is really just an FYI. If it's not an obvious fix for the v31 releases then feel free to close this as I will ultimately move to v32.

warpme, 2022-03-21:
Thx for info!
I conclude issue is regression in v31 branch and it is not present in v32.
v32 has highly reworked video stack so thats why probably v32 works again ok for you.  
Quick look on backports in v31 shows me low chances that regression was fixed.

Also: v31 is so old that i think it is not worth to further investigate where is v31 regression. 
v32 works well you - so let's close this.

closed 2022-03-21 by warpme