warpme/minimyth2 #22: X96_X6_RK3566 on another RK3566 based box

andyshrk, 2022-04-03:
First I should say thanks for your nice work。
When see you try this project on rk3566， I also want to have a try。
I have an RK3566 based box（Not X96）。
I Download your X96 image from the release page， and write to sdcard by Etcher。
After insert the sdcard to board and powerup，I see u-boot  start log from the debug uart：
`U-Boot 2017.09 (Apr 02 2022 - 15:51:26 +0200)

Model: X96 X6 RK3566 TV Box
PreSerial: 2, raw, 0xfe660000
DRAM:  3.7 GiB
Sysmem: init
Relocation Offset: ef4c4000, fdt: edbf9dd0 
no mmc device at slot 1
dwmmc@fe2b0000: 1 (SD), dwmmc@fe2c0000: 2, sdhci@fe310000: 0
Bootdev(atags): mmc 1
MMC1: Legacy, 50Mhz
PartType: EFI
No misc partition
boot mode: None
FIT: No boot partition
No resource partition
Failed to load DTB
Failed to get kernel dtb, ret=-19
io-domain: OK
*** Warning - bad CRC, using default environment

Model: X96 X6 RK3566 TV Box
rockchip_set_serialno: could not find efuse/otp device
CLK: (sync kernel. arm: enter 816000 KHz, init 816000 KHz, kernel 0N/A)
  apll 816000 KHz
  dpll 528000 KHz
  gpll 1188000 KHz
  cpll 1000000 KHz
  npll 24000 KHz
  vpll 24000 KHz
  hpll 24000 KHz
  ppll 100000 KHz
  armclk 816000 KHz
  aclk_bus 150000 KHz
  pclk_bus 50000 KHz
  aclk_top_high 300000 KHz
  aclk_top_low 200000 KHz
  hclk_top 150000 KHz
  pclk_top 50000 KHz
  aclk_perimid 300000 KHz
  hclk_perimid 150000 KHz
  pclk_pmu 50000 KHz
Net:   clk_get_by_name(master_bus) failed: -22eqos_probe_resources() failed: -22No ethernet found.
Hit key to stop autoboot('CTRL+C'):  0 
## Booting FIT Image FIT: No boot partition
FIT: No FIT image
Could not find misc partition
ANDROID: reboot reason: "(none)"
optee api revision: 2.0
TEEC: Waring: Could not find security partition
Not AVB images, AVB skip
android_image_load_by_partname: Can't find part: boot
Android image load failed
Android boot failed, error -1.
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:4...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
reading /extlinux/extlinux.conf
400 bytes read in 6 ms (64.5 KiB/s)
1:      MiniMyth2
Retrieving file: /Image
reading /Image
30980608 bytes read in 2604 ms (11.3 MiB/s)
append: root=/dev/mmcblk0p5 rw rootwait earlycon=uart8250,mmio32,0xfe660000 console=ttyS2,1500000n8 ogo.nologo vt.cur_def
ault=1
Retrieving file: /dtbs/rockchip/rk3566-x96_x6.dtb
reading /dtbs/rockchip/rk3566-x96_x6.dtb
47581 bytes read in 13 ms (3.5 MiB/s)
Fdt Ramdisk skip relocation
## Flattened Device Tree blob at 0x0a100000
   Booting using the fdt blob at 0x0a100000
   Using Device Tree in place at 000000000a100000, end 000000000a10e9dc
Adding bank: 0x00200000 - 0x08400000 (size: 0x08200000)
Adding bank: 0x09400000 - 0xf0000000 (size: 0xe6c00000)
Total: 3590.876 ms

Starting kernel ...

I/TC: Secondary CPU 1 initializing
I/TC: Secondary CPU 1 switching to normal world boot
I/TC: Secondary CPU 2 initializing
I/TC: Secondary CPU 2 switching to normal world boot
I/TC: Secondary CPU 3 initializing
I/TC: Secondary CPU 3 switching to normal world boot`

I plug an eth。
And I can see  MiniMyth2  logo on my HDMI monitor

Sometims I can see  the “MythTV Startup Status” UI， I got this by many times reset the board。

But I got “Database failed to start” message。
Does that means I have to install mythtv backend on somewhere？ I am sorry if I miss something， this is my fisrt time play with MythTV。

And it seems there is no linux kernel boot logo output on my debug uart？ how can i mdodify the configuration to get uart console work？

warpme, 2022-04-03:
Hi Andy,

Great you decided to try my work.
It is also valuable to get support for more rk35xx devices.
From Your report I see you get almost working MiniMyth2 on your box.
So I have some Q:

1. what exactly tvbox are you using?

2. you mention "_Sometims I can see the “MythTV Startup Status” UI， I got this by many times reset the board。_"
So i see boot is somehow predictable on your box.
When you don't see MythTV Startup Status” UI - what you see/what happens?

3. “_Database failed to start” message_" - it is expected when you don't have/have not running mythtv backed (with it's database). To get working MiniMyth2 - You need mythtv backend setup & running. You may look here: https://www.mythtv.org/wiki/Operating_system for distro you are using. Generally: mythtv setup is not easy thing - but once you get it running - you might be very satisfied by what it offers you...

4. _And it seems there is no linux kernel boot logo output on my debug uart？ how can i modify the configuration to get uart console work？_ This is result of extlinux.conf file in /boot partition on SDcard. Removing `logo.nologo`should bring back logo. 

5. May you do favour for me and: 
-comment line5 and uncomment line8 in extlinux.conf file in /boot
-boot your tvbox
-send me uart output

andyshrk, 2022-04-03:
> Hi Andy,
> 
> Great you decided to try my work. It is also valuable to get support for more rk35xx devices. From Your report I see you get almost working MiniMyth2 on your box. So I have some Q:
> 
> 1. what exactly tvbox are you using?
     I use a RK3566_BOX_DEMO1 which made as a box sdk board by rockchip. I think the hardware is very similar with your board。

> 2. you mention "_Sometims I can see the “MythTV Startup Status” UI， I got this by many times reset the board。_"
>    So i see boot is somehow predictable on your box.
>    When you don't see MythTV Startup Status” UI - what you see/what happens?

    It seems that the system will stuck a long time when loading wifi firmware. I can see this log now:

[   15.219483] rk_gmac-dwmac fe010000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[   15.221732] rk_gmac-dwmac fe010000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   15.232940] dwmac4: Master AXI performs any burst length
[   15.233421] rk_gmac-dwmac fe010000.ethernet eth0: No Safety Features support found
[   15.234103] rk_gmac-dwmac fe010000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   15.235081] rk_gmac-dwmac fe010000.ethernet eth0: registered PTP clock
[   15.236184] rk_gmac-dwmac fe010000.ethernet eth0: configuring for phy/rgmii link mode
[   17.282643] rk_gmac-dwmac fe010000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   33.759478] vcc5v0_usb20_host: disabling
[   67.893070] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac4359-sdio.bin failed with error -2
[   67.893933] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac4359-sdio.bin
[  130.336452] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

> 3. “_Database failed to start” message_" - it is expected when you don't have/have not running mythtv backed (with it's database). To get working MiniMyth2 - You need mythtv backend setup & running. [...]

 Thank you, I will try.

> 4. _And it seems there is no linux kernel boot logo output on my debug uart？ how can i modify the configuration to get uart console work？_ This is result of extlinux.conf file in /boot partition on [...]

Thank you again,  I can have the uart console output now:

U-Boot 2017.09 (Apr 02 2022 - 15:51:26 +0200)

Model: X96 X6 RK3566 TV Box
PreSerial: 2, raw, 0xfe660000
DRAM:  3.7 GiB
Sysmem: init
Relocation Offset: ef4c4000, fdt: edbf9dd0 
no mmc device at slot 1
dwmmc@fe2b0000: 1 (SD), dwmmc@fe2c0000: 2, sdhci@fe310000: 0
Bootdev(atags): mmc 1
MMC1: Legacy, 50Mhz
PartType: EFI
No misc partition
boot mode: None
FIT: No boot partition
No resource partition
Failed to load DTB
Failed to get kernel dtb, ret=-19
io-domain: OK
*** Warning - bad CRC, using default environment

Model: X96 X6 RK3566 TV Box
rockchip_set_serialno: could not find efuse/otp device
CLK: (sync kernel. arm: enter 816000 KHz, init 816000 KHz, kernel 0N/A)
  apll 816000 KHz
  dpll 528000 KHz
  gpll 1188000 KHz
  cpll 1000000 KHz
  npll 24000 KHz
  vpll 24000 KHz
  hpll 24000 KHz
  ppll 100000 KHz
  armclk 816000 KHz
  aclk_bus 150000 KHz
  pclk_bus 50000 KHz
  aclk_top_high 300000 KHz
[... 845 lines cut ...]
+ local LOG_DIR
+ local LOG_FILE
+ local RETURN
+ [ xyes = xyes ]
+ LOG_DIR=/var/log
+ LOG_FILE=/var/log/minimyth.log
+ [ ! -e /var/log/minimyth.log ]
+ RETURN=0
+ [ -w /var/log/minimyth.log ]
+ /bin/echo --- execution start: /etc/rc.d/init.d/dhcp start
+ /etc/rc.d/init.d/dhcp start
[   15.219483] rk_gmac-dwmac fe010000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[   15.221732] rk_gmac-dwmac fe010000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   15.232940] dwmac4: Master AXI performs any burst length
[   15.233421] rk_gmac-dwmac fe010000.ethernet eth0: No Safety Features support found
[   15.234103] rk_gmac-dwmac fe010000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   15.235081] rk_gmac-dwmac fe010000.ethernet eth0: registered PTP clock
[   15.236184] rk_gmac-dwmac fe010000.ethernet eth0: configuring for phy/rgmii link mode
[   17.282643] rk_gmac-dwmac fe010000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   33.759478] vcc5v0_usb20_host: disabling
[   67.893070] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac4359-sdio.bin failed with error -2
[   67.893933] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac4359-sdio.bin
[  130.336452] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

It stuck here。 And I can‘t do any input。

warpme, 2022-04-03:
I uploaded [MiniMyth2-armv8-master-11.29.2.r188-board-rk3566.x96_x6-SD-Image.img.xz](https://github.com/warpme/minimyth2/releases/download/v11.29.1-v32.0-13-gf69ce764b7/MiniMyth2-armv8-master-11.29.2.r188-board-rk3566.x96_x6-SD-Image.img.xz) which:
-added firmware to wifi You have in Your board
-has dedicated DT for Your board

So, pls:
-burn it on SD card
-edit extlinux.conf and:
1.replace rk3566-x96_x6.dtb to rk3566-box-demo-v10.dtb
2.comment line#5 & uncomment line#8
-look into uart log: hope wifi will load firmware

andyshrk, 2022-04-04:
Hi warpme：
When boot your new img without any modification， no edit extlinx.conf and nor replace dtb。The board can boot into MythTV startup status UI very soon。 But still “Database failed to start”， I think this is because i don’t setup mythtv database correctly。

This is very promising， I will check mythtv database setup again。

After I edit extlinx.con and replace rk3566-x96_x6.dtb with rk3566-box-demo-v10.dtb,  when the board run, I only get four penguin ， no  MythTV ui。
This is the log：
U-Boot 2017.09 (Apr 02 2022 - 15:51:26 +0200)

Model: X96 X6 RK3566 TV Box
PreSerial: 2, raw, 0xfe660000
DRAM:  3.7 GiB
Sysmem: init
Relocation Offset: ef4c4000, fdt: edbf9dd0 
no mmc device at slot 1
dwmmc@fe2b0000: 1 (SD), dwmmc@fe2c0000: 2, sdhci@fe310000: 0
Bootdev(atags): mmc 1
MMC1: Legacy, 50Mhz
PartType: EFI
No misc partition
boot mode: None
FIT: No boot partition
No resource partition
Failed to load DTB
Failed to get kernel dtb, ret=-19
io-domain: OK
*** Warning - bad CRC, using default environment

Model: X96 X6 RK3566 TV Box
rockchip_set_serialno: could not find efuse/otp device
CLK: (sync kernel. arm: enter 816000 KHz, init 816000 KHz, kernel 0N/A)
  apll 816000 KHz
  dpll 528000 KHz
  gpll 1188000 KHz
  cpll 1000000 KHz
  npll 24000 KHz
  vpll 24000 KHz
  hpll 24000 KHz
  ppll 100000 KHz
  armclk 816000 KHz
  aclk_bus 150000 KHz
  pclk_bus 50000 KHz
  aclk_top_high 300000 KHz
  aclk_top_low 200000 KHz
  hclk_top 150000 KHz
  pclk_top 50000 KHz
  aclk_perimid 300000 KHz
  hclk_perimid 150000 KHz
  pclk_pmu 50000 KHz
Net:   clk_get_by_name(master_bus) failed: -22eqos_probe_resources() failed: -22No ethernet found.
Hit key to stop autoboot('CTRL+C'):  0 
## Booting FIT Image FIT: No boot partition
FIT: No FIT image
Could not find misc partition
ANDROID: reboot reason: "(none)"
optee api revision: 2.0
TEEC: Waring: Could not find security partition
Not AVB images, AVB skip
android_image_load_by_partname: Can't find part: boot
Android image load failed
Android boot failed, error -1.
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:4...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
reading /extlinux/extlinux.conf
400 bytes read in 6 ms (64.5 KiB/s)
1:      MiniMyth2
Retrieving file: /Image
[... 818 lines cut ...]
+ mm_command_run /etc/rc.d/init.d/dhcp start
+ local COMMAND=/etc/rc.d/init.d/dhcp start
+ local LOG_DIR
+ local LOG_FILE
+ local RETURN
+ [ xyes = xyes ]
+ LOG_DIR=/var/log
+ LOG_FILE=/var/log/minimyth.log
+ [ ! -e /var/log/minimyth.log ]
+ RETURN=0
+ [ -w /var/log/minimyth.log ]
+ /bin/echo --- execution start: /etc/rc.d/init.d/dhcp start
+ /etc/rc.d/init.d/dhcp start
[   16.249605] rk_gmac-dwmac fe010000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[   16.251806] rk_gmac-dwmac fe010000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   16.263009] dwmac4: Master AXI performs any burst length
[   16.263491] rk_gmac-dwmac fe010000.ethernet eth0: No Safety Features support found
[   16.264172] rk_gmac-dwmac fe010000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   16.265156] rk_gmac-dwmac fe010000.ethernet eth0: registered PTP clock
[   16.266345] rk_gmac-dwmac fe010000.ethernet eth0: configuring for phy/rgmii link mode
[   18.322731] rk_gmac-dwmac fe010000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   33.759606] vcc5v0_usb20_host: disabling
[   67.893182] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac4359-sdio.txt failed with error -2
[   67.894048] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac4359-sdio.txt
[  130.341085] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50

warpme, 2022-04-04:
Andy

1.
You mention: _After I edit extlinx.con and replace rk3566-x96_x6.dtb with rk3566-box-demo-v10.dtb_

but in attached log i see:
```
1: MiniMyth2
Retrieving file: /Image
reading /Image
30980608 bytes read in 2605 ms (11.3 MiB/s)
append: root=/dev/mmcblk0p5 rw rootwait earlycon=uart8250,mmio32,0xfe660000 console=ttyS2,1500000n8 debug MM_DEBUG="yes"
Retrieving file: /dtbs/rockchip/rk3566-x96_x6.dtb
reading /dtbs/rockchip/rk3566-x96_x6.dtb
47613 bytes read in 14 ms (3.2 MiB/s)
```
this tell me you probably overwrite `rk3566-x96_x6.dtb` file with `rk3566-box-demo-v10.dtb`
there is no need to do that.
You can simply change line:
`FDT /dtbs/rockchip/rk3566-x96_x6.dtb`
to
`FDT /dtbs/rockchip/rk3566-box-demo-v10.dtb`

2.
boot hang with rk3566-box-demo-v10.dtb is because we are missing `brcmfmac4359-sdio.rockchip,rk3566-box-demo-v10.txt` firmware file. I already fixed this. Currently You may go with x96 DT.

3.
Regarding accessing database: pls look here: https://github.com/warpme/minimyth2/wiki/Running-on-ARM-boards-or-TVboxes

Shaggy013, 2022-04-04:
the box-demo dts runs on almost every tvbox  in downstream (android)

the ir ports and the vfd used mostly the biggest diffrence in the decomiled dtb files 
x88 thought was even a spi 

so far i could see were the h96 and m2 the only 2 with a rk809 
and the others are rk817

warpme, 2022-04-05:
> the box-demo dts runs on almost every tvbox in downstream (android)

Isn't android using 4.19 kernel?
I think so - so simple reuse DT from android is no an option.
(porting dt from android 4.19 to mainline 5.17 is almost full rewrite)  

>the ir ports and the vfd used mostly the biggest diffrence in the decomiled dtb files
x88 thought was even a spi

indeed. unfortunately i don't have demo box hw - so adding full support will be a bit difficult.

>and the others are rk817

i'm not sure....
i.e. x96-x6 is PMIC-less box....

andyshrk, 2022-04-05:
Hi warpme：
> Andy
> 
> 
> You mention: _After I edit extlinx.con and replace rk3566-x96_x6.dtb with rk3566-box-demo-v10.dtb_
> 
> but in attached log i see:
> 
> ```
> 1: MiniMyth2
> Retrieving file: /Image
> reading /Image
> 30980608 bytes read in 2605 ms (11.3 MiB/s)
> append: root=/dev/mmcblk0p5 rw rootwait earlycon=uart8250,mmio32,0xfe660000 console=ttyS2,1500000n8 debug MM_DEBUG="yes"
> Retrieving file: /dtbs/rockchip/rk3566-x96_x6.dtb
> reading /dtbs/rockchip/rk3566-x96_x6.dtb
> 47613 bytes read in 14 ms (3.2 MiB/s)
> ```
> 
> this tell me you probably overwrite `rk3566-x96_x6.dtb` file with `rk3566-box-demo-v10.dtb` there is no need to do that. You can simply change line: `FDT /dtbs/rockchip/rk3566-x96_x6.dtb` to `FDT /dtbs/rockchip/rk3566-box-demo-v10.dtb`
> 
> 
> boot hang with rk3566-box-demo-v10.dtb is because we are missing `brcmfmac4359-sdio.rockchip,rk3566-box-demo-v10.txt` firmware file. I already fixed this. Currently You may go with x96 DT.
> 
> 
> Regarding accessing database: pls look here: https://github.com/warpme/minimyth2/wiki/Running-on-ARM-boards-or-TVboxes

May I ask how to check what my database name/user/passwd? Are they the imformation stored in  cat /home/mythtv/.mythtv/config.xml  ?

warpme, 2022-04-05:
> May I ask how to check what my database name/user/passwd? Are they the imformation stored in cat /home/mythtv/.mythtv/config.xml ?

Yes.
user/pass in config.xml file must correspond to user/pass set in DB.
MythTV uses config.xml user/pass to access DB. When DB receives request access - DB compares this user/pass with stored-in-DB user/pass and if it matches - DB allows myth to access data stored in DB.

I think it is worth to look on some good wikis.
https://www.mythtv.org/wiki/Database_Setup

ArchLinux is really good one (imho): https://wiki.archlinux.org/title/MythTV

You may also look/ask on  forums: https://forum.mythtv.org

Shaggy013, 2022-04-05:
maybey adding the ir and vfd configs in a overlay and the box-demo-dts as base so they can activate afterwards 

like almost every tvbox firmware works on a quartz64 except the h96 and m2 

and a x88 works with the dowmstream box-demo-dts  

 do you have a img of the box-demo ?

warpme, 2022-04-05:
> maybey adding the ir and vfd configs in a overlay and the box-demo-dts as base so they can activate afterwards

Unfortunately there is no overlays support in 5.17 mainline :-(
But vfd is box specific - so i think dt is best place for vfd config.
But IR  - indeed is perfect use-case for overlays - for remote key mapping.
If demo-box has standard wiring - then IR should work as on rk35xx it is served by PWM3 or by PWM7 pin. 

> and a x88 works with the dowmstream box-demo-dts

isn't x88 an rk 3318 box?

> do you have a img of the box-demo ?

unfortunately no.
Ideally will be to have hw....or de-compiled android dts

andyshrk, 2022-04-06:
> maybey adding the ir and vfd configs in a overlay and the box-demo-dts as base so they can activate afterwards
> 
> like almost every tvbox firmware works on a quartz64 except the h96 and m2
> 
> and a x88 works with the dowmstream box-demo-dts
> 
> do you have a img of the box-demo ?

Do you mean the android image for rk3566-box-demo ?

Shaggy013, 2022-04-06:
wich one :) there isnt one

andyshrk, 2022-04-06:
> Yes. user/pass in config.xml file must correspond to user/pass set in DB. MythTV uses config.xml user/pass to access DB. When DB receives request access - DB compares this user/pass with [...]

I have use the user/passwd from my  `/home/mythtv/.mythtv/config.xml`, set them in `minimyth.conf` and also set the **MM_MASTER_SERVER** with the ip of the ubuntu 20.04 which I install Mythtv backend by MCP. And MythTV version is 32.
I can confirm this when the board startup and Launch the **MythTV Startup status UI**  at the **Setup** page.
I can also confirm that the mythbackend is running because i can see it when I run the **top** command.
The database seems has run because i can connect it with **mysql -u mythtv -pPASSWD mythconverg** command.
But the board still say **Database failed to start**.

is there a way to enable the uart console input ? so maybe i can find more information.

warpme, 2022-04-06:
Andy,

Running minimyth2 has for you:
-telnet (user:root; no password)
-web server (so you can see i.e. box status)
-web server port 8080 - you can browse filesystem i.e. to examine logs

in telnet you have i.e.  midnight commander

So what you see regarding DB when you type in webbrowser:  `http://<box ip>:8080/var/log/mythfrontend.log`

andyshrk, 2022-04-06:
> Andy,
>
> Running minimyth2 has for you: -telnet (user:root; no password) -web server (so you can see i.e. box status) -web server port 8080 - you can browse filesystem i.e. to examine logs
>
> in telnet [...]

The problem is my board run in an office with many computers, I still don't know the box ip.

warpme, 2022-04-06:
To check access to DB from box telnet:
`mysql -umythtv -pmythtv -Dmythconverg -h<box IP>`
you should see maria DB prompt....

warpme, 2022-04-06:
> The problem is my board run in an office with many computers, I still don't know the box ip.

If you know MAC address of box - you can use `arp -n` in terminal in any machine then find entry with box MAC addr...

andyshrk, 2022-04-10:
> http://<box ip>:8080/var/log/mythfrontend.log

After trying many times，I finally get the IP of my box_dmeo.
```
Myth-Frontend-06c7e973c2f1 login: root

BusyBox v1.33.1 (2022-02-19 23:23:15 CET) built-in shell (ash)

root@Myth-Frontend-06c7e973c2f1:~ # dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 5.17.1 (andy@B150) (aarch64-linux-gnu-gcc (Linaro GCC 6.3-2017.05) 6.3.1 20170404, GNU ld (Linaro_Binutils-2017.05) 2.27.0.20161019) #3 SMP PREEMPT Sun Apr 10 16:09:52 CST 2022
[    0.000000] Machine model: Rockchip RK3566 BOX DEMO V10 Board
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000083fffff]
[    0.000000]   node   0: [mem 0x0000000009400000-0x00000000efffffff]
[    0.000000]   node   0: [mem 0x00000001f0000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] On node 0, zone DMA: 4096 pages in unavailable ranges
[    0.000000] cma: Reserved 256 MiB at 0x00000000e0000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 29 pages/cpu s79640 r8192 d30952 u118784
[    0.000000] pcpu-alloc: s79640 r8192 d30952 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: patching kernel code
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1027656
[    0.000000] Kernel command line: root=/dev/mmcblk0p5 rw rootwait earlycon=uart8250,mmio32,0xfe660000 console=ttyS2,1500000n8 debug MM_DEBUG="yes"
[    0.000000] Unknown kernel command line parameters "MM_DEBUG=yes", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x00000000dc000000-0x00000000e0000000] (64MB)
[    0.000000] Memory: 3735728K/4175872K available (14272K kernel code, 3374K rwdata, 6472K rodata, 6208K init, 559K bss, 178000K reserved, 262144K cma-reserved)
[    0.000000] random: get_random_u64 called from cache_random_seq_create+0x88/0x170 with crng_init=0
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 48712 entries in 191 pages
[    0.000000] ftrace: allocated 191 pages with 7 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 30 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS [mem 0xfd440000-0xfd45ffff]
[... 617 lines cut ...]
1970-01-01 08:02:51.201090 I  OpenGL: Qt platform          : eglfs
1970-01-01 08:02:51.201221 I  OpenGL: EGL display          : Yes
1970-01-01 08:02:51.201227 I  OpenGL: Qt OpenGL module     : OpenGL ES
1970-01-01 08:02:51.201233 I  OpenGL: Qt OpenGL format     : OpenGLES 3.1
1970-01-01 08:02:51.201238 I  OpenGL: Qt OpenGL surface    : RGBA: 8:8:8:0 Depth: 0 Stencil: 0
1970-01-01 08:02:51.201245 I  OpenGL: Max texture size     : 8192
1970-01-01 08:02:51.201253 I  OpenGL: Shaders              : Vertex,Fragment,Compute
1970-01-01 08:02:51.201259 I  OpenGL: 16bit framebuffers   : No
1970-01-01 08:02:51.273517 I  OpenGL: Initialised MythRenderOpenGL
1970-01-01 08:02:51.273532 I  OpenGL: Using full range output
1970-01-01 08:02:51.356351 I  UICache: Pruning cache directory: /home/minimyth/.mythtv/cache/remotecache
1970-01-01 08:02:51.356583 I  UICache: Kept 0 files, deleted 0 files, stat error on 0 files
[... cut ...]
1970-01-01 08:03:17.377869 I  Start up testing connections. DB 172.16.12.141, BE , attempt 7, status dbAwake, Delay: 2000
1970-01-01 08:03:22.428551 I  Start up testing connections. DB 172.16.12.141, BE , attempt 8, status dbAwake, Delay: 2000
1970-01-01 08:03:27.478924 I  Start up testing connections. DB 172.16.12.141, BE , attempt 9, status dbAwake, Delay: 2000
1970-01-01 08:03:32.529625 I  Start up testing connections. DB 172.16.12.141, BE , attempt 10, status dbAwake, Delay: 2000
1970-01-01 08:03:37.581303 I  Start up failure. host 172.16.12.141, status dbAwakeFail
```

warpme, 2022-04-10:
Did you check access to DB from box telnet:
`mysql -umythtv -pmythtv -Dmythconverg -h<box IP>`
Did you see maria DB prompt....

andyshrk, 2022-04-11:
> Did you check access to DB from box telnet: `mysql -umythtv -pmythtv -Dmythconverg -h<box IP>` Do you see maria DB prompt with above command?

It seems the same error.

```
andy@B150:~$ telnet 172.16.12.191
Trying 172.16.12.191...
Connected to 172.16.12.191.
Escape character is '^]'.

? login: root

BusyBox v1.33.1 (2022-02-19 23:23:15 CET) built-in shell (ash)

root@?:~ # 
root@?:~ # 
root@?:~ # mysql -umythtv -p4Jess9tO -Dmythconverg -h172.16.12.141
ERROR 2002 (HY000): Can't connect to server on '172.16.12.141' (115)
root@?:~ # 
root@?:~ # ls
root@?:~ # ls /home/
minimyth
root@?:~ # ls /home/minimyth/.
.blackboxrc    .config/       .kodi/         .mame/         .mythtv/       .stella/       .xscreensaver
.cache/        .kde/          .lircrc        .minimyth/     .ssh/          .toprc
root@?:~ # ls /home/minimyth/.config/
htop/       weston.ini
root@?:~ # ls /home/minimyth/.
.blackboxrc    .config/       .kodi/         .mame/         .mythtv/       .stella/       .xscreensaver
.cache/        .kde/          .lircrc        .minimyth/     .ssh/          .toprc
root@?:~ # ls /home/minimyth/.
.blackboxrc    .config/       .kodi/         .mame/         .mythtv/       .stella/       .xscreensaver
.cache/        .kde/          .lircrc        .minimyth/     .ssh/          .toprc
root@?:~ # ls /home/minimyth/.mythtv/
3rdParty/      channels/      lircrc         pytmdb3.cache  tmp/
cache/         config.xml     mysql.txt      themes/
root@?:~ # cat /home/minimyth/.mythtv/config.xml 
<Configuration>
  <UPnP>
    <MythFrontend>
      <DefaultBackend>
        <LocalHostName>Myth-Frontend-06c7e973c2f1</LocalHostName>
        <DBHostName>172.16.12.141</DBHostName>
        <DBUserName>mythtv</DBUserName>
        <DBPassword>4Jess9tO</DBPassword>
        <DBName>mythconverg</DBName>
        <DBPort>0</DBPort>
      </DefaultBackend>
    </MythFrontend>
  </UPnP>
</Configuration>
root@?:~ # mysql -umythtv -p4Jess9tO -Dmythconverg -h172.16.12.141
ERROR 2002 (HY000): Can't connect to server on '172.16.12.141' (115)
root@?:~ # 
```

On my backend:
```
andy@B150:~$ sudo mysql -uroot -p mythconverg
[sudo] password for andy: 
Sorry, try again.
[sudo] password for andy: 
Enter password: 
Reading table information for completion of table and column names
You can turn off this feature to get a quicker startup with -A

Welcome to the MySQL monitor.  Commands end with ; or \g.
Your MySQL connection id is 2615
Server version: 8.0.28-0ubuntu0.20.04.3 (Ubuntu)

Copyright (c) 2000, 2022, Oracle and/or its affiliates.

Oracle is a registered trademark of Oracle Corporation and/or its
affiliates. Other names may be trademarks of their respective
owners.

Type 'help;' or '\h' for help. Type '\c' to clear the current input statement.

mysql> show grants for mythtv;
+---------------------------------------------------------+
| Grants for mythtv@%                                     |
+---------------------------------------------------------+
| GRANT USAGE ON *.* TO `mythtv`@`%`                      |
| GRANT ALL PRIVILEGES ON `mythconverg`.* TO `mythtv`@`%` |
+---------------------------------------------------------+
2 rows in set (0.00 sec)

mysql> 
```

warpme, 2022-04-11:
```
root@?:~ # mysql -umythtv -p4Jess9tO -Dmythconverg -h172.16.12.141
ERROR 2002 (HY000): Can't connect to server on '172.16.12.141' (115)
```
I think issue is with DB access rights on your server side.
Probably your DB has not set privileges to allow access from external hosts.

andyshrk, 2022-04-14:
After setting bind-address to 0.0.0.0
cat /etc/mysql/mysql.conf.d/mythtv.cnf 
```
[mysqld]
bind-address=0.0.0.0
max_connections=100
```
Now the box can connect to backend database now.
Now I can play a video stored at my backend computer.

I think I have a similary environment with you。
So if you have some develop issue with vop or vop2 drm， maybe I can sync the test case from you and help debug 。

warpme, 2022-04-15:
Andy,
Great you got it working!

For rk356x vop2 issue with drm planes rendering: latest status:
http://lists.infradead.org/pipermail/linux-rockchip/2022-April/031064.html

I think - from user perspective:
-we will develop it working by playing with vop2 code (and not by patching Qt sources as qt users are using pre-build qt binaries)
-if not above - vop2 (and rk356x) will be marked as not working with qt in drm planes due rk356x constrains....

closed 2022-04-22 by andyshrk