warpme/minimyth2 #23: h6 OrangePi 3 support

jimbolaya, 2022-04-05:
I was wondering how difficult it would be to support another h6 board, the OrangePi 3 (http://www.orangepi.org/Orange%20Pi%203/)? I've played with Armbian with this board, but not Arch.

warpme, 2022-04-05:
Hi,

H6 family needs bootloader (u-boot) compiled for target board - so i need to prepare dedicated SD image for OPi3.
I don't have this board to test - so i'll be a bit blind....
Let me try play with this.
Stay tuned....

jimbolaya, 2022-04-05:
Let me know if you would like me to test anything. I've got one of these boards sitting idle for the time being.

warpme, 2022-04-06:
Ok - so beta for opi3: https://github.com/warpme/minimyth2/releases/download/v12.0.0-v33-Pre-193-ge7a9d48089/MiniMyth2-armv8-master-12.0.1.r193-board-h6.orangepi_3-SD-Image.img.xz
it is blind work of me so expect issues....

warpme, 2022-04-08:
Oh.
This is bad.
A bit strange that there is on audio....
OPi3 don't have Eth support on non-patched mainline kernel. 
I was trying to get it working - but it looks i failed :-(
But missing audio is really strange for me!

You mention You have Armbian working ok on Your OPi3?
What kernel?
Are You able identify what patches Armbian applies on mainline kernel to get OPi3 working?

Key issue here is that I don't have HW.
Adding support for new HW without HW is always issue :-(

jimbolaya, 2022-04-08:
Ah. Hrm. I'll have to load the latest Armbian on there. I'm not sure how to get what patches it might have. 

I just remembered too, the last thing I ran on it was LibreELEC. There was a non-standard build for it from one of the threads in the forum: https://forum.libreelec.tv/thread/17565-nightly-images-for-a64-h3-h5-h6-and-r40-boards/

If you're in the US, I'd be happy to send it to you if that would help. 

For Armbian:
Kernel is 5.15.25-sunxi64

cat /proc/asound/cards
 0 [sun9ihdmi      ]: sun9i-hdmi - sun9i-hdmi
                      sun9i-hdmi

So, no analog audio. Reading the the forums, it doesn't seem like that's going to be likely.

From dmesg:
[  157.775011] dwmac-sun8i 5020000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet] (irq=POLL)

If you need extensive output from stuff like that let me know. I could probably arrange for you to be able to ssh into it if you think that would be helpful. Anyway, if it's too much hassle let me know.

jimbolaya, 2022-10-10:
Sure, I'll give it a shot and let you know.

warpme, 2022-10-18:
Any news?
May i close this ticket?

jimbolaya, 2022-10-18:
I apologize, I tried this earlier, but I think the SD card I was using was bad. 

In any case, out of several attempts, I was only able to get video output once. I also tried a different power supply, but still wasn't able to get it to work. 

I'm OK closing this. I've gone in a different direction with a MythTV client at this point.

Thank you for your effort on this, I'm sorry I couldn't help more.

jimbolaya, 2022-10-18:
I'm not sure it ever booted, except the once since I didn't ever get video. It could be my strange video setup which is HDMI to DVI-D in this case. When I did get video, it didn't seem very happy (it said couldn't initialize some hardware).

I haven't yet gone back to see if the device is still usable (maybe it got fried somehow). I am planning at some point to make an Armbian SD Card and see if that still works.

closed 2022-11-02 by warpme