warpme/minimyth2 #25: make install error

andyshrk, 2022-04-08:
Hi warpme:

I try to build this project myself on Ubuntu 20.04 LTS,
but when run `make install` after `make build` success, i get  failed following:
Andy suggestion is appreciated.

`==> Running make install in work/main.d/minimyth-32-11.29.0.r12
make[1]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12'
make[2]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/source'
make[2]: Nothing to be done for 'build'.
make[2]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/source'
make[2]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/source'
made SD card distribution
  using boards list (board-rk3566.x96_x6) from script command-line
  adding board-rk3566.x96_x6 to default.config & MiniMyth2.wks
  board-rk35xx detected. Using default-gpt.wks
  boards        : board-rk3566.x96_x6-
  mm2 home dir  : [/srv/WorkSpace/opensource/minimyth2]
  boot files    : [/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/boot]
  rootfs files  : [/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs]
  entering fakeroot enviroment...
  WIC output:
INFO: Creating image(s)...

ERROR: _exec_cmd: export PATH=/srv/WorkSpace/opensource/minimyth2/images/build/sbin:/srv/WorkSpace/opensource/minimyth2/images/build/usr/sbin:/srv/WorkSpace/opensource/minimyth2/images/build/usr/bin:$PATH;export PSEUDO_PREFIX=/srv/WorkSpace/opensource/minimyth2/images/build/usr;export PSEUDO_LOCALSTATEDIR=/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs/../pseudo;export PSEUDO_PASSWD=/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs;export PSEUDO_NOSYMLINKEXP=1;/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/tmp/sysroots-components/x86_64/pseudo-native/usr/bin/pseudo mkfs.ext4 -F -i 8192 ./tmp.wic.o0bt7zqv/rootfs_ROOT.7.ext4 -L ROOT -U 8caa6358-c5f7-473a-b142-d55e2f506b85 -d /srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs returned '1' instead of 0
output: ERROR: ld.so: object 'libpseudo.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
mkfs.ext4: /lib/x86_64-linux-gnu/libblkid.so.1: version `BLKID_2_36' not found (required by mkfs.ext4)
  removing working files...
  compressing SD image...
xz: MiniMyth2-armv8-32-11.29.0.r12-board-rk3566.x96_x6-SD-Image.img: No such file or directory
  SD image creation done!
  Image is here:/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/MiniMyth2-armv8-32-11.29.0.r12-board-rk3566.x96_x6-SD-Image.img.xz
installing SD Card image on [/home/andy/build]
cp: cannot stat '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/MiniMyth2-*-SD-Image.img.xz': No such file or directory
make[2]: *** [Makefile:1037: install] Error 1
make[2]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/source'
make[1]: *** [Makefile:7: install] Error 2
make[1]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12'
make: *** [../../gar.lib.mk:423: install-work/main.d/minimyth-32-11.29.0.r12/Makefile] Error 2`

warpme, 2022-04-10:
Andy,
Issue seems to be here:
`output: ERROR: ld.so: object 'libpseudo.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.`

When You will run `find / -name libpseudo.so` - what is output?

For future investigation you may:
-change in `{MM_HOME}/images/build/usr/bin/kickstart/create-image.sh`     `debug_verbosity="no"` to `debug_verbosity="yes"`

andyshrk, 2022-04-11:
> Andy, Issue seems to be here: `output: ERROR: ld.so: object 'libpseudo.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.`
> 
> When You will run `find / -name libpseudo.so` - what is output?

the find command output:

```
andy@B150:/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth$ sudo find /* -name libpseudo.so ! -path "/dev/" ! -path "/sys/" ! -path "/proc/"
[sudo] password for andy: 
find: warning: -path /dev/ will not match anything because it ends with /.
find: warning: -path /sys/ will not match anything because it ends with /.
find: warning: -path /proc/ will not match anything because it ends with /.
/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/work/armv8a-poky-linux/dbus/1.12.20-r0/recipe-sysroot-native/usr/lib/pseudo/lib64/libpseudo.so
/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/work/armv8a-poky-linux/rpcbind/1.2.5-r0/recipe-sysroot-native/usr/lib/pseudo/lib64/libpseudo.so
/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/work/armv8a-poky-linux/openssh/8.3p1-r0/recipe-sysroot-native/usr/lib/pseudo/lib64/libpseudo.so
/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/work/armv8a-poky-linux/nfs-utils/2.5.1-r0/recipe-sysroot-native/usr/lib/pseudo/lib64/libpseudo.so
/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/work/armv8a-poky-linux/avahi/0.8-r0/recipe-sysroot-native/usr/lib/pseudo/lib64/libpseudo.so
/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/work/armv8a-poky-linux/systemd/1_246.9-r0/recipe-sysroot-native/usr/lib/pseudo/lib64/libpseudo.so
/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/work/stratix10-poky-linux/linux-socfpga-lts/5.4.104-lts+gitAUTOINC+c7bfe97dd0-r0/recipe-sysroot-native/usr/lib/pseudo/lib64/libpseudo.so
/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/work/stratix10-poky-linux/gsrd-console-image/1.0-r0/recipe-sysroot-native/usr/lib/pseudo/lib64/libpseudo.so
/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/work/stratix10-poky-linux/console-image-minimal/1.0-r0/recipe-sysroot-native/usr/lib/pseudo/lib64/libpseudo.so
/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/work/stratix10-poky-linux/u-boot-socfpga/1_v2021.01+gitAUTOINC+60ef4c2b45-r0/recipe-sysroot-native/usr/lib/pseudo/lib64/libpseudo.so
/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+ee24ebec9e-r0/sysroot-destdir/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/work/x86_64-linux/pseudo-native/1.9.0+gitAUTOINC+ee24ebec9e-r0/recipe-sysroot-native/usr/lib/pseudo/lib64/libpseudo.so
/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/sysroots-components/x86_64/pseudo-native/usr/lib/pseudo/lib64/libpseudo.so
find: ‘/proc/4740/task/4740/net’: Invalid argument
find: ‘/proc/4740/net’: Invalid argument
find: ‘/run/user/1000/doc’: Permission denied
find: ‘/run/user/1000/gvfs’: Permission denied
/srv/WorkSpace/opensource/minimyth2/script/utils/pseudo/work/build.d/pseudo-1.9.0/lib64/libpseudo.so
/srv/WorkSpace/opensource/minimyth2/images/build/usr/lib64/libpseudo.so
andy@B150:/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth$ 

```
> 
> For future investigation you may: -change in `{MM_HOME}/images/build/usr/bin/kickstart/create-image.sh` `debug_verbosity="no"` to `debug_verbosity="yes"`

andyshrk, 2022-04-11:
Here is more information with debug_verbosity enabled.

https://pastebin.com/R0X1mNbu

warpme, 2022-04-11:
Ah - i see
In your system You have also pseudo lib installed by Yocto
`/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/sysroots-components/x86_64/pseudo-native/usr/lib/pseudo/lib64/libpseudo.so` 
May you do experiment with temp renaming/hide above file and try again with mm2 install?

andyshrk, 2022-04-11:
> Ah - i see In your system You have also pseudo lib installed by Yocto `/opt/WorkSpace/vendors/socfpga/yocto/stratix10-rootfs/tmp/sysroots-components/x86_64/pseudo-native/usr/lib/pseudo/lib64/libpseudo.so` May you do experiment with temp renaming/hide above file and try again with mm2 install?

I removed the whole yocto project, and reboot he computer, then run make install again, but still has no luck.

```
EBUG: _exec_cmd: output for du -ks /srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs (rc = 0): 608112	/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs
DEBUG: Requested partition size for /: 0
DEBUG: Added 524288 extra blocks to / to get to 1472120 total blocks
DEBUG: ['mkfs.ext4', '-F', '-i', '8192', './tmp.wic.trnnvnk5/rootfs_ROOT.7.ext4', '-L', 'ROOT', '-U', 'cd648e86-7bb8-45f1-b0ae-df77e47c6f71', '-d', '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs']
DEBUG: exec_native_cmd: export PATH=/srv/WorkSpace/opensource/minimyth2/images/build/sbin:/srv/WorkSpace/opensource/minimyth2/images/build/usr/sbin:/srv/WorkSpace/opensource/minimyth2/images/build/usr/bin:$PATH;export PSEUDO_PREFIX=/srv/WorkSpace/opensource/minimyth2/images/build/usr;export PSEUDO_LOCALSTATEDIR=/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs/../pseudo;export PSEUDO_PASSWD=/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs;export PSEUDO_NOSYMLINKEXP=1;/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/tmp/sysroots-components/x86_64/pseudo-native/usr/bin/pseudo mkfs.ext4 -F -i 8192 ./tmp.wic.trnnvnk5/rootfs_ROOT.7.ext4 -L ROOT -U cd648e86-7bb8-45f1-b0ae-df77e47c6f71 -d /srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs
DEBUG: _exec_cmd: export PATH=/srv/WorkSpace/opensource/minimyth2/images/build/sbin:/srv/WorkSpace/opensource/minimyth2/images/build/usr/sbin:/srv/WorkSpace/opensource/minimyth2/images/build/usr/bin:$PATH;export PSEUDO_PREFIX=/srv/WorkSpace/opensource/minimyth2/images/build/usr;export PSEUDO_LOCALSTATEDIR=/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs/../pseudo;export PSEUDO_PASSWD=/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs;export PSEUDO_NOSYMLINKEXP=1;/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/tmp/sysroots-components/x86_64/pseudo-native/usr/bin/pseudo mkfs.ext4 -F -i 8192 ./tmp.wic.trnnvnk5/rootfs_ROOT.7.ext4 -L ROOT -U cd648e86-7bb8-45f1-b0ae-df77e47c6f71 -d /srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs
DEBUG: ['export', 'PATH=/srv/WorkSpace/opensource/minimyth2/images/build/sbin:/srv/WorkSpace/opensource/minimyth2/images/build/usr/sbin:/srv/WorkSpace/opensource/minimyth2/images/build/usr/bin:$PATH;export', 'PSEUDO_PREFIX=/srv/WorkSpace/opensource/minimyth2/images/build/usr;export', 'PSEUDO_LOCALSTATEDIR=/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs/../pseudo;export', 'PSEUDO_PASSWD=/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs;export', 'PSEUDO_NOSYMLINKEXP=1;/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/tmp/sysroots-components/x86_64/pseudo-native/usr/bin/pseudo', 'mkfs.ext4', '-F', '-i', '8192', './tmp.wic.trnnvnk5/rootfs_ROOT.7.ext4', '-L', 'ROOT', '-U', 'cd648e86-7bb8-45f1-b0ae-df77e47c6f71', '-d', '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs']

ERROR: _exec_cmd: export PATH=/srv/WorkSpace/opensource/minimyth2/images/build/sbin:/srv/WorkSpace/opensource/minimyth2/images/build/usr/sbin:/srv/WorkSpace/opensource/minimyth2/images/build/usr/bin:$PATH;export PSEUDO_PREFIX=/srv/WorkSpace/opensource/minimyth2/images/build/usr;export PSEUDO_LOCALSTATEDIR=/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs/../pseudo;export PSEUDO_PASSWD=/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs;export PSEUDO_NOSYMLINKEXP=1;/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/tmp/sysroots-components/x86_64/pseudo-native/usr/bin/pseudo mkfs.ext4 -F -i 8192 ./tmp.wic.trnnvnk5/rootfs_ROOT.7.ext4 -L ROOT -U cd648e86-7bb8-45f1-b0ae-df77e47c6f71 -d /srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/image/rootfs returned '1' instead of 0
output: ERROR: ld.so: object 'libpseudo.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
mkfs.ext4: /lib/x86_64-linux-gnu/libblkid.so.1: version `BLKID_2_36' not found (required by mkfs.ext4)
  removing working files...
  compressing SD image...
[... cut ...]
find: warning: -path /proc/ will not match anything because it ends with /.
find: ‘/proc/4681/task/4681/net’: Invalid argument
find: ‘/proc/4681/net’: Invalid argument
find: ‘/run/user/1000/doc’: Permission denied
find: ‘/run/user/1000/gvfs’: Permission denied
/srv/WorkSpace/opensource/minimyth2/script/utils/pseudo/work/build.d/pseudo-1.9.0/lib64/libpseudo.so
/srv/WorkSpace/opensource/minimyth2/images/build/usr/lib64/libpseudo.so
andy@B150:/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth$ 

```

warpme, 2022-04-11:
So I have another idea:

May you try replace line62 in  `{MM_HOME}/images/build/usr/bin/kickstart/create-image.sh`
to
`export LD_LIBRARY_PATH=${mm_home}/images/build/lib:${mm_home}/images/build/usr/lib`

andyshrk, 2022-04-11:
> So I have another idea:
> 
> May you try replace line62 in `{MM_HOME}/images/build/usr/bin/kickstart/create-image.sh` to `export LD_LIBRARY_PATH=${mm_home}/images/build/lib:${mm_home}/images/build/usr/lib`

Yes, it is works now:
```
andy@B150:/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth$ make install
[===== NOW BUILDING:	minimyth-32-11.29.0.r12	=====]
make[1]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth'
[===== NOW BUILDING:	minimyth-32-11.29.0.r12	=====]
	[fetch] complete for minimyth.
	[checksum] complete for minimyth.
	[extract] complete for minimyth.
	[patch] complete for minimyth.
	[configure] complete for minimyth.
	[build] complete for minimyth.
checking ...
  basic binaries ...
  build system binaries ...
    'sh' (from package 'bash')
    'ar' (from package 'binutils')
    'as' (from package 'binutils')
    'ld' (from package 'binutils')
    'nm' (from package 'binutils')
    'ranlib' (from package 'binutils')
    'bison' (from package 'bison')
    'bzip2' (from package 'bzip2')
    'basename' (from package 'coreutils')
    'cat' (from package 'coreutils')
    'chmod' (from package 'coreutils')
    'chown' (from package 'coreutils')
    'comm' (from package 'coreutils')
    'cp' (from package 'coreutils')
    'cut' (from package 'coreutils')
    'date' (from package 'coreutils')
    'dirname' (from package 'coreutils')
    'echo' (from package 'coreutils')
    'env' (from package 'coreutils')
    'expr' (from package 'coreutils')
    'false' (from package 'coreutils')
    'head' (from package 'coreutils')
    'id' (from package 'coreutils')
    'install' (from package 'coreutils')
    'ln' (from package 'coreutils')
    'ls' (from package 'coreutils')
    'md5sum' (from package 'coreutils')
    'mkdir' (from package 'coreutils')
    'mv' (from package 'coreutils')
    'od' (from package 'coreutils')
    'rm' (from package 'coreutils')
    'rmdir' (from package 'coreutils')
    'sleep' (from package 'coreutils')
    'sort' (from package 'coreutils')
    'split' (from package 'coreutils')
    'su' (from package 'coreutils')
    'tail' (from package 'coreutils')
    'test' (from package 'coreutils')
    'touch' (from package 'coreutils')
    'tr' (from package 'coreutils')
    'true' (from package 'coreutils')
    'uname' (from package 'coreutils')
    'uniq' (from package 'coreutils')
    'wc' (from package 'coreutils')
    'cvs' (from package 'cvs')
    'cmp' (from package 'diffutils')
    'diff' (from package 'diffutils')
    'file' (from package 'file')
    'find' (from package 'findutils')
    'flex' (from package 'flex')
    'awk' (from package 'gawk')
[... 853 lines cut ...]
DEBUG: _exec_cmd: output for install -m 0644 -D /srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/boot/dtbs/rockchip/rk3566-quartz64-a.dtb ./tmp.wic.vg7apnuz/boot.6/dtbs/rockchip/rk3566-quartz64-a.dtb (rc = 0): 
DEBUG: Install dtbs/rockchip/rk3566-x96_x6.dtb as dtbs/rockchip/rk3566-x96_x6.dtb
DEBUG: _exec_cmd: install -m 0644 -D /srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/boot/dtbs/rockchip/rk3566-x96_x6.dtb ./tmp.wic.vg7apnuz/boot.6/dtbs/rockchip/rk3566-x96_x6.dtb
DEBUG: ['install', '-m', '0644', '-D', '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/boot/dtbs/rockchip/rk3566-x96_x6.dtb', './tmp.wic.vg7apnuz/boot.6/dtbs/rockchip/rk3566-x96_x6.dtb']
DEBUG: _exec_cmd: output for install -m 0644 -D /srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/boot/dtbs/rockchip/rk3566-x96_x6.dtb ./tmp.wic.vg7apnuz/boot.6/dtbs/rockchip/rk3566-x96_x6.dtb (rc = 0): 
DEBUG: Install dtbs/rockchip/rk3399-hugsun-x99.dtb as dtbs/rockchip/rk3399-hugsun-x99.dtb
DEBUG: _exec_cmd: install -m 0644 -D /srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/boot/dtbs/rockchip/rk3399-hugsun-x99.dtb ./tmp.wic.vg7apnuz/boot.6/dtbs/rockchip/rk3399-hugsun-x99.dtb
DEBUG: ['install', '-m', '0644', '-D', '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/boot/dtbs/rockchip/rk3399-hugsun-x99.dtb', './tmp.wic.vg7apnuz/boot.6/dtbs/rockchip/rk3399-hugsun-x99.dtb']
DEBUG: _exec_cmd: output for install -m 0644 -D /srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/boot/dtbs/rockchip/rk3399-hugsun-x99.dtb ./tmp.wic.vg7apnuz/boot.6/dtbs/rockchip/rk3399-hugsun-x99.dtb (rc = 0): 
DEBUG: Install dtbs/rockchip/rk3399-khadas-edge.dtb as dtbs/rockchip/rk3399-khadas-edge.dtb
DEBUG: _exec_cmd: install -m 0644 -D /srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/boot/dtbs/rockchip/rk3399-khadas-edge.dtb ./tmp.wic.vg7apnuz/boot.6/dtbs/rockchip/rk3399-khadas-edge.dtb
DEBUG: ['install', '-m', '0644', '-D', '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/boot/dtbs/rockchip/rk3399-khadas-edge.dtb', './tmp.wic.vg7apnuz/boot.6/dtbs/rockchip/rk3399-khadas-edge.dtb']
DEBUG: _exec_cmd: output for install -m 0644 -D /srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/boot/dtbs/rockchip/rk3399-khadas-edge.dtb ./tmp.wic.vg7apnuz/boot.6/dtbs/rockchip/rk3399-khadas-edge.dtb (rc = 0): 
[... cut ...]
DEBUG: ['install', '-m', '0644', '-D', '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/boot/dtbs/rockchip/rk3326-odroid-go2.dtb', './tmp.wic.vg7apnuz/boot.6/dtbs/rockchip/rk3326-odroid-go2.dtb']
DEBUG: _exec_cmd: output for install -m 0644 -D /srv/WorkSpace/opensource/minimyth2/script/meta/minimyth/work/main.d/minimyth-32-11.29.0.r12/build/stage/boot/dtbs/rockchip/rk3326-odroid-go2.dtb ./tmp.wic.vg7apnuz/boot.6/dtbs/rockchip/rk3326-odroi

warpme, 2022-04-11:
So - it is good it works now.
Issue was bug exposing when glibc used my mm2 is newer than glibc used by build machine os. It was result of overlook in LD_LIBRARY causing that script not looks for mm2 build libs and always looks for system libs - if searched lib is not in usr/lib. Now this is fixed in 363a615931.
I think we can close issue.
May thx for your help!