warpme/minimyth2 #26: u-boot for rk3566 not build

andyshrk, 2022-04-14:
When updated to the bellow commit, I got following compile error,

andy@B150:/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth$ rg 3566  ~/.minimyth2/minimyth.conf.mk 
74:# 'board-rk3566.x96_x6'
94:mm_BOARD_TYPE             ?= board-rk3566.x96_x6

It seems u-boot is not compiled.

ommit 55991fe3fe1dd9d1e90c7801cf44846c8407b1ee (HEAD -> master)
Author: Piotr Oniszczuk <piotr.oniszczuk@gmail.com>
Date:   Mon Apr 11 20:58:03 2022 +0200

    index.html: make monitorix .png on main page always refreshed on page load
```
 //srv/WorkSpace/opensource/minimyth2/images/build/usr/bin/mkdir -p '/srv/WorkSpace/opensource/minimyth2/images/main/usr/include'
 //srv/WorkSpace/opensource/minimyth2/images/build/usr/bin/install -c -m 644 ../../libctf/../include/ctf.h ../../libctf/../include/ctf-api.h '/srv/WorkSpace/opensource/minimyth2/images/main/usr/include'
make[7]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/devel/gdb/work/main.d/gdb-10.2/build/libctf'
make[6]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/devel/gdb/work/main.d/gdb-10.2/build/libctf'
make[5]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/devel/gdb/work/main.d/gdb-10.2/build/libctf'
make[4]: Nothing to be done for 'install-target'.
make[4]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/devel/gdb/work/main.d/gdb-10.2/build'
make[3]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/devel/gdb/work/main.d/gdb-10.2/build'
echo " ==> Installing GPL2 license text"
 ==> Installing GPL2 license text
install -d /srv/WorkSpace/opensource/minimyth2/images/main/usr/licenses/gdb/
install -m 644 ../../licenses/GPL2.txt /srv/WorkSpace/opensource/minimyth2/images/main/usr/licenses/gdb/
mkdir -p cookies/main.d/. && date >> cookies/main.d/install-license-GPL2
echo " ==> Installing LGPL2 license text"
 ==> Installing LGPL2 license text
install -d /srv/WorkSpace/opensource/minimyth2/images/main/usr/licenses/gdb/
install -m 644 ../../licenses/LGPL2.txt /srv/WorkSpace/opensource/minimyth2/images/main/usr/licenses/gdb/
mkdir -p cookies/main.d/. && date >> cookies/main.d/install-license-LGPL2
	[install] complete for gdb.
make[2]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/devel/gdb'
 ==> Building bootloaders/bootloader as a dependency
make[2]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/bootloader'
[===== NOW BUILDING:	bootloader-	=====]
	[fetch] complete for bootloader.
	[checksum] complete for bootloader.
install -d work/main.d
	[extract] complete for bootloader.
install -d work/main.d/bootloader-
	[patch] complete for bootloader.
make[3]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/bootloader'
make[3]: '../../devel/build-system-bins/cookies/build.d/install' is up to date.
make[3]: '../../utils/kickstart-oe/cookies/build.d/install' is up to date.
make[3]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/bootloader'
	[configure] complete for bootloader.
	[build] complete for bootloader.
make install -C ../board-rk3566.x96_x6
make[3]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rk3566.x96_x6'
[===== NOW BUILDING:	bootfiles-rk3566.x96_x6-2022-01-03	=====]
	[fetch] complete for bootfiles-rk3566.x96_x6.
	[checksum] complete for bootfiles-rk3566.x96_x6.
	[extract] complete for bootfiles-rk3566.x96_x6.
	[patch] complete for bootfiles-rk3566.x96_x6.
	[configure] complete for bootfiles-rk3566.x96_x6.
	[build] complete for bootfiles-rk3566.x96_x6.
cp: cannot stat '/srv/WorkSpace/opensource/minimyth2/images/main/usr/lib/u-boot/rk3566/rk3566-x96_x6_defconfig/idblock.bin': No such file or directory
make[3]: *** [Makefile:35: install-custom] Error 1
make[3]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rk3566.x96_x6'
make[2]: *** [Makefile:92: install-custom] Error 2
make[2]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/bootloader'
make[1]: *** [../../gar.lib.mk:573: ../../bootloaders/bootloader/cookies/main.d/install] Error 2
make[1]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth'
make: *** [../../gar.lib.mk:592: imgdep-main] Error 2
andy@B150:/srv/WorkSpace/opensource/minimyth2/script/meta/minimyth$
```

warpme, 2022-04-15:
Well,

I see `cp: cannot stat '/srv/WorkSpace/opensource/minimyth2/images/main/usr/lib/u-boot/rk3566/rk3566-x96_x6_defconfig/idblock.bin': No such file or directory`
Lack of this file means you had in past successful install of this file by building/installing `u-boot-rk3566`  - but now this file can't be found....
Why this?

Anyway You can clean things by:
-go to `bootloaders/board-rk3566.x96_x6` and do `make clean clean-all`
-also go to `bootloaders/u-boot-rk3566` and also do `make clean clean-all`

andyshrk, 2022-04-16:
> Well,
>
> I see `cp: cannot stat '/srv/WorkSpace/opensource/minimyth2/images/main/usr/lib/u-boot/rk3566/rk3566-x96_x6_defconfig/idblock.bin': No such file or directory` Lack of this file means you had [...]

I do `make clean` at build-root after update . Not sure if it caused by this.
> 
> Anyway You can clean things by: -go to `bootloaders/board-rk3566.x96_x6` and do `make clean clean-all` -also go to `bootloaders/u-boot-rk3566` and also do `make clean clean-all`

After make clean clean-all , the u-boot compile success. But the flowing error comes:
```
make[5]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/devel/apitrace/work/main.d/apitrace-9.0_build'
make[5]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/devel/apitrace/work/main.d/apitrace-9.0_build'
make[5]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/devel/apitrace/work/main.d/apitrace-9.0_build'
make[5]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/devel/apitrace/work/main.d/apitrace-9.0_build'
[ 80%] Linking CXX shared library egltrace.so
[ 80%] Linking CXX shared library glxtrace.so
make[5]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/devel/apitrace/work/main.d/apitrace-9.0_build'
[ 93%] Built target glretrace_common
make[5]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/devel/apitrace/work/main.d/apitrace-9.0_build'
make[5]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/devel/apitrace/work/main.d/apitrace-9.0_build'
Consolidate compiler generated dependencies of target eglretrace
Consolidate compiler generated dependencies of target glretrace
make[5]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/devel/apitrace/work/main.d/apitrace-9.0_build'
make[5]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/devel/apitrace/work/main.d/apitrace-9.0_build'
[ 96%] Built target glretrace
[ 96%] Built target eglretrace
/srv/WorkSpace/opensource/minimyth2/script/devel/apitrace/work/main.d/apitrace-9.0_build/wrappers/glxtrace.cpp: In function ‘_wrapProcAddress’:
/srv/WorkSpace/opensource/minimyth2/script/devel/apitrace/work/main.d/apitrace-9.0_build/wrappers/glxtrace.cpp:98858:24: note: variable tracking size limit exceeded with ‘-fvar-tracking-assignments’, retrying without
98858 | static __GLXextFuncPtr _wrapProcAddress(const GLubyte * procName, __GLXextFuncPtr procPtr) {
      |                        ^
/srv/WorkSpace/opensource/minimyth2/images/build/usr/lib/gcc/aarch64-minimyth-linux-gnu/11.2.0/../../../../aarch64-minimyth-linux-gnu/bin/ld: /srv/WorkSpace/opensource/minimyth2/images/build/usr/lib/gcc/aarch64-minimyth-linux-gnu/11.2.0/../../../../aarch64-minimyth-linux-gnu/bin/ld: DWARF error: could not find abbrev number 557056
/tmp/ccXfoszh.ltrans29.ltrans.o: in function `dlsym':
<artificial>:(.text+0x4540): undefined reference to `__libc_dlopen_mode'
/srv/WorkSpace/opensource/minimyth2/images/build/usr/lib/gcc/aarch64-minimyth-linux-gnu/11.2.0/../../../../aarch64-minimyth-linux-gnu/bin/ld: <artificial>:(.text+0x4550): undefined reference to `__libc_dlsym'
/srv/WorkSpace/opensource/minimyth2/images/build/usr/lib/gcc/aarch64-minimyth-linux-gnu/11.2.0/../../../../aarch64-minimyth-linux-gnu/bin/ld: /tmp/ccXfoszh.ltrans29.ltrans.o: in function `dlopen':
<artificial>:(.text+0x4880): undefined reference to `__libc_dlopen_mode'
/srv/WorkSpace/opensource/minimyth2/images/build/usr/lib/gcc/aarch64-minimyth-linux-gnu/11.2.0/../../../../aarch64-minimyth-linux-gnu/bin/ld: <artificial>:(.text+0x4890): undefined reference to `__libc_dlsym'
/srv/WorkSpace/opensource/minimyth2/images/build/usr/lib/gcc/aarch64-minimyth-linux-gnu/11.2.0/../../../../aarch64-minimyth-linux-gnu/bin/ld: /tmp/ccXfoszh.ltrans29.ltrans.o: in function `_get_glXChooseVisual(_XDisplay*, int, int*) [clone .lto_priv.0]':
<artificial>:(.text+0x75c8): undefined reference to `__libc_dlopen_mode'

```

warpme, 2022-04-17:
_I do make clean at build-root after update_
This cleans all packages in root.
So effectively  you will have vanilla sources without any build package (exactly like at first initial build when you clone git before going to scrip/meta/minimyth and issuing make build)

closed 2022-04-22 by warpme

andyshrk, 2022-04-25:
> _I do make clean at build-root after update_ This cleans all packages in root. So effectively you will have vanilla sources without any build package (exactly like at first initial build when you clone git before going to scrip/meta/minimyth and issuing make build)

I can reproduce this issue：
* after a successful make `build、mak clean` at minimyth2/script/meta/minimyth
* cd {build-root} ; make clean.   I can see many many  **Entering directory xxx .... Leaving directory xxx** log, and nor err
*  after the `make clean` done， I run ` ls script/bootloaders/board-rk3566.x96_x6/`

I can see cookies work dir are still there.

```
make[3]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/perl/perl-XML-SAX'
make[2]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/perl'
make[1]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script'
andy@B150:/srv/WorkSpace/opensource/minimyth2$ 
andy@B150:/srv/WorkSpace/opensource/minimyth2$ ls script/bootloaders/board-rk3566.x96_x6/
checksums  cookies  download  files  Makefile  work
```

So i do a modify of the makefile:
```
andy@B150:/srv/WorkSpace/opensource/minimyth2$ git diff
diff --git a/script/bootloaders/board-rk3566.x96_x6/Makefile b/script/bootloaders/board-rk3566.x96_x6/Makefile
index c31e4635..554475f2 100644
--- a/script/bootloaders/board-rk3566.x96_x6/Makefile
+++ b/script/bootloaders/board-rk3566.x96_x6/Makefile
@@ -39,9 +39,9 @@ install-custom:
        @# do not do $(MAKECOOKIE) as reinstall-board uses this install also to reinstall board files
 
 clean-custom:
-       @rm -f  $(DESTDIR)/boot/rk3566_u-boot
-       @rm -rf $(DESTDIR)/boot/rk3566loaderfiles*
-       @rm -f  $(DESTDIR)/boot/minimyth.conf
-       @rm -rf $(DESTDIR)/boot/extlinux*
+       rm -f  $(DESTDIR)/boot/rk3566_u-boot
+       rm -rf $(DESTDIR)/boot/rk3566loaderfiles*
+       rm -f  $(DESTDIR)/boot/minimyth.conf
+       rm -rf $(DESTDIR)/boot/extlinux*
 
 clean-all: clean-custom cookieclean
diff --git a/script/gar.mk b/script/gar.mk
index 43298b3b..494ff8cb 100644
--- a/script/gar.mk
+++ b/script/gar.mk
@@ -541,8 +541,8 @@ downloadclean:
        @rm -rf $(DOWNLOADDIR)
 
 cookieclean: buildclean
-       @rm -rf $(COOKIEROOTDIR)
-       @rm -rf tmp-$(COOKIEROOTDIR)
+       rm -rf $(COOKIEROOTDIR)
+       rm -rf tmp-$(COOKIEROOTDIR)
 
```

Run `make clean` again:
```
make[3]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-h616.x96_mate'
make[3]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-h616.x96_mate'
make[3]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rk3328.beelink_a1'
make[3]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rk3328.beelink_a1'
make[3]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rpi4.rpi64'
make[3]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rpi4.rpi64'
make[3]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/u-boot-rpi'
rm -rf cookies
rm -rf tmp-cookies
make[3]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/u-boot-rpi'
make[3]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rpi3.mainline32'
make[3]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rpi3.mainline32'
make[3]: Entering directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rk3399.orangepi_4'
make[3]: Leaving directory '/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rk3399.orangepi_4'
[... 18 lines cut ...]
Go to the board-rk3566.x96_x6 dir:
```
andy@B150:/srv/WorkSpace/opensource/minimyth2$ cd script/bootloaders/board-rk3566.x96_x6/
andy@B150:/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rk3566.x96_x6$ ls
checksums  cookies  download  files  Makefile  work
andy@B150:/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rk3566.x96_x6$ make clean
rm -f  /srv/WorkSpace/opensource/minimyth2/images/main/boot/rk3566_u-boot
rm -rf /srv/WorkSpace/opensource/minimyth2/images/main/boot/rk3566loaderfiles*
rm -f  /srv/WorkSpace/opensource/minimyth2/images/main/boot/minimyth.conf
rm -rf /srv/WorkSpace/opensource/minimyth2/images/main/boot/extlinux*
andy@B150:/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rk3566.x96_x6$ ls
checksums  cookies  download  files  Makefile  work
andy@B150:/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rk3566.x96_x6$ make clean-all
rm -f  /srv/WorkSpace/opensource/minimyth2/images/main/boot/rk3566_u-boot
rm -rf /srv/WorkSpace/opensource/minimyth2/images/main/boot/rk3566loaderfiles*
rm -f  /srv/WorkSpace/opensource/minimyth2/images/main/boot/minimyth.conf
rm -rf /srv/WorkSpace/opensource/minimyth2/images/main/boot/extlinux*
rm -rf cookies
rm -rf tmp-cookies
andy@B150:/srv/WorkSpace/opensource/minimyth2/script/bootloaders/board-rk3566.x96_x6$ ls
checksums  download  files  Makefile
```

It can only be cleaned by `make clean-all`
Not sure if this is the case.