warpme/minimyth2 #31: Raspberry Pi 4 needs newer software

kjeacle, 2022-08-10:
Thought I'd give this a quick try on a Raspberry Pi 4 but got a message saying the board needs newer software.

I used this image:

MiniMyth2-armv8-master-12.3.2.r720-board-sm1.tanix_tx5_plus-board-rpi34.mainline64-SD-Image.img.xz

[image]

kjeacle, 2022-08-12:
Perhaps this was an SD card problem. Rewrote the image and it looks like it attempts to boot. The rainbow screen appears for a split-second but then the screen just goes black. If this works for you, feel free to close issue as I'm not confident this isn't a local problem here.

warpme, 2022-08-13:
Karl,
I'm on vacations so replays will be delayed.
I suspect Your issue is because in mm2 ver. I published on GitHub I'm using very latest rpi firmware and maybe rom initial program loader (first stage boot sw. residing in board rom) on your board is too old. 
Let me provide you mm2 image with older rpi firmware - as I already found new firmware problematic (https://github.com/warpme/minimyth2/commit/752aefb1034238db94d9b349f7628f8e341176ee).
stay tuned!

kjeacle, 2022-08-14:
Hi Piotr, I only want to try it out as possible future option. No immediate plans to use it. PXE+x86 working fine and I'm still on v31(!) so would use frontend schema override options to try watching videos etc. Enjoy your holidays. No hurry here. Thanks!

warpme, 2022-08-18:
Karl,
may you try https://github.com/warpme/minimyth2/releases/download/v12.5.0-v33-Pre-763-g6034adfa56/MiniMyth2-armv8-master-12.5.0.r762-board-sm1.tanix_tx5_plus-board-rpi34.mainline64-SD-Image.img.xz

I build it remotely. it has older rpi firmware.
In case of any problems - pls try telneting to board and provide me some logs...

kjeacle, 2022-08-19:
Still no luck with this image. I tried two different SD cards but got the same result: rainbow screen and then just a black screen - nothing happens. The card appears to be OK because it boots on a Raspberry Pi 3, however there didn't appear to be any themes present on the card? So I had to add these lines to /media/boot/minimyth.conf to get mythfrontend to start:

MM_THEME_NAME='mythbuntu'
MM_THEME_URL='http://w.x.y.z/path/to/mythbuntu.sfs'

After that, it worked on the RPi3 but I didn't do any testing as it's the RPi4 that I really want to try.

warpme, 2022-08-22:
Karl,
1. I think boot issue might be related to: a\combination of your board rom and rpi firmware or b\yours power supply.
Ideally will be to find what rpi firmware books ok with your rpi4 bard.
I think we may try following:
-try v12.6.1. It it will still not boot, then:
-try to find booting ok OS i.e. raspbian
-try to use this OS firmware with mm2 by copying rpi firmware files from OS to /boot part on mm2 sd card.
(IIRC fixup4cd.dat  fixup4.dat   fixup4x.dat   start4cd.elf   start4.elf  start4x.elf bootcode.bin should be enough.)

2. Regarding no themes on SD card:
It was bug in code i overlooked due no testing possibilities. 
May you try this: 
https://github.com/warpme/minimyth2/releases/download/v12.6.1-v33-Pre-763-g6034adfa56/MiniMyth2-armv8-master-12.6.1.r762-board-sm1.tanix_tx5_plus-board-rpi34.mainline64-SD-Image.img.xz
Also pls try this .img without any changes in minimyth.conf. I want to see is it capable to zero-touch-provisioning...

kjeacle, 2022-08-22:
I have two official Raspberry Pi 4 power supplies so power problem is possible but seems unlikely. I have also tried two different SD cards.

12.6.1 has the same problem: rainbow screen then nothing.

12.6.1 with boot files (that you mentioned) copied from working RPi 4 SD card also has same problem.

12.6.1 with modified boot files works in RPi 3 out of box apart from setting MM_MASTER_SERVER.

warpme, 2022-08-22:
Karl,
Im really surprised by this.
Only thing coming to my mind is thing you mentioned in Your very first post: _"Raspberry Pi 4 but got a message saying the board needs newer software"_

I asked RPI guys and they told me "_first, do what bard tells you to do: upgrade sw...._"

kjeacle, 2022-08-22:
I've updated the bootloader to the most recent beta that I could find: pieeprom-2022-08-02.bin. Unfortunately there is no difference. Rainbow screen and then nothing.

Other images (e.g. FreeBSD, DietPi) boot without problems so obviously I'm suspicious that there's something not right with the MM2 image, but perhaps there is something else weird going on with my setup.

I will keep an eye out for future images and update here if I have success.

warpme, 2022-08-24:
Karl,
If you still have energy/motivation to play with mm2 on rpi4....
I prepared .img which boots in chain-load mode (rpi4 spl chainloaded to mainline u-boot):
https://github.com/warpme/minimyth2/releases/download/v12.6.1-v33-Pre-763-g6034adfa56/MiniMyth2-armv8-master-12.6.1.r762-board-rpi4.mainline64-SD-Image.img.xz
it boots ok on mine rpi4....

kjeacle, 2022-08-24:
After the rainbow screen, it gets as far as "Starting kernel ...". Then the HDMI switches off and the green activity LED repeatedly flashes twice with a short pause between each double flash. I don't see any DHCP request coming in on the server.

warpme, 2022-08-24:
Geez,
This RPI nightmare never ends.....
Here are rpi error led codes:  

```
Long flashes | Short flashes | Status
-- | -- | --
0 | 3 | Generic failure to boot
0 | 4 | start*.elf not found
0 | 7 | Kernel image not found
0 | 8 | SDRAM failure
0 | 9 | Insufficient SDRAM
0 | 10 | In HALT state
2 | 1 | Partition not FAT
2 | 2 | Failed to read from partition
2 | 3 | Extended partition not FAT
2 | 4 | File signature/hash mismatch - Pi 4
3 | 1 | SPI EEPROM error - Pi 4
3 | 2 | SPI EEPROM is write protected - Pi 4
4 | 4 | Unsupported board type
4 | 5 | Fatal firmware error
4 | 6 | Power failure type A
4 | 7 | Power failure type B
```

is there any match with your's case?

warpme, 2022-08-24:
Karl,
Just 2 things:
1. you may enable verbose booting of kernel - so we may see when kernel hangs at boot. To do so: put https://github.com/warpme/minimyth2/blob/2f620bcc80f039f742c89d8ff4ceacff25b711b0/script/bootloaders/board-rpi4.mainline64/files/boot-debug.scr as boot.scr on SD card /BOOT part and try to boot

2. I testing mm2 on mine rpi4 which is: Machine model: Raspberry Pi 4 Model B 
Do you have the same model? Maybe this will explain why mm2 boots ok on my rpi4 and fails on rpi4 (as there is something wrong in mm2 for yours rpi4 variant maybe?) 
 
btw: try booting few times. on mine rpi4 with mainline u-boot - boot sometimes hangs with symptoms like you described in https://github.com/warpme/minimyth2/issues/31#issuecomment-1225525930

kjeacle, 2022-08-24:
1. Tried boot-debug.scr. Text flies off the screen and the screen goes blank. Took a video and paused it. Last line before screen goes blank is "fb0: switching to vc4 from simple". Tried two different monitors but same thing happens.

2. It's a 2GB RPi 4 Model B as shown in photo in first comment. Bought it recently so presumably a new revision if there are multiple revisions.

warpme, 2022-08-24:
> 1. Tried boot-debug.scr. Text flies off the screen and the screen goes blank. Took a video and paused it. Last line before screen goes blank is "fb0: switching to vc4 from simple". Tried two different monitors but same thing happens.

And Eth is still dead?

kjeacle, 2022-08-24:
Yes, nothing in DHCP server log. Both ethernet LEDs are solid and the board activity LED does two short flashes, pause, too short flashes, pause... repeating forever.

warpme, 2022-08-25:
Karl,

I found this: https://patchwork.kernel.org/project/linux-arm-kernel/cover/20220815-rpi-fix-4k-60-v1-0-c52bd642f7c6@cerno.tech/
while this is titled about 4k - patch seems to be related to vc4 clock area (i believe our issue is related to 5.19 vc4 code changes).
I produced .img with this patch + most recent rpi4 firmware in 2 versions: rpi4 proprietary bootloader and mainline uboot.
May you try?
(in case problems try boot with boot-debug.scr on mainline uboot )
https://github.com/warpme/minimyth2/releases/tag/v12.6.1-v33-Pre-763-g6034adfa56

btw: this whole mess is because - contrary to dietpi/raspbian which uses rpi downstream kernel - mm2 - is single universal rootfs image prepared for all supported boards and based on mainline kernel. This design decision allows me to offer multi-board capability - so theoretically - mm2 is capable to be single SD card image for all SoC. 
Price for decision this is i.e. rpi mess (btw: rpi is known VERY upstream reluctant + strong drive to use downstream kernel which is mainline with 600+ downstream patches. for me this is hell tio support...)

kjeacle, 2022-08-25:
Tried but exactly the same as previous release. Stops at the "switching to vc4" line. I am happy to keep testing but it's very frustrating for you to debug remotely like this. Feel free to park this and wait for a mainline change that might address this?

Thanks for the background information - that's very interesting. FWIW, from an end-user perspective the range of boards that are supported on the releases page is slightly overwhelming. For me, it would be enough to have a wiki page that recommended a very small number of well supported devices that are available for purchase online e.g. small x86 box, small android box, a couple of SoC boards.

warpme, 2022-09-13:
Karl,
I can't reproduce this ticket original issue (nor 2 other users I asked).
I propose close this ticket and return to testing on your's rpi4 when i will release mm2 with 6.0 kernel.
Thx!

closed 2022-09-13 by warpme