warpme/minimyth2 #35: MM_FONT_FILE_TTF_ADD doesn't work

kjeacle, 2022-08-26:
In v12.6.2-v32 MythCenter-wide has gone, so I have set MM_THEME_NAME and MM_THEME_URL to download it on boot.

But there are fonts missing. I tried MM_FONT_FILE_TTF_ADD='DroidSans.ttf' but nothing happened. Documentation says that this will download font from server. Looking at init.d/font, it doesn't appear to do any downloading but maybe that happens elsewhere.

Also, init.d/font has a line that makes a link:

``
/bin/ln -sf /usr/share/fonts/X11/TTF/${font} /usr/share/mythtv/${font}
``

But surely this should be:

``
/bin/ln -sf /usr/share/fonts/X11/TTF/${font} /usr/share/mythtv/fonts/${font}
``

warpme, 2022-08-27:
Oh this is bug.
Should be flxed in v12.6.3...

kjeacle, 2022-08-27:
Working again, thanks. I added the following to get MythCenter-wide looking the same as before: MM_FONT_FILE_TTF_ADD='DroidSans.ttf DroidSansMono.ttf DroidSans-Bold.ttf'

closed 2022-08-27 by kjeacle