warpme/minimyth2 #37: rockpi and raspberrypi.

rottnest, 2022-10-29:
Hi
I've got a year old and a week old rockpi 4A (2G on the old, 4G on the new)
Both display no action at boot.

I made a card for a 4G raspberry pi 4.
It boots, I see 'scripts' in the messages then the display goes blank. It is a 1080 display that does work from 768x1024. The blue led flashes in regular pairs of flashes.

I downloaded src.tgz and will try to build

I've said more on the mythtv-users list

warpme, 2022-10-29:
James,
I see you are using RockPi 4A.
Image I’m building is for RockPi 4B and for sure it will not work on Your board without change device tree(*) file
Now DT is for RockPi 4B while Your board requires DT for RockPi 4A.
So pls copy+rename `boot/dtbs/rockchip/rk3399-rock-pi-4a.dtb` file to `/boot` partition named as `rk3399_dtb`

warpme, 2022-10-29:
Regarding RPI4:
May you pls provide me MiniMyth2 Diagnostic Info content?
It is in `/var/log/system-info`
Alternatively in browser _TopMenu/Logs/MiniMyth2 Diagnostic Info_

rottnest, 2022-10-30:
> James,
> I see you are using RockPi 4A.
> Image I’m building is for RockPi 4B and for sure it will not work on Your board without change device tree(*) file
> Now DT is for RockPi 4B while Your board [...]

Piotr I’m not trying to give you a hard time - promise :-)
I’ve played with lots of things, just not ARM so I’m pretty much completely naive !

Because ra-pi gives no display, I mounted the SD card elsewhere to get the logs you asked for and while playing around after 2nd reboot MiniMyth is playing !

Now to play with the rockpi. There is NO board support package on your release, I got one elsewhere and will play with it.
Cheers
James

warpme, 2022-10-30:
> Wiadomość napisana przez rottnest ***@***.***> w dniu 30.10.2022, o godz. 10:41:
> 
> 
> 
> 
> Because ra-pi gives no display, I mounted the SD card elsewhere to get the logs you asked for and while playing around after 2nd reboot MiniMyth is playing !
> 
> Now to play with the rockpi. There is NO board support package on your release, I got one elsewhere and will play with it.
> Cheers
> James

James,

let me explain who things are going here:

-my distro (minimyth2) has rk3399 u-boot (bootloader) built with rock-pi-4-rk3399_defconfig.
IIRC this config should cover all rockpi 4 boards.

-minimyth2 boot script (boot.scr) is prepared for rockpi-4b (uses dtbs/rockchip/rk3399-rock-pi-4b.dtb)

-for cases like yours (board is different than declared in boot.scr) - I developed solution where user can add custom DT. This procedure is programmed in boot.scr and is based on logic like this:

1.look for rk3399_dtb is /boot
2.if present - use it
3.if not present - use DT declared in boot.scr

So by putting rk3399_dtb into /boot - you will ask minimyth2 to use right DT for your bard
(rk3399-rock-pi-4a.dtb)

pls do above and let me know how it goes

rottnest, 2022-10-30:
> On 30 Oct 2022, at 5:55 pm, Piotr Oniszczuk ***@***.***> wrote:
> 
> 
> 
> 
> > Wiadomość napisana przez rottnest ***@***.***> w dniu 30.10.2022, o godz. 10:41:
> > 
> > 
> > 
> > 
> > Because ra-pi gives no display, I mounted the SD card elsewhere to get the logs you asked for and while playing around after 2nd reboot MiniMyth is playing !
> > 
> > Now to play with the rockpi. There is NO board support package on your release, I got one elsewhere and will play with it.
> > Cheers
> > James
> 
> James,
> 
> let me explain who things are going here:
> 
> -my distro (minimyth2) has rk3399 u-boot (bootloader) built with rock-pi-4-rk3399_defconfig.
> IIRC this config should cover all rockpi 4 boards.
> 
> -minimyth2 boot script (boot.scr) is prepared for rockpi-4b (uses dtbs/rockchip/rk3399-rock-pi-4b.dtb)
> 
> -for cases like yours (board is different than declared in boot.scr) - I developed solution where user can add custom DT. This procedure is programmed in boot.scr and is based on logic like this:
> 
> 1.look for rk3399_dtb is /boot
> 2.if present - use it
> 3.if not present - use DT declared in boot.scr
> 
> So by putting rk3399_dtb into /boot - you will ask minimyth2 to use right DT for your bard
> (rk3399-rock-pi-4a.dtb)
> 
> pls do above and let me know how it goes 

Thanks, better understanding.
My error was assuming 4A is exactly like 4B without wifi n bluetooth.
I made a card with rk3399.dtb from debian-buster, went for supper, and there it was - GONE. If hunting does not find then I'll craft another in the morning.
How much ram do you need?

James

warpme, 2022-10-30:
> 
> Thanks, better understanding.
> My error was assuming 4A is exactly like 4B without wifi n bluetooth.
> I made a card with rk3399.dtb from debian-buster, went for supper, and there it was - GONE. If hunting does not find then I'll craft another in the morning.

DTB is kernel version specific!
Pls use right dtb file from my SD card.

So:
1. flash SD card with image from my github repo

2. mount r/w SD card boot partition

3. copy file from this SD card: boot/dtbs/rockchip/rk3399-rock-pi-4a.dtb file to /boot partition root dir as rk3399_dtb
(exact such name. no rk3399.dtb. it must be „rk3399_dtb”)

4. unmout SD card

5. insert in bard. it should boot.

rottnest, 2022-10-31:
> On 30 Oct 2022, at 9:13 pm, Piotr Oniszczuk ***@***.***> wrote:
> 
> 
> 
> > 
> > Thanks, better understanding.
> > My error was assuming 4A is exactly like 4B without wifi n bluetooth.
> > I made a card with rk3399.dtb from debian-buster, went for supper, and there it was - GONE. If hunting does not find then I'll craft another in the morning.
> 
> DTB is kernel version specific!
> Pls use right dtb file from my SD card.
> 
> So:
> 1. flash SD card with image from my github repo
> 
> 2. mount r/w SD card boot partition
> 
> 3. copy file from this SD card: boot/dtbs/rockchip/rk3399-rock-pi-4a.dtb file to /boot partition root dir as rk3399_dtb
> (exact such name. no rk3399.dtb. it must be „rk3399_dtb”)
> 
> 4. unmout SD card
> 
> 5. insert in bard. it should boot.
> 

Hi Piotr

I downloaded the image this morning. Mounted. 
# cp /mnt/rockchip/rk3399-rock-pi-4a.dtb rk3399_dtb
# sync

Plugged in, power-on, nothing:

green led
no blue activity

I’m going to try build from src. (native, not cross) what I am trying to see is if the display is smoother.
My card I’m using is 2G, my 4G card has a m2 card. I’ve broken the m2 cable 3 times over the years so I’d rather not undo the m2 to mount an SD. 

FYI I’ll keep you informend
Cheers
James

PS Jim Abernathy is using frame buffer on EGLS. That seems somwhat hard.

warpme, 2022-10-31:
James,
May you point me to exact URL with linux distro booting ok on your board?

rottnest, 2022-10-31:
> On 31 Oct 2022, at 3:47 pm, Piotr Oniszczuk ***@***.***> wrote:
> 
> 
> James,
> May you point me to exact URL with linux distro booting ok on your board?
> 

Hi

this boots and runs on ra-pi 4

MiniMyth2-armv8-32-12.6.2.r60-board-sm1.tanix_tx5_plus-board-rpi34.mainline64-SD-Image.img.xz

this does not boot

MiniMyth2-armv8-32-12.6.2.r60-board-rk3399.rockpi4-b-SD-Image.img.xz

I copied the dtbs/rockchip/rk3399-rock-pi-4a.dtb to rk3399_dtb

On power on: green led. The End. No blue led activity. No network presense.

The url is your git hub ie
https://github.com/warpme/minimyth2/releases/download/v12.6.2-v32.0-60-ge5c974e402/MiniMyth2-armv8-32-12.6.2.r60-board-rk3399.rockpi4-b-SD-Image.img.xz

Cheers
James

warpme, 2022-10-31:
_linux distro booting ok on your board_ - i was meant booting on rockpi4a - not on rpi4....

rottnest, 2022-10-31:
> On 31 Oct 2022, at 4:43 pm, Piotr Oniszczuk ***@***.***> wrote:
> 
> 
> linux distro booting ok on your board - i was meant booting on rockpi4a - not on rpi4….

I have no booting rockpi4 images :-(
James

rottnest, 2022-10-31:
> On 31 Oct 2022, at 5:22 pm, James Linder ***@***.***> wrote:
> 
> 
> 
>> On 31 Oct 2022, at 4:43 pm, Piotr Oniszczuk ***@***.***> wrote:
>> 
>> 
>> linux distro booting ok on your board - i was meant booting on rockpi4a - not on rpi4….
> 
> I have no booting rockpi4 images :-(

Piotr
I do not know how to get omax support. I built 22 from src and the display performance is aweful
The only reason I was looking at rockpi was to compare display.
Do you have a rockpi, if so what is performance like?

The video he shows seems quite nice, but it's hard to tell
https://www.explainingcomputers.com/rockpi4.html <https://www.explainingcomputers.com/rockpi4.html>

James

rottnest, 2022-10-31:
> On 31 Oct 2022, at 5:22 pm, James Linder ***@***.*** ***@***.***>> wrote:
> 
> 
> 
>> On 31 Oct 2022, at 4:43 pm, Piotr Oniszczuk ***@***.*** ***@***.***>> wrote:
>> 
>> 
>> linux distro booting ok on your board - i was meant booting on rockpi4a - not on rpi4….
> 
> I have no booting rockpi4 images :-(

Piotr
I do not know how to get omax support. I built 22 from src and the display performance is aweful
The only reason I was looking at rockpi was to compare display.
Do you have a rockpi, if so what is performance like?

The video he shows seems quite nice, but it's hard to tell
https://www.explainingcomputers.com/rockpi4.html <https://www.explainingcomputers.com/rockpi4.html>

sorry 1st video 17:45

James

warpme, 2022-10-31:
James,

I switch rockpi4 from mainline u-boot to radxa (producer of your's board).
On mine rockpi-4b works perfect.
Here is mage: https://github.com/warpme/minimyth2/releases/download/v12.9.0-v33-922-g61b82f2b5a/MiniMyth2-armv8-master-12.9.0.r933-board-rk3399.rockpi4-b-SD-Image.img.xz
It should work on yours rockpi-4a.
If not - try change device-tree like I advised here: https://github.com/warpme/minimyth2/issues/37#issuecomment-1296716529

re performance of rockpi4:
mpeg2/h264/vp8/vp9/hevc - all are hardware accelerated.
i have silky smooth playback - even of 4k@60fps
playback of 4k@60fps gets 3..5% of cpu on mm2
on rk3399 - myth just flying

rottnest, 2022-11-01:
> On 1 Nov 2022, at 5:07 am, Piotr Oniszczuk ***@***.***> wrote:
> 
> 
> James,
> 
> I switch rockpi4 from mainline u-boot to radxa (producer of your's board).
> On mine rockpi-4b works perfect.
> Here is mage: https://github.com/warpme/minimyth2/releases/download/v12.9.0-v33-922-g61b82f2b5a/MiniMyth2-armv8-master-12.9.0.r933-board-rk3399.rockpi4-b-SD-Image.img.xz
> It should work on yours rockpi-4a.
> If not - try change device-tree like I advised here: #37 (comment)
> 
> re performance of rockpi4:
> mpeg2/h264/vp8/vp9/hevc - all are hardware accelerated.
> i have silky smooth playback - even of ***@***.***
> playback of ***@***.*** gets 3..5% of cpu on mm2
> on rk3399 - myth just flying

Yup that boots - thanks
My main BE is 32, so I’m just building a master be to test with
Cheers
James

warpme, 2022-11-01:
Oh - great to hear that!
Pls feel free to express any comments, suggestions, bugs, etc

rottnest, 2022-11-01:
> On 1 Nov 2022, at 3:56 pm, Piotr Oniszczuk ***@***.***> wrote:
> 
> 
> Oh - great to hear that!
> Pls feel free to express any comments, suggestions, bugs, etc

Poitr Hi

I am using 32 but built a master backend to test. The ‘master’ image I downloaded looks at, then plays my 32 backend !!!
That suits me ideally, but I guess it is not what you want.

Side-by-side my evalution is

raspberrry pi 5/10
rockpi 7/10
intel + i3 8/10
mac 4xi5 8/10

I guess the limiting bit is OTA recording. They are pretty crappy doing 2-4 Mb/s (DVB) at 25 fps.
but all in all, good and thanks lots for your help.

I’ve got the jellyfish egs. Just looked at those.
Wow at 10 Mb/s I’d give the the rockpit 10/10. absolutly silkysmooth.

I should do the a/b again with good quality recordings OTH gonna be watch crappy OTA stuff
James

warpme, 2022-11-02:
James,

just 3 things before i will close this ticket:
1. for sure you are using v4l2 video decoding profile right?
2. you can periodically update mm2 to current most actual version over Internet. to do this you can:

option1: use option in GUI
`Setup/System Maintenance/System Updates/Check for updates` (if you are using Default menu scheme)
`Advanced/Maintenance/System Updates/Check for updates` (if you are using Media Centre menu scheme)

option2: log to device via telnet
run `mm_do_online_update` to check are there any updates
run `mm_do_online_update doupdate` to update then
```
sync
reboot
```
3. if you found mm2 nice to use - pls put word about this in mythtv-users mailing list in topic where all this was initiated (Raspbery PI topic iirc)