warpme/minimyth2 #38: quartz-64-a

Shaggy013, 2022-10-30:
no boot 

no leds
no hdmi
console stops after kernel gets started 

Starting kernel ...

I/TC: Secondary CPU 1 initializing
I/TC: Secondary CPU 1 switching to normal world boot
I/TC: Secondary CPU 2 initializing
I/TC: Secondary CPU 2 switching to normal world boot
I/TC: Secondary CPU 3 initializing
I/TC: Secondary CPU 3 switching to normal world boot

warpme, 2022-10-30:
Is this vanilla image downloaded from my git?
If so then image is for quartz64-b
Pls change DT in SD card boot part `/extlinux/extlinux.conf` to correct one (rk3566-quartz64-a.dtb)
Also you may try add `debug` in kernel command line

Shaggy013, 2022-10-30:
yeah i saw i just the B :(

well it does blink now :)

only backlight switches on 

and serial console nothing more 

is console enabled in kernel ?

will try on a 4k tv if got some output

Shaggy013, 2022-10-30:
well i got some output on a 4k tv only dhcp dont likes me today 

ow forgot debug

it is the same behaviour as what the other distros also  had no hdmi on all my tv's.
only the latest armbian and libreelec images works on all my tv's  

ow and rk3566 runs nice on 1560 memory and the 1992 cpu settings just as rk3568 ;)

warpme, 2022-10-30:
"only the latest armbian and libreelec images works on all my tv's" - saying libreelec (LE) - do you mean LE on rk356x?
afaik LE ont supports yet rk356x?
Anyway - indeed Alex Bee (he is doing LE rk patches) is providing extra kernel code patches related to rk hdmi.
Unfortunately this patch - while works for Kodi - breaks i.e. drm atomic commits in Qt. 
Thats why mm2 uses only part of Alex code (this not breaking Qt) - and it looks i was forced remove patch code giving you working hdmi in LE :-(

warpme, 2022-10-30:
is it possible to find exact armbian hdmi patch giving you working hdmi on your's tv's?

Shaggy013, 2022-10-30:
balbes150 has a rk356x image of LE wich works only the source ?  , the armbian hdmi workd the last few releases i tried  

that i trying to find last few days :)
got a nice gentoo ebuild now with warpme armbian manjaro le patch selection for the kernel

Shaggy013, 2022-10-30:
only i can think of is something with edid reading or clocks and hdmi speeds

closed 2022-10-30 by warpme

warpme, 2022-11-01:
FYI:

I played a bit with rk drm code and produced: https://github.com/warpme/minimyth2/releases/download/v12.9.0-v33-922-g61b82f2b5a/MiniMyth2-armv8-master-12.9.1.r933-board-rk3566.quartz64-SD-Image.img.xz

You mention:

> it is the same behaviour as what the other distros also had no hdmi on all my tv's.

curious how it will go on your's 4k tv?

btw:
it you want to have mythtv gui on rk356x in drm planes mode  - remember to set in minimyth.conf

`MM_MYTHTV_DRM_VIDEO="yes"`

and add:

`MM_MYTHTV_SET_ENV_VAR="QT_QPA_EGLFS_KMS_PLANE_INDEX=0"`

this is because vop2 reports drm planes formats in unusual way and mythtv automatic planes election selects wrong plane for gui :-(
Unfortunately OSD is still not visible on rk356x on drm planes mode :-\

Shaggy013, 2022-11-01:
i did found little of the diffrence between yours ,armbian and le hdmi patch .
the armbian has some more pll hdmi , like some  more clocks and resolution ,

and your image did work in the end on my 4k tv only not on the other screen where i have also a serial console , it did hang on getting a ip adress .

[armbian.txt](https://github.com/warpme/minimyth2/files/9913416/armbian.txt)
[warpme.txt](https://github.com/warpme/minimyth2/files/9913419/warpme.txt)

i will try it later at evening

Shaggy013, 2022-11-02:
quartz64-a

hangs on dhclient

[    9.346759] Bluetooth: HCI UART protocol Marvell registered
[    9.444309] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/9 wl0: Feb 11 2020 11:54:51 version 7.45.96.61 (be7af2d@shgit) (r745790) FWID 01-a41d86bd es7.c5.n4.a3
[   11.789943] Bluetooth: hci0: command 0x0c03 tx timeout
[   19.683273] Bluetooth: hci0: BCM: Reset failed (-110)
[   21.816605] Bluetooth: hci0: command 0x0c03 tx timeout
[   29.709940] Bluetooth: hci0: BCM: Reset failed (-110)
[   31.304461] loop: module loaded
[   33.763273] Bluetooth: hci0: command 0x0c03 tx timeout
[   41.869940] Bluetooth: hci0: BCM: Reset failed (-110)
[   42.651457] rk_gmac-dwmac fe010000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   42.653150] rk_gmac-dwmac fe010000.ethernet eth0: PHY [stmmac-0:00] driver [Generic PHY] (irq=POLL)
[   42.663253] dwmac4: Master AXI performs any burst length
[   42.663741] rk_gmac-dwmac fe010000.ethernet eth0: No Safety Features support found
[   42.664425] rk_gmac-dwmac fe010000.ethernet eth0: IEEE 1588-2008 Advanced[   42.668307] rk_gmac-dwmac fe010000.ethernet eth0: configuring for phy/rgmii link mode
[   42.672273] rk_gmac-dwmac fe010000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx

warpme, 2022-11-02:
Argh - unfortunately I don't have model A to test.
6.1 kernel DT in mm2 seems to be almost vanilla 6.1 mainline DT....
Phy seems to be working. 
Also speed is already negotiated to 1Gbps.
1. i assume LEDs on Eth port are lightning ok (activity and speed)?
2. Also LEDs on other side (etc switch or...) are OK?
3. Have you any other distro with mainline 6.1 working ok on your's model A?
4. If yes - then can you use DT from this distro in mm2?
   
Also - maybe issue is in bootloader?   
Is it possible for you to copy (via dd) bootloader from working distro SD card to mm2 SD card and see?

btw: do you know: is pIne64 providing vendor uboot sources for Quartz64 boards?
Im using Peter Geis uboot which is quite old....

Shaggy013, 2022-11-04:
i check at evening but think motocom

i assume LEDs on Eth port are lightning ok (activity and speed)?Also LEDs on other side (etc switch or...) are OK?Have you any other distro with mainline 6.1 working ok on your's model A?If yes - then can you use DT from this distro in mm2?

Shaggy013, 2022-11-04:
yeah there are 2 u-boots in circulation 

the old that works

and the newer wich  has issues 

on the quartz chat there are lot of issues with new u-boot 
and counterpillow , pg , neggles are discussing it

Shaggy013, 2022-11-04:
i had your patches in a gentoo kernel and then it works so it is just a setting for network.

warpme, 2022-11-04:
fyi: i'm developing support for radxa rock3-c (another rk3566 sbc) and radxa bootloader boots nicely on quartz64-b.  
next week i'll receive board - so i can start to develop dt for it....