warpme/minimyth2 #40: X96 tv box is not booting

sickosarm, 2022-11-11:
Thanks for your good work. 

Last version for this device must have an incorrect u-boot because sd card is not booting  but android from emmc does.

warpme, 2022-11-11:
Hmm...

I assume you are reporting issue about x96-x6 tv box.
  
Saying 

> Last version for this device must have an incorrect u-boot 

Do you mean:
-version of tv box or
-version of mm2 SD image?

If you mean tv box:
-is there any older ver. of mm2 working ok for you on this exact x96-x6 box
-if not - may you pls provide exact pictures of pcb?  

btw: I flashed on my x96-x6 tvbox and all works as expected. u-boot on my box gives on uart log like this:

```
DDR Version V1.09 20210630
ln
ddrconfig:7
LPDDR4X, 324MHz
BW=32 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
tdqss: cs0 dqs0: -72ps, dqs1: -144ps, dqs2: -96ps, dqs3: -217ps, 
tdqss: cs1 dqs0: -72ps, dqs1: -144ps, dqs2: -96ps, dqs3: -217ps, 

change to: 324MHz
PHY drv:clk:36,ca:36,DQ:29,odt:60
vrefinner:16%, vrefout:41%
dram drv:40,odt:0
clk skew:0x65

change to: 528MHz
PHY drv:clk:36,ca:36,DQ:29,odt:60
vrefinner:16%, vrefout:41%
dram drv:40,odt:0
clk skew:0x58

change to: 780MHz
PHY drv:clk:36,ca:36,DQ:29,odt:60
vrefinner:16%, vrefout:41%
dram drv:40,odt:0
clk skew:0x58

change to: 1056MHz(final freq)
PHY drv:clk:36,ca:36,DQ:29,odt:60
vrefinner:16%, vrefout:22%
dram drv:40,odt:80
vref_ca:00000071
clk skew:0x46
cs 0:
the read training result:
DQS0:0x3e, DQS1:0x3c, DQS2:0x45, DQS3:0x3b, 
min  : 0xf 0x10 0x11  0xf  0x1  0x2  0x6  0x2 , 0x9  0x8  0x3  0x1  0x9  0x7  0x9  0x5 ,
      0x14 0x13  0xf  0xd  0x5  0x2  0x4  0x5 , 0xa  0x7  0x6  0x1  0xa  0x8  0x7  0x8 ,
mid  :0x2d 0x2e 0x2f 0x2c 0x20 0x20 0x24 0x20 ,0x27 0x27 0x20 0x20 0x27 0x26 0x26 0x23 ,
      0x33 0x33 0x2e 0x2c 0x25 0x22 0x24 0x24 ,0x2a 0x27 0x25 0x21 0x2a 0x28 0x26 0x28 ,
max  :0x4c 0x4d 0x4d 0x49 0x40 0x3f 0x43 0x3f ,0x45 0x46 0x3e 0x3f 0x46 0x45 0x44 0x42 ,
      0x52 0x53 0x4d 0x4c 0x46 0x42 0x44 0x44 ,0x4a 0x48 0x45 0x41 0x4a 0x49 0x46 0x48 ,
range:0x3d 0x3d 0x3c 0x3a 0x3f 0x3d 0x3d 0x3d ,0x3c 0x3e 0x3b 0x3e 0x3d 0x3e 0x3b 0x3d ,
      0x3e 0x40 0x3e 0x3f 0x41 0x40 0x40 0x3f ,0x40 0x41 0x3f 0x40 0x40 0x41 0x3f 0x40 ,
the write training result:
DQS0:0x3d, DQS1:0x33, DQS2:0x3a, DQS3:0x29, 
min  :0x5d 0x62 0x62 0x62 0x54 0x54 0x58 0x57 0x57 ,0x51 0x50 0x4c 0x4c 0x53 0x51 0x52 0x53 0x4e ,
      0x5a 0x58 0x55 0x56 0x4e 0x4c 0x4c 0x50 0x50 ,0x4a 0x48 0x45 0x42 0x4b 0x49 0x48 0x4c 0x46 ,
mid  :0x79 0x7d 0x7d 0x7c 0x6f 0x70 0x73 0x72 0x72 ,0x6c 0x6c 0x66 0x67 0x6e 0x6a 0x6b 0x6b 0x68 ,
      0x76 0x75 0x70 0x71 0x69 0x63 0x67 0x6a 0x6b ,0x66 0x62 0x5e 0x5d 0x66 0x65 0x60 0x66 0x5e ,
max  :0x96 0x98 0x99 0x97 0x8b 0x8c 0x8f 0x8d 0x8d ,0x87 0x88 0x80 0x82 0x8a 0x83 0x85 0x84 0x83 ,
      0x92 0x92 0x8b 0x8d 0x84 0x7b 0x82 0x85 0x87 ,0x82 0x7d 0x78 0x78 0x82 0x82 0x79 0x81 0x77 ,
[... 130 lines cut ...]
mmc1 is current device
Scanning mmc 1:4...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
reading /extlinux/extlinux.conf
401 bytes read in 4 ms (97.7 KiB/s)
1:.MiniMyth2
Retrieving file: /Image
reading /Image
31187456 bytes read in 2582 ms (11.5 MiB/s)
append: root=/dev/mmcblk0p5 rw rootwait earlycon=uart8250,mmio32,0xfe660000 console=ttyS2,1500000n8 logo.nologo vt.cur_default=1
Retrieving file: /dtbs/rockchip/rk3566-x96_x6.dtb
reading /dtbs/rockchip/rk3566-x96_x6.dtb
51251 bytes read in 12 ms (4.1 MiB/s)
Fdt Ramdisk skip relocation
## Flattened Device Tree blob at 0x0a100000
   Booting using the fdt blob at 0x0a100000
   Using Device Tree in place at 000000000a100000, end 000000000a10f832
Adding bank: 0x00200000 - 0x08400000 (size: 0x08200000)
Adding bank: 0x09400000 - 0xf0000000 (size: 0xe6c00000)
Adding bank: 0x1f0000000 - 0x200000000 (size: 0x10000000)
Total: 3484.969 ms

Starting kernel ...
```

sickosarm, 2022-11-11:
I am sorry for my bad explanation.
The box is an x96 x6 that works with previous releases. Your release 12.9.0 is not booting and just want you to know this. I have checked your log from u-boot and is strange that is loading x96_x6 dtb instead of the quartz64 dtb that is present in the extlinux.conf.
Anyway, excellent work and thanks very much again for supporting an rk3566 TV box, you are the only one I know that makes such thing.

warpme, 2022-11-12:
Here i'm a bit lost...
Here is extlinux.conf you should have in your SD card:
https://github.com/warpme/minimyth2/blob/99931f2fe0f4255b9303e92c02aeafc4ad41ad3d/script/bootloaders/board-rk3566.x96_x6/files/extlinux.conf#L3

u-boot for this box is build with x96_x6 defconfig:
https://github.com/warpme/minimyth2/blob/99931f2fe0f4255b9303e92c02aeafc4ad41ad3d/script/bootloaders/board-rk3566.x96_x6/Makefile#L10

it looks like uboot is correctly build as log says:

```
U-Boot 2017.09 (Sep 14 2022 - 11:52:24 +0200)

Model: X96 X6 RK3566 TV Box
```

also it looks like kernel gets correct DT as log says:

```
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
reading /extlinux/extlinux.conf
401 bytes read in 4 ms (97.7 KiB/s)
1:.MiniMyth2
Retrieving file: /Image
reading /Image
31187456 bytes read in 2582 ms (11.5 MiB/s)
append: root=/dev/mmcblk0p5 rw rootwait earlycon=uart8250,mmio32,0xfe660000 console=ttyS2,1500000n8 logo.nologo vt.cur_default=1
Retrieving file: /dtbs/rockchip/rk3566-x96_x6.dtb
reading /dtbs/rockchip/rk3566-x96_x6.dtb
51251 bytes read in 12 ms (4.1 MiB/s)
```

for me all looks ok. And my x96_x6 boots OK

You are saying:

> I have checked your log from u-boot and is strange that is loading x96_x6 dtb instead of the quartz64 dtb that is present in the extlinux.conf.

mm2 design is that for x96_x6 tv box we are using uboot rk3566-x96_x6_defconfig and /dtbs/rockchip/rk3566-x96_x6.dtb kernel DT.

If Yours tv box is not booting with v12.9.0 then may you pls:
1. provide me uart log from failing boot
2. provide me exact pictures of this box pcb ?

thx!

warpme, 2022-11-18:
closing as:
-i can't reproduce original issue
-there seems no replay from issue reporter

closed 2022-11-18 by warpme