warpme/minimyth2 #42: X96 X6 Ethernet is broken

makarioarm, 2022-12-06:
Thanks for your excellent work.

I have tried to connect this box with the ethernet port but fails. Maybe is just my box or  this brand has the ethernet broken?

Wifi is working.

warpme, 2022-12-06:
Hi,
For verification - I just downloaded image from my github, flash and insert into my x96_x6 box. Eth works so this might be issue that you have different hw than me (this is quite frequent practice in cheap Chinese tv boxes)
   
May you pls connect via wifi and send me diagnostic info?
Easier way to do this is:
1.telnet to box
2.type in telnet `tb4`
3.provide me url you are getting when issuing `tb4` command

Also - may you try last ver. I just uploaded: https://github.com/warpme/minimyth2/releases/download/v12.9.0-v33-Pre-935-g53dbe79f11/MiniMyth2-armv8-master-12.9.2.r1015-board-rk3566.x96_x6-SD-Image.img.xz

I new version also fails - may you boot factory Android and make sure is Eth working ok?

makarioarm, 2022-12-06:
I can do something wrong but telnet is not possible, no route to host. I cant ping to the box neither, host unreachable. I think the box has not get the correct gateway. I can see mythtv frontend and media server in network but database is not found by the box.

Android has also the ethernet broken and dmesg gives - can not attach to phy - error. Hardware must be different or faulting.

Your last image boots but hdmi is broken.

TiBismarthome, 2024-07-14:
> Xin chào, Để xác minh - Tôi vừa tải xuống hình ảnh từ github của mình, flash và chèn vào hộp x96_x6 của tôi. Eth hoạt động nên vấn đề này có thể là do bạn có phần cứng khác với tôi (đây là thông lệ khá phổ biến trong các hộp TV giá rẻ của Trung Quốc)
> 
> Bạn vui lòng kết nối qua wifi và gửi cho tôi thông tin chẩn đoán? Cách dễ hơn để thực hiện là: 1. telnet đến hộp 2. nhập telnet `tb4` 3. cung cấp cho tôi url bạn nhận được khi phát `tb4`lệnh
> 
> Ngoài ra - bạn có thể thử phiên bản cuối cùng. Tôi vừa tải lên: https://github.com/warpme/minimyth2/releases/download/v12.9.0-v33-Pre-935-g53dbe79f11/MiniMyth2-armv8-master-12.9.2.r1015-board-rk3566.x96_x6-SD-Image.img.xz
> 
> Phiên bản mới cũng bị lỗi - bạn có thể khởi động lại Android gốc và kiểm tra xem Eth có hoạt động bình thường không?

can you help me this link?