warpme/minimyth2 #43: Fail to compile for Orange Pi Zero2 libpython2.7.so.1.0

paulsykes, 2022-12-23:
Failing to compile for Orange Pi zero 2

```
ln: failed to access 'libpython2.7.so.1.0': No such file or directory
```

Steps to Reproduce:
1. Install Ubuntu Server 22.04 .1inside KVM VM (8 Cores, 8GB Memory, default Kernel 5.15.0-56-generic)
2. Create minimyth user during install
4. Log in as minimyth 
5. Apt update and upgrade
6. Follow instructions from Wiki

I've tried current Head and also commit id from last release 43b517beda92bfbc400222f0b453c677e7f8780  

Build runs for some time without any issues. Prior to Kernel build I accept defaults for all NEW kernel config options. This seems strange to me that this is a manual operation?

Config for you reference (.txt added so it would upload)
[minimyth.conf.mk.txt](https://github.com/warpme/minimyth2/files/10293642/minimyth.conf.mk.txt)

Console output at error
```
arm-minimyth-linux-gnueabi-ar rc libpython2.7.a Modules/threadmodule.o  Modules/signalmodule.o  Modules/posixmodule.o  Modules/errnomodule.o  Modules/pwdmodule.o  Modules/_sre.o  Modules/_codecsmodule.o  Modules/_weakref.o  Modules/zipimport.o  Modules/symtablemodule.o  Modules/xxsubtype.o
arm-minimyth-linux-gnueabi-ranlib libpython2.7.a
lto-wrapper: warning: using serial compilation of 27 LTRANS jobs
lto-wrapper: note: see the ‘-flto’ option documentation for more information
/tmp/ccgFcciB.s: Assembler messages:
/tmp/ccgFcciB.s:73: Error: invalid constant (0) after fixup
/tmp/ccgFcciB.s:76: Error: invalid constant (0) after fixup
/tmp/ccgFcciB.s:302: Error: invalid constant (1) after fixup
/tmp/ccgFcciB.s:570: Error: invalid constant (1) after fixup
/tmp/ccgFcciB.s:1336: Error: invalid constant (0) after fixup
/tmp/ccgFcciB.s:1770: Error: invalid constant (0) after fixup
/tmp/ccgFcciB.s:2239: Error: invalid constant (0) after fixup
/tmp/ccgFcciB.s:2457: Error: invalid constant (1) after fixup
/tmp/ccgFcciB.s:2489: Error: invalid constant (0) after fixup
/tmp/ccgFcciB.s:2493: Error: invalid constant (1) after fixup
/tmp/ccgFcciB.s:2496: Error: invalid constant (80) after fixup
/tmp/ccgFcciB.s:2500: Error: invalid constant (3) after fixup
/tmp/ccgFcciB.s:2534: Error: invalid constant (0) after fixup
/tmp/ccgFcciB.s:2554: Error: invalid constant (100) after fixup
/tmp/ccgFcciB.s:2589: Error: invalid constant (17) after fixup
/tmp/ccgFcciB.s:2713: Error: invalid constant (0) after fixup
/tmp/ccgFcciB.s:2781: Error: invalid constant (8000) after fixup
/tmp/ccgFcciB.s:2823: Error: invalid constant (20000) after fixup
/tmp/ccgFcciB.s:2833: Error: invalid constant (10000) after fixup
/tmp/ccgFcciB.s:2970: Error: invalid constant (1) after fixup
/tmp/ccgFcciB.s:3032: Error: invalid constant (ffffffffffffffff) after fixup
/tmp/ccgFcciB.s:3062: Error: invalid constant (ffffffffffffffff) after fixup
/tmp/ccgFcciB.s:3232: Error: invalid constant (ffffffffffffffff) after fixup
/tmp/ccgFcciB.s:3501: Error: invalid constant (1) after fixup
/tmp/ccgFcciB.s:3527: Error: invalid constant (0) after fixup
/tmp/ccgFcciB.s:3530: Error: invalid constant (1) after fixup
/tmp/ccgFcciB.s:3532: Error: invalid constant (80) after fixup
/tmp/ccgFcciB.s:3538: Error: invalid constant (100) after fixup
/tmp/ccgFcciB.s:3598: Error: invalid constant (0) after fixup
/tmp/ccgFcciB.s:3615: Error: invalid constant (3) after fixup
/tmp/ccgFcciB.s:3629: Error: invalid constant (17) after fixup
/tmp/ccgFcciB.s:3732: Error: invalid constant (7) after fixup
/tmp/ccgFcciB.s:4007: Error: invalid constant (1) after fixup
/tmp/ccgFcciB.s:4071: Error: invalid constant (1) after fixup
/tmp/ccgFcciB.s:4382: Error: invalid constant (7) after fixup
/tmp/ccgFcciB.s:4855: Error: invalid constant (398) after fixup
/tmp/ccgFcciB.s:4888: Error: invalid constant (a) after fixup
/tmp/ccgFcciB.s:4945: Error: invalid constant (2000) after fixup
/tmp/ccgFcciB.s:5711: Error: invalid constant (0) after fixup
/tmp/ccgFcciB.s:6687: Error: invalid constant (1) after fixup
/tmp/ccgFcciB.s:7303: Error: invalid constant (0) after fixup
/tmp/ccgFcciB.s:7308: Error: invalid constant (0) after fixup
/tmp/ccgFcciB.s:7361: Error: invalid constant (0) after fixup
[... 211 lines cut ...]
make[11]: Leaving directory '/home/minimyth/minimyth2/script/opengl/mesa'
make[10]: *** [../../gar.lib.mk:591: imgdep-main] Error 2
make[10]: Leaving directory '/home/minimyth/minimyth2/script/opengl/mesa'
make[9]: *** [../../gar.lib.mk:572: ../../opengl/mesa/cookies/main.d/install] Error 2
make[9]: Leaving directory '/home/minimyth/minimyth2/script/opengl/opengl'
make[8]: *** [../../gar.lib.mk:591: imgdep-main] Error 2
make[8]: Leaving directory '/home/minimyth/minimyth2/script/opengl/opengl'
make[7]: *** [../../gar.lib.mk:572: ../../opengl/opengl/cookies/main.d/install] Error 2
make[7]: Leaving directory '/home/minimyth/minimyth2/script/xorg/xorg'
make[6]: *** [../../gar.lib.mk:591: imgdep-main] Error 2
make[6]: Leaving directory '/home/minimyth/minimyth2/script/xorg/xorg'
make[5]: *** [../../gar.lib.mk:572: ../../xorg/xorg/cookies/main.d/install] Error 2
make[5]: Leaving directory '/home/minimyth/minimyth2/script/system/lirc'
make[4]: *** [../../gar.lib.mk:591: imgdep-main] Error 2
make[4]: Leaving directory '/home/minimyth/minimyth2/script/system/lirc'
make[3]: *** [../../gar.lib.mk:572: ../../system/lirc/cookies/main.d/install] Error 2
make[3]: Leaving directory '/home/minimyth/minimyth2/script/system/irxkeys'
make[2]: *** [../../gar.lib.mk:591: imgdep-main] Error 2
make[2]: Leaving directory '/home/minimyth/minimyth2/script/system/irxkeys'
make[1]: *** [../../gar.lib.mk:572: ../../system/irxkeys/cookies/main.d/install] Error 2
make[1]: Leaving directory '/home/minimyth/minimyth2/script/meta/minimyth'
make: *** [../../gar.lib.mk:591: imgdep-main] Error 2
```

I've used the supplied binary image and it works great, however I was wanting to have a poke around a bit so was attempting to compile from source. Many thanks, the project looks really good.

warpme, 2022-12-26:
Paul,

I fixed building for armv7 target. 
I tested build on rpi3 and It seems work well for wayland and eglfs. 
x11 gives black screen.
Generally:  have hell of problems with building bootloaders for armv7 target for other than rpi boards :-(
Is there any reason why you are building for armv7 instead of aarch64 target?

paulsykes, 2022-12-27:
Oh dear, I shouldn't have been building for armv7, I should have been building for aarch64. I didn't read far enough down the config file. I wrongly assumed that selecting  mm_BOARD_TYPE would also select the correct mm_GARCH.  I will update my mm_GARCH to aarch64 and try again. 

Many thanks for your assistance.

warpme, 2023-01-06:
Paul,
fyi: https://github.com/warpme/miniarch

closed 2023-01-11 by warpme