warpme/minimyth2 #44: Rockchip -- Wayland Black Screen

stvhay, 2023-01-28:
Hey @warpme , cool project! I noticed you had a note that you are getting a black screen in Wayland on Rockchip boards. I am also struggling to get these boards working and for me its not even a black screen. Im finding when I launch sway, seatd completely locks up and sway segfaults. 

Anyways, thought maybe we could work together to try to figure out what's going on here to see if we can get it working.

warpme, 2023-01-28:
Steve,
Thx for kind words!
Interesting is that with https://github.com/warpme/miniarch i can get nicely working EndeavourOS (EOS) on rk on wayland.
This means mm2 kernel (used in miniarch) seems to be working ok with wayland (i tested EOS with KDE DE; you may verify EOS bootstrapping with MiniArch) 
So maybe archlinux guys added some patches in mesa/wayland for rk to get wayland working? (i doubt)

Currently i don't have idea why the same kernel works ok in EOS and fails in MiniMyth2 :-\

stvhay, 2023-01-29:
I recently set up buildroot to have wayland provide the libwayland-egl.so and I’m wondering if we need just the gem version of the Mali driver.  Going to try that.Sent from my iPhone

stvhay, 2023-01-30:
Let's list out the differences that could exist:

| Difference | Notes | 
|------------|-------|
| Who provides libwayland-egl.so ? | mali or wayland provided. check if links to blob |
| Are the GPU drivers compiled with a "hook" ? | Check for libmali_hook in /usr/lib |
| What is the kernel DDK | Check kernel sources `MALI_RELEASE_NAME ?= '"g15p0-01eac0"'` |
| What firmware is used? | Check SHA256 |

What I've got:
- I've tried both providers of wayland-egl.so 
- The drivers do not work without a hook, so I am currently using the hook.
- Kernel DDK = g15p0-01eac0
- sha256 ad91511a77fa110af46a2a29cad9e225524d9cfa9fe35bdce96606d354e607de mali_csffw.bin

warpme, 2023-01-30:
Steve,
I'm a bit lost here....

Situation i have:
1. single binary having full video stack (drivers/kernel/drm/egl_provider/wayland_display_server/media_player)
2. aw/aml/rk/api hw
 
Binary from p.1 gives me ok wayland (ui & playback) on all hw - except: rk
As binary is the same - diff is in code which is hw dependent.

So imho i have 2 candidates: drm and egl provider (mesa).
I have black screen on rk3328 (mali450), 3399 (t860) and rk3566/3568 (g52).
So it is v.low probable that issue is in mesa (i.e. mali450 on s905 plays ok with weston)
Looking on weston log - i see absolutely normal log (no errors, standard info entries).
All above hints me: most probable root cause is drm.

Now - in my case: 
- kernel/weston_disp_server/media_player have absolutely normal logs
- i can remotely launch video playback and logs shows me nice playback (including hw accell), audio is playing ok, etc

with above i think issue is drm z-order related
(i.e. looks to me weston draws to drm plane which is overlaid by other plane thus screen is black)

I will consult this with dev. who write majory of patch code required to get rk drm working with dam-planes media playback...

closed 2023-02-11 by warpme