warpme/minimyth2 #45: nvidia-340.108 with linux 6.1 (and maybe 6.0)

pv4, 2023-02-10:
Hi!

I migrated to linux 6.1 from 5.19 a couple of days ago and found the newer kernel requires nvidia-340.108 to be patched more to be successfully built. The patch I use is attached, worked for me fine for a couple of days. But the changes are so that there is a probability some issues can happen (or not, I only have one machine to test) on setups other than my one. Could you please consider reviewing, testing and committing the patch?

Related kernel changes I based the patch on:
- removal of the necessary fields: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/include/acpi/acpi_bus.h?h=v6.2-rc3&id=e6bdbcc764af822ff7172a4a78d437dc433a0c74
- the commit before the previous one, with an example how to migrate the old api to the new one https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v6.2-rc3&id=d674553009afc9b24cab2bbec71628609edbbb27
- definitions of list_for_each_* https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/include/linux/list.h?h=v6.2-rc3

[0013-fix-compile-on-6.1.patch.gz](https://github.com/warpme/minimyth2/files/10703730/0013-fix-compile-on-6.1.patch.gz)

warpme, 2023-02-22:
Sorry for late replay and many thx for your contribution.
I already integrated it into mm2 6.1 kernel - but not tested yet.
(key issue for me is that going with nvidia property driver needs downgrade xorg-server :-(( )

pv4, 2023-02-24:
That's sad :(

What xorg-server version started to break the nvidia driver?

warpme, 2023-02-25:
IIRC last working (with compatible ABI) was: 1.20.x

warpme, 2023-05-25:
I switched mm2 to in-kernel nouveau.
By this nvidia proprietary drivers are unsupported now.
By this I'm closing this ticket.
Thx for your help!

closed 2023-05-25 by warpme