warpme/minimyth2 #46: TanixTX6s - no ap6330 wifi

warpme, 2023-03-02:
Hi Warpme,
I have the same issue with the latest MiniArch on Tanix TX6S with SP6330 wifi chip.

SDIO doesn't show up in dmesg, modprobe brcmfmac does nothing.

Another strange thing is that the board has 4GB of RAM, but linux sees only 3GB.
u-boot SPL shows 4GB, and next u-boot 3GB.

For the rest, MiniArch works perfectly.

Here is log:

> 

U-Boot SPL 2021.04 (Feb 25 2023 - 15:19:32 +0100)
**DRAM: 4096 MiB**
Failed to set core voltage! Can't set CPU frequency
Trying to boot from MMC1
NOTICE:  BL31: v2.8(debug):v12.13.0-v33-Pre-1175-g91e3da68e6-5-g0518a270
NOTICE:  BL31: Built : 15:19:28, Feb 25 2023
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a085a28, model: Tanix TX6s
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
WARNING: PMIC: No regulators DT node, skipping setup
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.

U-Boot 2021.04 (Feb 25 2023 - 15:19:32 +0100) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: Tanix TX6s
I2C:   ready
**DRAM:  3 GiB**
MMC:   mmc@4020000: 0, mmc@4022000: 1
Loading Environment from FAT... Card did not respond to voltage select! : -110
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   No ethernet found.
starting USB...
No working controllers found
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1501 bytes read in 2 ms (732.4 KiB/s)
Executing script at 4fc00000
32115200 bytes read in 2665 ms (11.5 MiB/s)
Failed to load '/h616_dtb'
27993 bytes read in 9 ms (3 MiB/s)
7630304 bytes read in 636 ms (11.4 MiB/s)
Moving Image from 0x40080000 to 0x40200000, end=42140000 #1 
Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
   Loading Ramdisk to 498b9000, end 49fffde0 ... OK
   Loading Device Tree to 00000000498af000, end 00000000498b8fff ... OK

Starting kernel ...

:: running early hook [udev]
Starting systemd-udevd version 253-1-arch
:: running hook [udev]
:: Triggering uevents...
:: running hook [keymap]
:: Loading keymap...kbd_mode: KDSKBMODE: Inappropriate ioctl for device
done.
:: performing fsck on '/dev/mmcblk0p2'
ROOT: clean, 31331/1866384 files, 332874/3811584 blocks
:: mounting '/dev/mmcblk0p2' on real root
:: running cleanup hook [udev]

Welcome to Arch Linux ARM!

[  OK  ] Created slice Slice /system/getty.
[  OK  ] Created slice Slice /system/modprobe.
[  OK  ] Created slice Slice /system/serial-getty.
[  OK  ] Created slice User and Session Slice.
[  OK  ] Started Dispatch Password …ts to Console Directory Watch.
[  OK  ] Started Forward Password R…uests to Wall Directory Watch.
[UNSUPP] Starting of Arbitrary Exec…m Automount Point unsupported.
[  OK  ] Reached target Local Encrypted Volumes.
[  OK  ] Reached target Local Integrity Protected Volumes.
[  OK  ] Reached target Path Units.
[  OK  ] Reached target Remote File Systems.
[  OK  ] Reached target Slice Units.
[  OK  ] Reached target Swaps.
[  OK  ] Reached target Local Verity Protected Volumes.
[  OK  ] Listening on Device-mapper event daemon FIFOs.
[  OK  ] Listening on Process Core Dump Socket.
[  OK  ] Listening on Journal Socket (/dev/log).
[  OK  ] Listening on Journal Socket.
[  OK  ] Listening on Network Service Netlink Socket.
[  OK  ] Listening on udev Control Socket.
[  OK  ] Listening on udev Kernel Socket.
         Mounting POSIX Message Queue File System...
         Mounting Kernel Debug File System...
         Mounting Kernel Trace File System...
         Mounting Temporary Directory /tmp...
         Starting Create List of Static Device Nodes...
         Starting Load Kernel Module configfs...
         Starting Load Kernel Module dm_mod...
         Starting Load Kernel Module drm...
         Starting Load Kernel Module fuse...
         Starting Load Kernel Module loop...
         Starting Journal Service...
         Starting Generate network …ts from Kernel command line...
         Starting Remount Root and Kernel File Systems...
         Starting Apply Kernel Variables...
         Starting Coldplug All udev Devices...
[  OK  ] Mounted POSIX Message Queue File System.
[  OK  ] Mounted Kernel Debug File System.
[  OK  ] Mounted Kernel Trace File System.
[  OK  ] Mounted Temporary Directory /tmp.
[  OK  ] Finished Create List of Static Device Nodes.
[  OK  ] Finished Load Kernel Module configfs.
[  OK  ] Started Journal Service.
[  OK  ] Finished Load Kernel Module dm_mod.
[  OK  ] Finished Load Kernel Module drm.
[  OK  ] Finished Load Kernel Module fuse.
[  OK  ] Finished Load Kernel Module loop.
[  OK  ] Finished Generate network units from Kernel command line.
[  OK  ] Finished Remount Root and Kernel File Systems.
[  OK  ] Finished Apply Kernel Variables.
[  OK  ] Reached target Preparation for Network.
         Mounting Kernel Configuration File System...
         Starting Flush Journal to Persistent Storage...
         Starting Load/Save OS Random Seed...
         Starting Create Static Device Nodes in /dev...
[  OK  ] Mounted Kernel Configuration File System.
[  OK  ] Finished Coldplug All udev Devices.
[  OK  ] Finished Flush Journal to Persistent Storage.
[  OK  ] Finished Create Static Device Nodes in /dev.
[  OK  ] Reached target Preparation for Local File Systems.
         Starting Rule-based Manage…for Device Events and Files...
[  OK  ] Started Rule-based Manager for Device Events and Files.
         Starting Network Configuration...
[  OK  ] Found device /dev/ttyS0.
[  OK  ] Started Network Configuration.
[  OK  ] Found device /dev/disk/by-label/BOOT.
         Mounting /boot...
[  OK  ] Mounted /boot.
[  OK  ] Reached target Local File Systems.
[  OK  ] Reached target Hardware activated USB gadget.
         Starting Create Volatile Files and Directories...
[  OK  ] Finished Load/Save OS Random Seed.
[  OK  ] Reached target Sound Card.
[  OK  ] Listening on Load/Save RF …itch Status /dev/rfkill Watch.
[  OK  ] Finished Create Volatile Files and Directories.
         Starting Network Name Resolution...
         Starting Network Time Synchronization...
         Starting Record System Boot/Shutdown in UTMP...
[  OK  ] Finished Record System Boot/Shutdown in UTMP.
[  OK  ] Started Network Time Synchronization.
[  OK  ] Reached target System Time Set.
[  OK  ] Started Network Name Resolution.
[  OK  ] Reached target Network.
[  OK  ] Reached target Host and Network Name Lookups.
[  OK  ] Reached target System Initialization.
[  OK  ] Started Refresh existing P…f archlinux-keyring regularly.
[  OK  ] Started Daily verification of password and group files.
[  OK  ] Started Daily Cleanup of Temporary Directories.
[  OK  ] Reached target Timer Units.
[  OK  ] Listening on D-Bus System Message Bus Socket.
[  OK  ] Reached target Socket Units.
[  OK  ] Reached target Basic System.
         Starting D-Bus System Message Bus...
[  OK  ] Started OpenSSH Daemon.
         Starting User Login Management...
         Starting Permit User Sessions...
[  OK  ] Finished Permit User Sessions.
[  OK  ] Started Getty on tty1.
[  OK  ] Started Serial Getty on ttyS0.
[  OK  ] Reached target Login Prompts.
[  OK  ] Started User Login Management.
[  OK  ] Started D-Bus System Message Bus.
[  OK  ] Reached target Multi-User System.
[  OK  ] Reached target Graphical Interface.

Arch Linux 6.2.0 (ttyS0)

alarm login:

_Originally posted by @arg7 in https://github.com/warpme/minimyth2/issues/10#issuecomment-1451470824_

warpme, 2023-03-02:
Hi,

No SDIO in dmesg most probably means no power is enabled for SDIO interface by sdio-pwrseq.
Most probably because DT has wrong pin declaration for GPIO pin used to turn-on power on SDIO interface.
May you boot android on this box and get GPIO info?
Probably below commands in android terminal will be enough:

```
mount -t debugfs debugfs /sys/kernel/debug
cat /sys/kernel/debug/gpio 
```

arg7, 2023-03-02:
Thanks for your support:
cat /sys/kernel/debug/gpio shows:
```
gpiochip1: GPIOs 0-287, parent: platform/pio, pio:
 gpio-207 (                    |wlan_hostwake       ) in  lo
 gpio-208 (                    |bt_hostwake         ) in  hi
 gpio-209 (                    |bt_wake             ) out hi
 gpio-210 (                    |wlan_regon          ) out hi
 gpio-211 (                    |bt_rst              ) out lo
 gpio-230 (                    |?                   ) out hi
 gpio-231 (                    |?                   ) out lo
 gpio-232 (                    |?                   ) out hi
 gpio-267 (                    |fd650_clk           ) out hi
 gpio-268 (                    |fd650_dat           ) in  hi
 gpio-272 (                    |cd                  ) in  hi

```

also i've noticed that lsmod shows:
```
Module                  Size  Used by
bcmdhd               1323008  0
bcm_btlpm              16384  0
uvcvideo              102400  0
videobuf2_v4l2         28672  1 uvcvideo
videobuf2_vmalloc      16384  1 uvcvideo
videobuf2_memops       16384  1 videobuf2_vmalloc
videobuf2_core         49152  2 uvcvideo,videobuf2_v4l2
mali_kbase            544768  7

```

and  dmesg | grep wlan shows:
```
[   33.418772] ======== dhd_wlan_init_plat_data ========
[   33.428718] dhd_wlan_init_gpio: WL_HOST_WAKE=-1, oob_irq=192, oob_irq_flags=0x4084
[   33.437789] dhd_wlan_init_gpio: WL_REG_ON=-1
[   33.788535] sunxi-wlan soc@03000000:wlan: bus_index: 1
[   34.333760] Register interface [wlan0]  MAC: e0:76:d0:4f:0e:e4\x0a
[   34.535457] init: Created socket '/dev/socket/wpa_wlan0', mode 660, user 1010, group 1010
```

looks like **bcmdhd** is responsible for wifi, but it is not present in MiniArch.

Best regards,
AR

arg7, 2023-03-02:
I've made a donation of 200PL to your cause.

warpme, 2023-03-05:
Sorry for delay. (i was a bit busy with adding usb/pcie/eth for rk3588 soc)

Many thx for donation!

Offer of sending to me box for fixing wifi is very kind - lets park this as plan B :-)
My plan A is do my best to get wifi working with minimal effort from your side.

GPIO info you provided me conforms GPIO pin used to power-on wifi module is correct one (android says gpio-210 wich is PG18) DT seems to set it correctly (https://github.com/warpme/minimyth2/blob/4f5f0d2bd8d43b80d655a7d6a2ee151273bd92b2/script/kernel/linux-6.2/files/0633-arm64-dts-allwinner-h616-add-Tanix-TX6s-TVbox.patch#L87)). 

I have another box with working ok ap6330. 
It looks like your's tx6s has dead wifi because no 32kHz clock is supplied to ap6330 module.
I'm investigating how to add such 32k clock in h616 platforms.

Stay tuned!

ps: once again: many thx for recognising my work!

warpme, 2023-03-05:
I added 32k clock in tx6s dt.
May you pls try update kernel in miniarch?
(in terminal pls type '`start`' and select '4')
This should update kernel to 6.2.2.
Reboot and pls provide me:
-dmesg
-`cat /sys/kernel/debug/clk/clk_summary`

It there will be still issue with wifi, pls provide me `cat /sys/kernel/debug/clk/clk_summary` also from android

warpme, 2023-04-08:
qll - i have progress here:

```
[    6.041712] brcmfmac mmc1:0001:1: loading /lib/firmware/brcm/brcmfmac-ap6330-sdio.tanix,tx6s.clm_blob failed with error -22
[    6.070315] rc rc0: sunxi-ir as /devices/platform/soc/7040000.ir/rc/rc0
[    6.070603] rc rc0: lirc_dev: driver sunxi-ir registered at minor = 0, raw IR receiver, no transmitter
[    6.070818] input: sunxi-ir as /devices/platform/soc/7040000.ir/rc/rc0/input0
[    6.076725] sunxi-ir 7040000.ir: initialized sunXi IR driver
[    6.361365] brcmfmac mmc1:0001:1: loading /lib/firmware/brcm/brcmfmac-ap6330-sdio.clm_blob failed with error -22
[    6.361403] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac-ap6330-sdio.clm_blob failed with error -22
[    6.361410] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac-ap6330-sdio.clm_blob
[    6.687773] brcmfmac_wcc: brcmf_wcc_attach: executing
[    6.688877] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    6.689233] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4330/4 wl0: Jan 23 2013 17:46:35 version 5.90.195.114 FWID 01-cafa6b3e
```

```
root@Myth-Frontend-0200252ec1ea:~ # ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq qlen 1000
    link/ether 02:00:25:2e:c1:ea brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop qlen 1000
    link/ether e0:76:d0:4f:0e:e4 brd ff:ff:ff:ff:ff:ff
```

warpme, 2023-04-18:
I uploaded new versions of images with ap6330 support.
I think we may close this ticket.
Many thx for your's support!

closed 2023-04-18 by warpme