warpme/minimyth2 #47: TanixTX6 4G - kernel discovers only 3G of RAM

warpme, 2023-03-02:
Board has 4GB of RAM, but linux sees only 3GB.
u-boot SPL shows 4GB, and next u-boot 3GB.

For the rest, MiniArch works perfectly.

Here is log:

> 

U-Boot SPL 2021.04 (Feb 25 2023 - 15:19:32 +0100)
**DRAM: 4096 MiB**
Failed to set core voltage! Can't set CPU frequency
Trying to boot from MMC1
NOTICE:  BL31: v2.8(debug):v12.13.0-v33-Pre-1175-g91e3da68e6-5-g0518a270
NOTICE:  BL31: Built : 15:19:28, Feb 25 2023
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a085a28, model: Tanix TX6s
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
WARNING: PMIC: No regulators DT node, skipping setup
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.

U-Boot 2021.04 (Feb 25 2023 - 15:19:32 +0100) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: Tanix TX6s
I2C:   ready
**DRAM:  3 GiB**
MMC:   mmc@4020000: 0, mmc@4022000: 1
Loading Environment from FAT... Card did not respond to voltage select! : -110
In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   No ethernet found.
starting USB...
No working controllers found
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1501 bytes read in 2 ms (732.4 KiB/s)
Executing script at 4fc00000
32115200 bytes read in 2665 ms (11.5 MiB/s)
Failed to load '/h616_dtb'
27993 bytes read in 9 ms (3 MiB/s)
7630304 bytes read in 636 ms (11.4 MiB/s)
Moving Image from 0x40080000 to 0x40200000, end=42140000 #1 
Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
   Loading Ramdisk to 498b9000, end 49fffde0 ... OK
   Loading Device Tree to 00000000498af000, end 00000000498b8fff ... OK

Starting kernel ...

:: running early hook [udev]
Starting systemd-udevd version 253-1-arch
:: running hook [udev]
:: Triggering uevents...
:: running hook [keymap]
:: Loading keymap...kbd_mode: KDSKBMODE: Inappropriate ioctl for device
done.
:: performing fsck on '/dev/mmcblk0p2'
ROOT: clean, 31331/1866384 files, 332874/3811584 blocks
:: mounting '/dev/mmcblk0p2' on real root
:: running cleanup hook [udev]

Welcome to Arch Linux ARM!

[  OK  ] Created slice Slice /system/getty.
[  OK  ] Created slice Slice /system/modprobe.
[  OK  ] Created slice Slice /system/serial-getty.
[  OK  ] Created slice User and Session Slice.
[  OK  ] Started Dispatch Password …ts to Console Directory Watch.
[  OK  ] Started Forward Password R…uests to Wall Directory Watch.
[UNSUPP] Starting of Arbitrary Exec…m Automount Point unsupported.
[  OK  ] Reached target Local Encrypted Volumes.
[  OK  ] Reached target Local Integrity Protected Volumes.
[  OK  ] Reached target Path Units.
[  OK  ] Reached target Remote File Systems.
[  OK  ] Reached target Slice Units.
[  OK  ] Reached target Swaps.
[  OK  ] Reached target Local Verity Protected Volumes.
[  OK  ] Listening on Device-mapper event daemon FIFOs.
[  OK  ] Listening on Process Core Dump Socket.
[  OK  ] Listening on Journal Socket (/dev/log).
[  OK  ] Listening on Journal Socket.
[  OK  ] Listening on Network Service Netlink Socket.
[  OK  ] Listening on udev Control Socket.
[  OK  ] Listening on udev Kernel Socket.
         Mounting POSIX Message Queue File System...
         Mounting Kernel Debug File System...
         Mounting Kernel Trace File System...
         Mounting Temporary Directory /tmp...
         Starting Create List of Static Device Nodes...
         Starting Load Kernel Module configfs...
         Starting Load Kernel Module dm_mod...
         Starting Load Kernel Module drm...
         Starting Load Kernel Module fuse...
         Starting Load Kernel Module loop...
         Starting Journal Service...
         Starting Generate network …ts from Kernel command line...
         Starting Remount Root and Kernel File Systems...
         Starting Apply Kernel Variables...
         Starting Coldplug All udev Devices...
[  OK  ] Mounted POSIX Message Queue File System.
[  OK  ] Mounted Kernel Debug File System.
[  OK  ] Mounted Kernel Trace File System.
[  OK  ] Mounted Temporary Directory /tmp.
[  OK  ] Finished Create List of Static Device Nodes.
[  OK  ] Finished Load Kernel Module configfs.
[  OK  ] Started Journal Service.
[  OK  ] Finished Load Kernel Module dm_mod.
[  OK  ] Finished Load Kernel Module drm.
[  OK  ] Finished Load Kernel Module fuse.
[  OK  ] Finished Load Kernel Module loop.
[  OK  ] Finished Generate network units from Kernel command line.
[  OK  ] Finished Remount Root and Kernel File Systems.
[  OK  ] Finished Apply Kernel Variables.
[  OK  ] Reached target Preparation for Network.
         Mounting Kernel Configuration File System...
         Starting Flush Journal to Persistent Storage...
         Starting Load/Save OS Random Seed...
         Starting Create Static Device Nodes in /dev...
[  OK  ] Mounted Kernel Configuration File System.
[  OK  ] Finished Coldplug All udev Devices.
[  OK  ] Finished Flush Journal to Persistent Storage.
[  OK  ] Finished Create Static Device Nodes in /dev.
[  OK  ] Reached target Preparation for Local File Systems.
         Starting Rule-based Manage…for Device Events and Files...
[  OK  ] Started Rule-based Manager for Device Events and Files.
         Starting Network Configuration...
[  OK  ] Found device /dev/ttyS0.
[  OK  ] Started Network Configuration.
[  OK  ] Found device /dev/disk/by-label/BOOT.
         Mounting /boot...
[  OK  ] Mounted /boot.
[  OK  ] Reached target Local File Systems.
[  OK  ] Reached target Hardware activated USB gadget.
         Starting Create Volatile Files and Directories...
[  OK  ] Finished Load/Save OS Random Seed.
[  OK  ] Reached target Sound Card.
[  OK  ] Listening on Load/Save RF …itch Status /dev/rfkill Watch.
[  OK  ] Finished Create Volatile Files and Directories.
         Starting Network Name Resolution...
         Starting Network Time Synchronization...
         Starting Record System Boot/Shutdown in UTMP...
[  OK  ] Finished Record System Boot/Shutdown in UTMP.
[  OK  ] Started Network Time Synchronization.
[  OK  ] Reached target System Time Set.
[  OK  ] Started Network Name Resolution.
[  OK  ] Reached target Network.
[  OK  ] Reached target Host and Network Name Lookups.
[  OK  ] Reached target System Initialization.
[  OK  ] Started Refresh existing P…f archlinux-keyring regularly.
[  OK  ] Started Daily verification of password and group files.
[  OK  ] Started Daily Cleanup of Temporary Directories.
[  OK  ] Reached target Timer Units.
[  OK  ] Listening on D-Bus System Message Bus Socket.
[  OK  ] Reached target Socket Units.
[  OK  ] Reached target Basic System.
         Starting D-Bus System Message Bus...
[  OK  ] Started OpenSSH Daemon.
         Starting User Login Management...
         Starting Permit User Sessions...
[  OK  ] Finished Permit User Sessions.
[  OK  ] Started Getty on tty1.
[  OK  ] Started Serial Getty on ttyS0.
[  OK  ] Reached target Login Prompts.
[  OK  ] Started User Login Management.
[  OK  ] Started D-Bus System Message Bus.
[  OK  ] Reached target Multi-User System.
[  OK  ] Reached target Graphical Interface.

Arch Linux 6.2.0 (ttyS0)

alarm login:

_Originally posted by @arg7 in https://github.com/warpme/minimyth2/issues/10#issuecomment-1451470824_

warpme, 2023-03-16:
I uploaded test build with potential 4G uboot fix: https://github.com/warpme/miniarch/releases/download/20230226-6.2-6b34b2ce/MiniArch-20230212-6.2.6-board-h616.tanix_tx6s-SD-Image.img.xz

May you pls try and report what uboot says on this build?

warpme, 2023-03-17:
Indeed: "_8GB of ram on cold boot, and 4GB after reboot_" - i heard about this uboot bug(issue).
I will try investigate further.
But should I understand: does Linux sees whole 4G and works well with full RAM?

re: tx6s: it looks like phy presence will be helpful....
Nature of issue requires probably use of oscilloscope / logical analyser to see what working Android provides and what I missing with in MiniArch....   

If it is ok for you - pls send me PM - i will give you logistic details :-)

arg7, 2023-03-17:
linux sees and uses correctly 4GB of ram, I have done some quick tests with RAM disk of 3GB (all free ram after boot), it works ok.

arg7, 2023-03-17:
but, how can I send you PM? looks like github doesn't support such a feature.
can you send me your address on my mobile phone +393936593340 via WhatsApp or sms?

warpme, 2023-03-17:
Argh - forgive me!
my PM: piotr.oniszczuk@gmail.com

re: 4G issue - many thx for testing.
I will close this ticket and potentially reopen when 8GB of ram on cold boot, and 4GB after reboot will be resolved by uboot team.

closed 2023-03-17 by warpme