warpme/minimyth2 #48: H313 X96Q testing

xAbdaziz, 2023-03-17:
Hello, I saw this [commit](https://github.com/warpme/minimyth2/commit/49ffa0a746d507e4dfb0fdffc43bca053bfe73c0), and I would like to help you in testing it if you need someone, I have UART cable so I can send logs

xAbdaziz, 2023-03-17:
> Thx for help! I still can't get my x96-q box showing me anything on uart.... (uart hw is ok as vendor Android shows me boot messages). So if you can send me: -android boot log -linux boot log (if you have any form of linux booting on your box) -pcb pictures (if you have) -RAM chips type you have on your box (if you can read them) thx in advance

I'm glad to be able to help, I forgot to mention that I have the second revision of the box (X96Q2)

android boot log: https://pastebin.com/LZzaDjPA
linux boot log: couldn't get linux to boot on it, I tried booting [this](https://github.com/hexdump0815/imagebuilder/releases/tag/230222-02), but this what shows in console:
`U-Boot SPL 2021.04-18166-g647b392bf2-dirty (Jan 27 2023 - 08:09:08 +0100) 
DRAM:This DRAM setup is currently not supported.`

pcb: I couldn't get a clear picture, but here is one I found online: https://i.ibb.co/8gP74QN/photo-2022-11-02-15-18-06.jpg, seems to be the same as my pcb except that I have a Micron EMMC
RAM chips: couldn't read them unfortunately. but it's LPDDR3 according to android boot log.

warpme, 2023-03-17:
Hmm, 
thats good that you have any uart output !
may you pls try https://github.com/warpme/minimyth2/releases/download/v12.13.0-v33-Pre-1175-g91e3da68e6/MiniMyth2-armv8-master-12.13.1.r1175-board-h313.x96_q-SD-Image.img.xz

of course uart bootlog pls

to develop support - getting android dt will be a must i think.
Is it possible for you to attach adb (android debugger) via usb cable to this box?
If you have usb-usb cable (example https://nvidia.custhelp.com/app/answers/detail/a_id/4367/~/how-to-use-the-new-shield-tv-in-mtp-or-adb-mode) - i can provide you help....

xAbdaziz, 2023-03-17:
> 

Yes, I can connect to the box using adb.
I will try the image you just sent and I will report back

xAbdaziz, 2023-03-17:
> Hmm, thats good that you have any uart output ! may you pls try [...]

Yes, I can connect to the box using adb.
I will try the image you just sent and I will report back

xAbdaziz, 2023-03-17:
> Hmm, thats good that you have any uart output ! may you pls try [...]

This is the output of the image you sent, it's stuck and nothing shows after that:

`U-Boot SPL 2023.01-rc3 (Mar 17 2023 - 14:15:21 +0100)
DRAM: 128 MiB
Failed to set core voltage! Can't set CPU frequency`

warpme, 2023-03-17:
Ok - this sounds encouraging!
It is possible DRAM params i put into U-Boot are wrong for your box.
Here we may try box Android device-tree - maybe it has correct values. 
You may try following procedure to get DT: https://github.com/warpme/minimyth2/wiki/Getting-device-tree-from-TVbox-vendor-Android

With box DT we may try address PMIC issue with `Failed to set core voltage! Can't set CPU frequency`

warpme, 2023-03-18:
Argh - your box has axp1530 and lpddr3.
dt has 31 different dram param sets...
huh
without having box physically i think it will be real pain to get it working....

xAbdaziz, 2023-03-21:
> Argh - your box has axp1530 and lpddr3. dt has 31 different dram param sets... huh without having box physically i think it will be real pain to get it working....

Oh sad, I hoped that I can help you :(

xAbdaziz, 2023-03-30:
@warpme Hi, I've tried your Tanix LPDDR3 build.

warpme, 2023-04-02:
Recently I bought x96-q box with lpddr3 memory and successfully add support of MiniMyth and MiniAch on this HW.
You may try: https://github.com/warpme/miniarch/releases/download/20230226-6.2-6b34b2ce/MiniArch-20230212-6.2.9-board-h313.x96_q_lpddr3-SD-Image.img.xz

xAbdaziz, 2023-04-02:
> Recently I bought x96-q box with lpddr3 memory and successfully add support of MiniMyth and MiniAch on this HW. You may try: https://github.com/warpme/miniarch/releases/download/20230226-6.2-6b34b2ce/MiniArch-20230212-6.2.9-board-h313.x96_q_lpddr3-SD-Image.img.xz

Cool!
I will try it now

xAbdaziz, 2023-04-02:
> Recently I bought x96-q box with lpddr3 memory and successfully add support of MiniMyth and MiniAch on this HW. You may try: https://github.com/warpme/miniarch/releases/download/20230226-6.2-6b34b2ce/MiniArch-20230212-6.2.9-board-h313.x96_q_lpddr3-SD-Image.img.xz

Yup, I can confirm that it works.
Thank you

xAbdaziz, 2023-04-02:
@warpme btw is it possible to to add to MiniArch's start script an option to install the image on eMMC?
SDcard is very slow and takes forever to do anything.

xAbdaziz, 2023-04-02:
@warpme also there's no wifi interface in `ifconfig`

warpme, 2023-04-09:
[xAb](https://github.com/xAbdaziz) may you try this image:  https://github.com/warpme/miniarch/releases/download/20230226-6.2-6b34b2ce/MiniArch-20230212-6.2.10-board-h313.x96_q_lpddr3-SD-Image.img.xz

xAbdaziz, 2023-04-11:
> @xAbdaziz may you try this image: https://github.com/warpme/miniarch/releases/download/20230226-6.2-6b34b2ce/MiniArch-20230212-6.2.10-board-h313.x96_q_lpddr3-SD-Image.img.xz

Sure

xAbdaziz, 2023-04-11:
> @xAbdaziz may you try this image: https://github.com/warpme/miniarch/releases/download/20230226-6.2-6b34b2ce/MiniArch-20230212-6.2.10-board-h313.x96_q_lpddr3-SD-Image.img.xz

Unfortunately, there is still no WiFi interface
[image]

xAbdaziz, 2023-04-11:
> may you give me output of `dmesg` ?

Sure, http://ix.io/4td9

warpme, 2023-04-18:
Hi,
I uploaded new ver of miniarch.
may you try it?
If boot will still fail - may you give me uart output + pictures of pcb (including numbers on dram chips)
thx!

xAbdaziz, 2023-05-05:
@warpme sorry, i went on a vacation and totally forgot about testing.
the build boots, but still no wifi interface unfortunately.

closed 2023-05-25 by warpme

kerta1n, 2023-09-24:
@warpme I just booted your 6.3.5 X96Q (H313, LPDDR3 2GB) image, and there is indeed WiFi, but no ethernet. `dmesg | grep eth` reports `EMAC reset timeout` and `probe of 5030000.ethernet failed with error -110`. Attempting to use a USB to Ethernet dongle fails with a firmware load error of -2. Going to try booting a 6.2 build

kerta1n, 2023-09-24:
Found out there were no builds for 6.2, but I got my dongle working!! Upon logging in, I ran `su`, used `root` as pwd, and then `depmod -a`. Gave the dongle time to think, then got an IP address about 20 seconds later. Going to extract eMMC contents before I install Linux to it (will host somewhere later).

kipz, 2024-07-03:
FWIW, everything is working on my X96Q v1.3: https://gist.github.com/kipz/93caf0f38944aa726d48135d7261b605