warpme/minimyth2 #49: Tanix H616 no booting

jaum20, 2023-03-19:
I downloaded `MiniMyth2-armv8-master-12.13.0.r1175-board-h616.tanix_tx6s-SD-Image.img.xz`, but [my box](https://www.aliexpress.com/item/1005004347477416.html?spm=a2g0o.order_list.order_list_main.5.208ccaa4kLtg91) sun50iw9p1 is not booting up the image. Any Ideas?

warpme, 2023-03-19:
Hi,
Maybe you have capability to connect UART to this box and see output?
There is more and more different variants of tx6 with different memory types (ddr3 or lpddr3, etc)
When box fails to boot - finding where issue is without uart output might be very difficult....

jaum20, 2023-03-19:
>Maybe you have capability to connect UART to this box and see output?
Sorry, I dont't know how to to that. Which type of hardware do I need ?

warpme, 2023-03-19:
Oh there is plenty uart-usb modules and guides how to do this. Pls try google search :-)
but you also will need to solder 2 cables from usb-uart module to pads on pcb...

jaum20, 2023-03-21:
> Oh there is plenty uart-usb modules and guides how to do this

can you link me one, please?

warpme, 2023-03-21:
Im using module similar to https://www.elektor.com/ch340-usb-to-ttl-converter-uart-module-ch340g-3-3-v-5-5-v
For connecting to board - pls look: https://linux-sunxi.org/Tanix_TX6s

jaum20, 2023-03-22:
Thank you. Some day I will try this

jaum20, 2023-04-18:
I will try it

jaum20, 2023-04-18:
None of them worked 

```
MiniArch-20230212-6.2.11-board-h616.tanix_tx6s-SD-Image.img.xz
```

```
MiniArch-20230212-6.2.11-board-h616.tanix_tx6s_axp313-SD-Image.img.xz
```

warpme, 2023-04-19:
Ah. Sad.
Without uart output I can't help....

jaum20, 2024-08-10:
Tried the last tx6s_axp313 and now it's working. No sure if something changed in the new images or I was doing something wrong before