warpme/minimyth2 #51: OrangePi Zero 2 missing firmware for UWE5622 (aw859a)

takase1121, 2023-05-26:
Wifi did not work with OrangePi Zero 2. I poked around with the dts files, but seems like mainline kernel doesn't have support for it. I looked around the patch and decompiled the dtb currently running on my system, and it shows wifi-pwrseq but not the actual wifi chip on SDIO. I also cannot find the firmware for UWE5622, which should be included.

References:
https://github.com/jernejsk/linux-1/commit/e7222b1f0dfde22a3b60de7f3add6599fd942e84
https://github.com/Ran-Thegoth/uwe5622
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/allwinner/sun50i-h616-orangepi-zero2.dts?h=v6.3.4

warpme, 2023-05-26:
Thx for reporting this!

I fixed this in 9ffb356430406ab401009b7561540f3579f24c2f
and uploaded update Miniarch SD image

`modprobe sprdwl_ng`

```
dmesg
[  162.908045] [ADDR_MGT] addr_mgt_probe: module version: v1.0.9
[  162.908200] [ADDR_MGT] addr_mgt_probe: success.
[  162.933983] WCN: marlin_init entry!
[  162.935046] WCN: wcn config bt wake host
[  162.935255] WCN_ERR: dts node for bt_wake not found
[  162.935262] WCN: marlin2 parse_dt some para not config
[  162.935279] sdiohal:sdiohal_parse_dt adma_tx:1, adma_rx:1, pwrseq:0, irq type:data, gpio_num:0, blksize:840
[  162.946558] sdiohal:sdiohal_init ok
[  162.947699] WCN: marlin_probe ok!
[  162.984190] WCN: start_marlin [MARLIN_WIFI]
[  162.984220] WCN: marlin power state:0, subsys: [MARLIN_WIFI] power 1
[  162.984229] WCN: the first power on start
[  163.089230] WCN: marlin chip en dummy pull up -- need manually set GPIO
[  163.115879] sdiohal:sdiohal_scan_card
[  163.116146] sdiohal:sdiohal_probe: func->class=0, vendor=0x0000, device=0x0000, func_num=0x0001, clock=50000000
[  163.116484] WCN: marlin_get_wcn_chipid: chipid: 0x2355b001
[  163.116494] WCN: marlin_scan_finish!
[  163.116499] sdiohal:probe ok
[  163.116704] sdiohal:scan end!
[  163.116717] WCN: then marlin start to download
[  163.116755] WCN: marlin_request_firmware from /lib/firmware/uwe5622/wcnmodem.bin start!
[  163.156463] WCN: marlin_firmware_parse_image imagepack is WCNM type,need parse it
[  164.665952] WCN: combin_img 0 marlin_firmware_write finish and successful
[  164.666188] WCN: marlin_start_run read reset reg val:0x1
[  164.666195] WCN: after do marlin_start_run reset reg val:0x0
[  164.666448] WCN: s_marlin_bootup_time=164666448162
[  164.672924] WCN: clock mode: TSX
[  164.673193] WCN: marlin_write_cali_data sync init_state:0xf49f2e22
[  164.699516] WCN: marlin_write_cali_data sync init_state:0xf49f2e22
[  164.726174] WCN: marlin_write_cali_data sync init_state:0xf0f0f0f1
[  164.726197] WCN: sdio_config bt_wake_host trigger:[high]
[  164.726201] WCN: sdio_config irq:[inband]
[  164.726205] WCN: sdio_config wake_host_level_duration_time:[20ms]
[  164.726211] WCN: sdio_config wake_host_data_separation:[bt/wifi reuse]
[  164.726215] WCN: marlin_send_sdio_config_to_cp sdio_config:0x80f01 (enable config)
[  164.726716] WCN: marlin_write_cali_data finish
[  164.726974] WCN: check_cp_ready sync val:0xf0f0f0f2, prj_type val:0x0
[  164.752860] WCN: check_cp_ready sync val:0xf0f0f0f2, prj_type val:0x0
[  164.779515] WCN: check_cp_ready sync val:0xf0f0f0f2, prj_type val:0x0
[  164.806168] WCN: check_cp_ready sync val:0xf0f0f0f2, prj_type val:0x0
[  164.832849] WCN: check_cp_ready sync val:0xf0f0f0f2, prj_type val:0x0
[  164.859511] WCN: check_cp_ready sync val:0xf0f0f0f6, prj_type val:0x0
[  164.859530] WCN: marlin_bind_verify confuse data: 0x9a09621a17108a98b0beb1aa8a12b210
[  164.859554] WCN: marlin_bind_verify verify data: 0xf5920a7267261c0fd5ba8927b860860b
[  164.860313] WCN: check_cp_ready sync val:0xf0f0f0f7, prj_type val:0x0
[  164.886194] WCN: check_cp_ready sync val:0xf0f0f0ff, prj_type val:0x0
[  164.886214] sdiohal:sdiohal_runtime_get entry
[  164.886488] WCN: get_cp2_version entry!
[  164.925608] WCN: WCND at cmd read:WCN_VER:Platform Version:MARLIN3_19B_W21.05.3~Project Version:sc2355_marlin3_lite_ott~01-27-2021 01:11:42~
[  164.925666] WCN: switch_cp2_log - close entry!
[  164.926991] WCN: WCND at cmd read:OK
[  164.927088] WCN: then marlin download finished and run ok
[  164.927097] WCN: start_loopcheck
[  164.966930] WCN: get_board_ant_num [one_ant]
[  164.966956] wifi ini path = /lib/firmware/uwe5622/wifi_2355b001_1ant.ini
[  164.993833] sprdwl:sprdwl_get_fw_info length mismatch: len_count=83, r_len=89
[  164.993860] sprdwl:sprdwl_get_fw_info, drv_version=1, fw_version=2, compat_ver=0
[  164.993867] sprdwl:chip_model:0x2355, chip_ver:0x0
[  164.993873] sprdwl:fw_ver:38222, fw_std:0x7f, fw_capa:0x120fff
[  164.993879] sprdwl:mac_addr:40:aa:56:44:ef:99
[  164.993887] sprdwl:credit_capa:TX_WITH_CREDIT
[  164.993891] sprdwl:ott support:0
[  165.027739] unisoc_wifi unisoc_wifi wlan0: mixed HW and IP checksum settings.
[  165.076462] audit: type=1130 audit(1685098588.383:70): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
```
```
[root@alarm alarm]# ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether 02:00:49:ec:9a:1a brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether c4:8d:3c:38:41:1b brd ff:ff:ff:ff:ff:ff
```

closing as fixed

closed 2023-05-26 by warpme

takase1121, 2023-05-26:
I assume I can simply update my miniarch artifact (run `start` and press 4)?

warpme, 2023-05-26:
I think yes :-)

takase1121, 2023-05-26:
I spotted some errors when scanning for Wifi with wpa_supplicant (it failed saying no networks are found):

dmesg:

```
[  119.141663] WCN: get_board_ant_num [one_ant]
[  119.141685] wifi ini path = /lib/firmware/uwe5622/wifi_2355b001_1ant.ini
[  119.170236] sprdwl:sprdwl_get_fw_info length mismatch: len_count=83, r_len=89
[  119.170260] sprdwl:sprdwl_get_fw_info, drv_version=1, fw_version=2, compat_ver=0
[  119.170270] sprdwl:chip_model:0x2355, chip_ver:0x0
[  119.170277] sprdwl:fw_ver:38222, fw_std:0x7f, fw_capa:0x120fff
[  119.170286] sprdwl:mac_addr:e0:75:26:b6:c6:fa
[  119.170296] sprdwl:credit_capa:TX_WITH_CREDIT
[  119.170301] sprdwl:ott support:0
[  119.205666] unisoc_wifi unisoc_wifi wlan0: mixed HW and IP checksum settings.
[  245.387360] INFO: task SPRDWL_TX_QUEUE:286 blocked for more than 122 seconds.
[  245.387395]       Tainted: G         C         6.3.4 #1
[  245.387406] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  245.387414] task:SPRDWL_TX_QUEUE state:D stack:0     pid:286   ppid:2      flags:0x00000008
[  245.387441] Call trace:
[  245.387448]  __switch_to+0xc0/0xf0
[  245.387476]  __schedule+0x278/0x778
[  245.387493]  schedule+0x5c/0xf8
[  245.387509]  schedule_timeout+0x1a0/0x1c0
[  245.387530]  wait_for_completion+0xb8/0x180
[  245.387548]  sprdwl_tx_work_queue+0x154/0x2b0 [sprdwl_ng]
[  245.387760]  kthread+0xe4/0xf8
[  245.387777]  ret_from_fork+0x10/0x20
[  368.267301] INFO: task SPRDWL_TX_QUEUE:286 blocked for more than 245 seconds.
[  368.267337]       Tainted: G         C         6.3.4 #1
[  368.267348] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  368.267357] task:SPRDWL_TX_QUEUE state:D stack:0     pid:286   ppid:2      flags:0x00000008
[  368.267383] Call trace:
[  368.267390]  __switch_to+0xc0/0xf0
[  368.267419]  __schedule+0x278/0x778
[  368.267436]  schedule+0x5c/0xf8
[  368.267451]  schedule_timeout+0x1a0/0x1c0
[  368.267472]  wait_for_completion+0xb8/0x180
[  368.267490]  sprdwl_tx_work_queue+0x154/0x2b0 [sprdwl_ng]
[  368.267702]  kthread+0xe4/0xf8
[  368.267719]  ret_from_fork+0x10/0x20
[  491.146440] INFO: task SPRDWL_TX_QUEUE:286 blocked for more than 368 seconds.
[  491.146476]       Tainted: G         C         6.3.4 #1
[  491.146487] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  491.146495] task:SPRDWL_TX_QUEUE state:D stack:0     pid:286   ppid:2      flags:0x00000008
[  491.146522] Call trace:
[  491.146529]  __switch_to+0xc0/0xf0
[  491.146557]  __schedule+0x278/0x778
[  491.146574]  schedule+0x5c/0xf8
[  491.146589]  schedule_timeout+0x1a0/0x1c0
[  491.146610]  wait_for_completion+0xb8/0x180
[  491.146628]  sprdwl_tx_work_queue+0x154/0x2b0 [sprdwl_ng]
[  491.146840]  kthread+0xe4/0xf8
[  491.146857]  ret_from_fork+0x10/0x20
[  614.021826] INFO: task SPRDWL_TX_QUEUE:286 blocked for more than 491 seconds.
[  614.021869]       Tainted: G         C         6.3.4 #1
[  614.021880] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  614.021888] task:SPRDWL_TX_QUEUE state:D stack:0     pid:286   ppid:2      flags:0x00000008
[  614.021916] Call trace:
[  614.021923]  __switch_to+0xc0/0xf0
[  614.021952]  __schedule+0x278/0x778
[  614.021969]  schedule+0x5c/0xf8
[  614.021985]  schedule_timeout+0x1a0/0x1c0
[  614.022006]  wait_for_completion+0xb8/0x180
[  614.022024]  sprdwl_tx_work_queue+0x154/0x2b0 [sprdwl_ng]
[  614.022237]  kthread+0xe4/0xf8
[  614.022254]  ret_from_fork+0x10/0x20
[  714.866346] unisoc_wifi unisoc_wifi wlan0: Current addr:  fc f8 1d bd 54 15 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  714.866378] unisoc_wifi unisoc_wifi wlan0: Expected addr: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  714.866388] ------------[ cut here ]------------
[... 12 lines cut ...]
[  714.866757] x20: ffff000003d7e000 x19: ffff000003d7e000 x18: 0000000000000000
[  714.866773] x17: 0000000000000000 x16: 0000000000000000 x15: 3030203030203030
[  714.866789] x14: 2030302030302030 x13: 726464615f766564 x12: 3e2d76656474656e
[  714.866805] x11: 2074636572726f63 x10: ffff800009c84428 x9 : ffff800008116a5c
[  714.866821] x8 : 00000000ffffdfff x7 : ffff800009c84428 x6 : 80000000ffffe000
[  714.866837] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000027
[  714.866852] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff000007f30000
[  714.866868] Call trace:
[  714.866873]  dev_addr_check+0xa8/0x140
[  714.866882]  __dev_open+0x44/0x200
[  714.866895]  __dev_change_flags+0x1ac/0x228
[  714.866906]  dev_change_flags+0x28/0x78
[  714.866917]  devinet_ioctl+0x310/0x688
[  714.866932]  inet_ioctl+0x1cc/0x1f0
[  714.866940]  sock_ioctl+0x338/0x460
[  714.866954]  __arm64_sys_ioctl+0xb8/0xe0
[  714.866967]  el0_svc_common.constprop.0+0x74/0x178
[  714.866983]  do_el0_svc+0x28/0x50
[  714.866994]  el0_svc+0x30/0x128
[  714.867004]  el0t_64_sync_handler+0x10c/0x118
[  714.867012]  el0t_64_sync+0x190/0x198
[  714.867022] ---[ end trace 0000000000000000 ]---
```

But I can connect to network properly with SSID and password. So I guess issue is resolved. Might need further investigation, but this is good enough for me.

warpme, 2023-05-26:
yeah. quality of aw859a driver is TERRIBLE....