warpme/minimyth2 #53: hint: maybe use pwm regulator for cpu on x96x6 box as well?

hexdump0815, 2023-08-25:
hello,

first a very big thank you for all your efforts to get new hardware running (h313/h616/h618/rk2566/rk3528 tv boxes for instance)!

it looks like you are using a fixed voltage for the opp points in your x96x6 dts - for me using a pwm regulator like in https://github.com/gregkh/linux/blob/master/arch/arm64/boot/dts/rockchip/rk3566-box-demo.dts or in your new rk3538 dts seems to be working better - seems to get better performance for me and should result in less power consumption. i did not test it deeply, but something like this worked quite well for some rk3328 tv box without a pmic in the past for me as well. actually my only testing so far was to use the mentioned box-demo dts instead of your x96x6 dts, which had the side effects that the builtin ethernet worked with that, but usb not. when i find some time, my plan is to try to make something like the best of both worlds out of the demo-box and your x96x6 dts.

all the best wishes and keep up your good work - hexdump

warpme, 2023-09-15:
@hexdump0815 
Thx for kind words!
And sorry for late replay - vacation period....
May you pls try dts with https://github.com/warpme/minimyth2/commit/06be37cd77d4a64578480aa560c7e493a5d0848f changes?
In this commit i switched cpu_vdd to pwm-regulator and went back to dtsi cpu opp modes

hexdump0815, 2023-09-16:
@warpme - thanks a lot, i'll give it a try during the next days

hexdump0815, 2023-09-30:
sorry for the long delay, but i can finally confirm that this is working well and gives better performance than with the old fixed regulator

closed 2023-09-30 by warpme