warpme/minimyth2 #54: H313 board ethernet issue

kernelgurumeditation, 2023-11-20:
Hi, 

first thanks for the project, it's very useful!
I have an INEXT h313 board that uses lddr3, your sdimage h313.x96_q_lpddr3, boots and is stable, but has an issue, the ethernet doesnt work.
This board doesn't have wifi/Bluetooth so having ethernet working is a must.

I'm adding dmesg and uboot logs in case you have a chance to give it a look.

thanks

[dmesg.txt](https://github.com/warpme/minimyth2/files/13405176/dmesg.txt)
[u-boot.txt](https://github.com/warpme/minimyth2/files/13405177/u-boot.txt)

warpme, 2023-12-08:
Well - I'm afraid without access to real hardware i can't help you...

warpme, 2024-02-12:
Closed due no hardware to work-on.

closed 2024-02-12 by warpme