warpme/minimyth2 #59: N100 support?

kjeacle, 2024-05-04:
I need to replace some old Nvidia ION devices so I decided to try an Asus Prime N100I-D D4 motherboard. 

A clean first-time boot with MM2 worked but no VAAPI decoders were found. When I tried to boot with a different theme, X11 kept switching resolution to 640x480.

Is there something incompatible about this particular motherboard or is N100 just not fully supported by MM2 yet?

Debug logs attached if it helps. Thanks!

[n100-system-info.gz](https://github.com/warpme/minimyth2/files/15208330/n100-system-info.gz)

warpme, 2024-05-04:
Karl,
May you pls try this: http://warped.inet2.org/ram-x86_64-minimyth2-master-13.2.0.r100.tar.bz2

kjeacle, 2024-05-04:
Thanks for the new version to test! I tried it but don't see any obvious difference. First clean boot (with database settings for frontend wiped) works as per attached photo. Audio works and "Minimyth" decoder profile works, but no VAAPI.

On every subsequent boot, the bottom half of the screen is shifted up and resolution is 640x480. Playback does not work in this case but when existing from failed playback the whole screen is visible again but resolution is just 640x480.

[image]
[image]
[system-info-first-boot.gz](https://github.com/warpme/minimyth2/files/15210149/system-info-first-boot.gz)
[system-info-second-boot.gz](https://github.com/warpme/minimyth2/files/15210150/system-info-second-boot.gz)

warpme, 2024-05-04:
Karl,
Thx for inputs.
From logs I see n100 needs new firmware which i need to include as build-ins into kernel.
Ill provide you new image soon. 
pls stay tuned :-)

warpme, 2024-05-09:
Karl,
I uploaded new ver. 
pls download and try:
http://warped.inet2.org/ram-x86_64-minimyth2-master-13.2.0.r100.tar.bz2

kjeacle, 2024-05-09:
Thanks for new version! The good news: VAAPI works!

The bad news: same problem as before with only first clean boot working. Second and subsequent boots start up in 640x480 with screen shifted up as shown in image above and playback does not work.

After some experimentation, I found that I could manually set the correct resolution for GUI and Video Output in: Setup - Appearance - Separate Video Modes for GUI and TV Playback. So, now, everything appears to be working. It's just that this is something that I have never had to do with previous versions of MM2.

[system-info-first-boot.gz](https://github.com/warpme/minimyth2/files/15266311/system-info-first-boot.gz)
[system-info-second-boot.gz](https://github.com/warpme/minimyth2/files/15266313/system-info-second-boot.gz)

warpme, 2024-05-22:
Karl,
It looks like 640x480 issue is normal MythTV behavior.
By https://github.com/MythTV/mythtv/blob/382deba302d39aa45e0dea67e0b3d86ee6f2988f/mythtv/programs/mythfrontend/globalsettings.cpp#L2407 - at first boot - mythtv inits setting to 640x480. At next boot it uses it so that's why you are getting up with 640x480.
I added fix to this: now GUI tries to start with 1980x1080 and falls back to lower res when 1980x1080 is not available. Playback res is also defaulted to 1920x1080. If user tv has no capability to display at 1920x1080 - user needs to manually change TV playback resolution in MythTV settings.
Pls try this new ver: http://minimyth2.mooo.com/ram-x86_64-minimyth2-master-13.2.1.r130.tar.bz2

btw: is boot splash working ok on yours N100?
On mine n3550 it is frozen at initial "starting ..." screen. (it works o on amd and 23 different arm boards - only inteh has this issue)

kjeacle, 2024-05-23:
The fix to try 1920x1080 works here so that problem is solved. Splashscreen freezes here too as you describe. Otherwise this N100 motherboard is working well so I can definitely recommend. Feel free to close this issue unless there is something else you want me to try. Thanks!

warpme, 2024-05-23:
Thx for your support!

closed 2024-05-23 by warpme