warpme/minimyth2 #61: Strange clone x96q

lexmulya, 2024-07-02:
Hello.

I am the owner of a strange x96q clone, it looks like lpddr3+emmc. I tried MiniMyth2-armv8-master-13.2.1.r130-board-h313.x96_q_lpddr3_v1.3-SD-Image.img and MiniMyth2-armv8-master-13.2.1.r130-board-h313.x96_q_lpddr3-SD-Image. img and they both do not load to the kernel.

I get the following log
`U-Boot SPL 2024.04 (May 18 2024 - 02:16:28 +0200)
DRAM: 2048 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.10.0  (debug):v13.2.0-3-ga1bb0b907-dirty
NOTICE:  BL31: Built : 02:16:23, May 18 2024
NOTICE:  BL31: Detected Allwinner H616 SoC (1823)
NOTICE:  BL31: Found U-Boot DTB at 0x4a0ae9d8, model: hechuang,x96-q LPDDR3 v1.3
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
INFO:    PMIC: Probing AXP305 on RSB
ERROR:   RSB: set run-time address: 0x10003
INFO:    Could not init RSB: -65539
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for erratum 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for erratum 1530924 was applied
INFO:    PSCI: Suspend is unavailable
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
INFO:    Changed devicetree.

U-Boot 2024.04 (May 18 2024 - 02:16:28 +0200) Allwinner Technology

CPU:   Allwinner H616 (SUN50I)
Model: hechuang,x96-q LPDDR3 v1.3
DRAM:  2 GiB (effective 4.7 EiB)`

The memory chip I have installed is h9tq64aaetac
I booted into the built-in android and downloaded the DTB file
[x96q.dtb.gz](https://github.com/user-attachments/files/16072896/x96q.dtb.gz)
[image]

If you have free time, I ask for your help.
I'm sorry for my English.

lexmulya, 2024-07-02:
Uboot log when loading android
`U-Boot SPL 2024.01 (Jun 13 2024 - 15:40:46 +0300)
DRAM: 128 MiB
[82]HELLO! BOOT0 is starting!
[85]BOOT0 commit : 3ae35eb
[87]set pll start
[90]periph0 has been enabled
[93]set pll end
[95]unknow PMU
[96]unknow PMU
[98]PMU: AXP1530
[100]dram return write ok
[103]board init ok
[104]DRAM BOOT DRIVE INFO: V0.648
[108]the chip id is 0x5c00
[110]chip id check OK
[114]DRAM_VCC set to 1200 mv
[117]DRAM CLK =600 MHZ
[119]DRAM Type =7 (3:DDR3,4:DDR4,7:LPDDR3,8:LPDDR4)
[127]Actual DRAM SIZE =1536 M
[130]DRAM SIZE =1536 MBytes, para1 = 30fa, para2 = 6001000, dram_tpr13 = 26061
[143]DRAM simple test OK.
[145]rtc standby flag is 0x0, super standby flag is 0x0
[151]dram size =1536
[154]card no is 2
[156]sdcard 2 line count 8
[158][mmc]: mmc driver ver 2021-10-12 13:56
[163][mmc]: b mmc 2 bias 4
[171][mmc]: Wrong media type 0x0, but host sdc2, try mmc first
[177][mmc]: ***Try MMC card 2***
[223][mmc]: RMCA OK!
[228][mmc]: MMC 5.0
[230][mmc]: HSSDR52/SDR25 8 bit
[233][mmc]: 50000000 Hz
[235][mmc]: 7456 MB
[237][mmc]: ***SD/MMC 2 init OK!!!***
[318]Loading boot-pkg Succeed(index=0).
[322][mmc]: b mmc 2 bias 4
[324]Entry_name        = u-boot
[334]Entry_name        = monitor
[338]Entry_name        = dtbo
[341]Entry_name        = dtb
[345]tunning data addr:0x4a0003e8
[348]Jump to second Boot.
NOTICE:  BL3-1: v1.0(debug):05d6c57
NOTICE:  BL3-1: Built : 13:35:35, 2021-10-28
NOTICE:  BL3-1 commit: 8
NOTICE:  cpuidle init version V1.0
ERROR:   Error initializing runtime service tspd_fast
NOTICE:  BL3-1: Preparing for EL3 exit to normal world
NOTICE:  BL3-1: Next image address = 0x4a000000
-OTICE:  BL3-1: Next image spsr = 0x1d3

U-Boot 2018.05 (Dec 04 2023 - 17:33:48 +0800) Allwinner Technology

[00.428]CPU:   Allwinner Family
[00.431]Model: sun50iw9
I2C:   ready
[00.435]DRAM:  1.5 GiB
[00.438]Relocation Offset is: 55ebf000
[00.482]secure enable bit: 0
[00.485]pmu_axp152_probe pmic_bus_read fail
[00.489]PMU: AXP1530
[00.494]CPU=1008 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=400Mhz
[00.502]drv_disp_init
[00.532]__clk_enable: clk is null.
[00.538]drv_disp_init finish
[00.541]gic: sec monitor mode
[00.570]flash init start
[00.573]workmode = 0,storage type = 2
[00.576]MMC:     2
[00.577][mmc]: mmc driver ver uboot2018:2021-07-19 14:09:00
[00.584][mmc]: get sdc_type fail and use default host:tm4.
[00.595][mmc]: SUNXI SDMMC Controller Version:0x40502
[00.648][mmc]: Best spd md: 3-HS200/SDR104, freq: 4-150000000, Bus width: 8
[00.654]sunxi flash init ok
[00.658]Loading Environment from SUNXI_FLASH... OK
[00.668]usb burn from boot
delay time 0
weak:otg_phy_config
[00.682]usb prepare ok
[01.485]overtime
[01.489]do_burn_from_boot usb : no usb exist
[01.493]boot_gui_init:start
FAT: Misaligned buffer address (9be7b598)
32 bytes read in 4 ms (7.8 KiB/s)
[01.777]boot_gui_init:finish
[01.780]bmp_name=bootlogo.bmp
3686456 bytes read in 31 ms (113.4 MiB/s)
[01.825][mmc]: delete mmc-hs400-1_8v from dtb
[01.832]update dts
** Unrecognized filesystem type **
[01.843]load file(ULI/factory/rootwait init.txt) error.
** Unrecognized filesystem type **
[01.858]load file(ULI/factory/snum.txt) error.
[01.862]name in map mac
** Unrecognized filesystem type **
[01.874]load file(ULI/factory/wifi_mac.txt) error.
** Unrecognized filesystem type **
[01.888]load file(ULI/factory/bt_mac.txt) error.
** Unrecognized filesystem type **
[01.901]load file(ULI/factory/selinux.txt) error.
** Unrecognized filesystem type **
[01.915]load file(ULI/factory/specialstr.txt) error.
[01.926]update part info
[01.949]update bootcmd
[01.951]No ethernet found.
Hit any key to stop autoboot:  0
[02.182]Starting kernel ...`

AlabamaWebp, 2024-07-24:
I have the same device as in the photo above and it does not start if you find a solution please write here

warpme, 2024-08-13:
Well - without hw at hand i can't help....

warpme, 2024-10-09:
closed due no hw to work on

closed 2024-10-09 by warpme