warpme/minimyth2 #62: Can’t connect to database

jaum20, 2024-08-10:
Can’t connect to database using a box with minimyth instaled. Using another pc in the network I can connect. If I click "search" in the frontend it will prompt me asking for a PIN. I tried "mythtv" but didn‘t work

[image]

warpme, 2024-08-13:
Hi,
What hw are you using?
Generally - without logs i can't say anything.
Is it possible to i.e. use USB-Eth dongle (to get Internet access) and provide me output of system info (i.e. by logging with telnet then typing 'tb4' then providing me url you will get as output)

jaum20, 2024-08-13:
I'm using TX6s (`MiniMyth2-armv8-master-13.4.0.r239-board-h616.tanix_tx6s_axp313-SD-Image.img.xz`). I never used telnet, just ssh. How can I do that?

warpme, 2024-10-09:
closed due no activity

closed 2024-10-09 by warpme

jaum20, 2024-10-09:
no activity??