warpme/minimyth2 #63: Latest Image doesn't work on Radxa Zero 3W

coolirisme, 2024-11-27:
I flashed the images from the latest release on both EMMC and SDCARD but it doesn't boot in both cases.

warpme, 2024-11-28:
Hi,
I just downloaded https://github.com/warpme/minimyth2/releases/download/v13.8.0/MiniMyth2-armv8-master-13.8.0.r381-board-rk3566.zero3w-SD-Image.img.xz, flashed to sdcard, powered and all works (https://gist.github.com/warpme/1f817fdc5d760c27b6d28f84dd1a925f)
Is it possible for you to get uart log from booting process of your's zero3w?

ps: rockchip uboot prefers booting from emmc over sd card.
So it you have e.g. broken bootloader on emmc - even it you inserted well booting sdcard - board boot will fail due emmc broken bootloader. In such case you need to erase emmc to make board booting from sd card....

coolirisme, 2024-11-28:
@warpme I erased the emmc using rkdeveloptool and reflashed the image on sdcard but thing no HDMI signal. Unfortunately its not possible for me to get UART as my radxa doesnt have the headers soldered on it.

The LED is stuck on a solid green color

warpme, 2025-01-16:
May you provide me pls uart output?
Without this info i can't move forward....

closed 2025-02-03 by warpme