warpme/minimyth2 #66: Commit 962ac7d breaks boot for some H618 devices

SiliconExarch, 2025-02-23:
I am currently working on adding proper support for the H618 version of the T95Z Plus TV box, which is mostly compatible with the Vontar H618, aside from the WiFi/BT, Seven-segment display and RGB LEDs. 

Since commit 962ac7d the device appears not to boot as there is no longer any HDMI output after applying power and waiting for a few seconds, regardless of whether I use the Vontar dtb or my own WIP one. Unfortunately I don't have UART on the device yet so I can't check the u-boot console for any relevant errors.

Reverting this commit allows the device to boot again with either dtb.

warpme, 2025-02-23:
Well - https://github.com/warpme/minimyth2/commit/962ac7da6cb84d0e41826c5555cb101b70a53a8d is quite useful as it greatly  improves ram size detection on 3 different devices i have. 
I suspect it just exposes issue of using no correct dram params for your tvbox....
Ideally will be to:

1. boot box with any working image
2. telnet and dump first 16MB of box eMMC.
E.g. if your box has emmc as mmc2, run:
`dd if=/dev/mmcblk2 of=/mmc.bin bs=1M count=16`
mmc.bin contains vendor dram exact params used for dram chips your box has
3. provide this dump for me

With this dump - i can prepare dedicated uboot using exact dram params from your box

Generally: dram area is bumpy road...
Without uart we are totally blind so - if you want to get this box supported - i'm pretty sure you will need uart sooner or latter!

warpme, 2025-03-04:
Hi,
In 2e267842b1033bbc4c2c5d80c1756a142e347cc5 i added potential fix....
may you try how it goes for you?

SiliconExarch, 2025-03-04:
> Hi, In [2e26784](https://github.com/warpme/minimyth2/commit/2e267842b1033bbc4c2c5d80c1756a142e347cc5) i added potential fix.... may you try how it goes for you?

Now booting again with this fix. 

Here is the [requested dump](https://github.com/user-attachments/files/19077741/mmc.zip) in case it's still of any use.

warpme, 2025-03-07:
Hi, i added optimised code for ram sizing (and emmc booting).
may you verify with current head?
thx!

warpme, 2025-03-18:
closing as fixed.
thx!

closed 2025-03-18 by warpme