warpme/minimyth2 #68: How to update database

andyshrk, 2025-07-12:
Hello, these past couple of days, I've decided to give minimyth2 a try again.
I down load Myth2-armv8-master-14.2.0.r292-board-rk3566.x96_x6-SD-Image.img.xz  from thre rease page, write it on a SD card.
Then change the FDT to
FDT /dtbs/rockchip/rk3566-box-demo.dtb
Then bootup on my rk35660box-demo board.

Finally, all I saw on the HDMI screen was a prompt: This version of MythTV requires an updated database(schema is 5 versions behind), Please run myttv-seup or mythbackend to update y....

For the backend:
I installed mythbackend 34 on ubuntu 24.04:
And on the setupwizard web page, I can see: Database schema version is 1379, But I can't find where to do the database update.
I'm not sure if you can give some help information. Thank you very much.

warpme, 2025-07-13:
Hi Andy,

Thx for trying minimyth2!

I see you are using quite old ver of mythtv (v34).
mythtv is quite actively developed and current ver. (v35) has many fixes/improvements over 1.5y old v34.

Your DB schema ver. issue is direct result versions mismatch between backend (v34) and frontend (minimyth2 has is pre-36).
Pls try to use v35 backend. For frontend - i prepared v35 based build of minimyth2: https://github.com/warpme/minimyth2/releases/download/v14.2.0/MiniMyth2-armv8-35-14.2.1.r30-board-rk3566.x96_x6-SD-Image.img.xz  

Also i prepared small tutorial for backend setup: https://github.com/warpme/miniarch/wiki/Installing-and-Setup-MythTV-on-MiniArch 
It is for Archlinux - but you may find it helpful maybe at backend configure process.

Pls let me know how it goes for you.

andyshrk, 2025-07-15:
> Andy, Good to hear thing started somehow work you you!
> 
> 1. mouse issue: how app (mythtv) reacts on input devices (mouse) depends on how qt is configured to handle input. In need to check why left/right mouse keys are not working. Pls use keyboard (or ir remote - assuming you have selected right ir keymappings; this we can sort later)
> 2. for cec - you need enable it in mythtv settings. Pls go to Setup/General/RemoteControl/EnableCECcontrol interface and tick "on"

After connecting the keyboard, I have enabled the CEC function.
Now, I can operate it using the CEC remote control.

> 4. for issue with language setup screen after restart: i have few q:>    3.1: issue happens after backed restart or frontend restart?
>    3.2: are you getting the same box mac address after frontend reboot (cold and warm)?
>    3.3: may you provide me backend logs: after first boot (with ok ui) and after second boot (when you get language ui)
>    3.4: also frontend (mm2) logs will be useful. Simply: boot mm2; telnet to box; type in terminal `tb4`; and send me url. For both cases (first boot with ok ui & second boot when you getting language ui)
> 

After connecting the keyboard, it seemed that I pressed the ESC key, 
and then I returned to the normal UI settings interface.

> And: did you follow backend setup like i.e in https://github.com/warpme/miniarch/wiki/Installing-and-Setup-MythTV-on-MiniArch ?

I will take a closer look at these setting documents again.
Thank you so much.

warpme, 2025-07-21:
Andy,
If all is ok for you - may we close this ticket?

andyshrk, 2025-07-22:
> Andy, If all is ok for you - may we close this ticket?

It works. Yes, we can close it, thank you very much.