warpme/minimyth2 #69: [Request Help] About the sdio wifi speed problem of nanopi r76s

qingtian110, 2025-07-26:
I compiled the openwrt system according to your project dts (mainline kernel to implement nanopi r76s device support), the sdio interface cannot be used sdr104 mode.

```
[    1.346862] dwmmc_rockchip 2a310000.mmc: Successfully tuned phase to 55
[    1.346944] mmc0: new ultra high speed SDR104 SDHC card at address aaaa
[    1.347816] mmcblk0: mmc0:aaaa SC32G 29.7 GiB
[    1.372260] init: Console is alive
[    1.372354] init: - watchdog -
[    1.770720] dwmmc_rockchip 2a320000.mmc: Unexpected CMD11 timeout
[    1.810338] kmodloader: loading kernel modules from /etc/modules-boot.d/*
[    1.824235] button_hotplug: loading out-of-tree module taints kernel.
[    1.825216] Button Hotplug driver version 0.4.1
[    1.828092] raid6: skipped pq benchmark and selected neonx8
[    1.828138] raid6: using neon recovery algorithm
[    1.830981] xor: measuring software checksum speed
[    1.831264]    8regs           : 13569 MB/sec
[    1.831534]    32regs          : 13701 MB/sec
[    1.831902]    arm64_neon      :  9663 MB/sec
[    1.831930] xor: using function: 32regs (13701 MB/sec)
[    1.882508] Btrfs loaded, zoned=no, fsverity=no
[    1.898466] SGI XFS with ACLs, security attributes, no debug enabled
[    1.903861] usbcore: registered new interface driver uas
[    1.904040] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
[    1.910667] init: - preinit -
[    2.013493] mmc_host mmc1: Timeout sending command (cmd 0x10202000 arg 0x0 status 0x90202000)
[    2.014132]  mmcblk0: p1 p2
[    2.255133] mmc_host mmc1: Bus speed (slot 0) = 50000000Hz (slot req 50000000Hz, actual 50000000HZ div = 0)
[    2.256188] mmc1: new high speed SDIO card at address 0001
```

**[    1.770720] dwmmc_rockchip 2a320000.mmc: Unexpected CMD11 timeout**

**[    2.256188] mmc1: new high speed SDIO card at address 0001**

CMD11 Timeout occurred while trying to switch to 1.8v, this causes it to only work at "high speed" instead of "ultra high speed (SDR104)".

I confirmed that sdio works very well with SDR104 in friendlyarm official system (bsp kernel). 

What do you think about this?

qingtian110, 2025-07-26:
`[    4.968096] mmc1: new ultra high speed SDR104 SDIO card at address 0001`

On friendlyarm systems, sdio wifi speeds can reach 450Mbps (Client & AP), while on the mainline kernel they are only Client 80~120Mbps and AP 50Mbps.

qingtian110, 2025-07-27:
This is my dmesg

<details>
<summary>dmesg</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.12.40 (admin@cooluc.com) (clang version 20.0.0 (OpenWrt), LLD 20.0.0) #0 SMP Mon Jun 23 20:40:36 2025
[    0.000000] KASLR disabled due to lack of seed
[    0.000000] Machine model: FriendlyElec NanoPi R76S
[    0.000000] earlycon: uart8250 at MMIO32 0x000000002ad40000 (options '')
[    0.000000] printk: legacy bootconsole [uart8250] enabled
[    0.000000] OF: reserved mem: Reserved memory: No reserved-memory node in the DT
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040200000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x000000043fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040200000-0x000000043fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040200000-0x000000043fffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 16 MiB at 0x00000000ff000000 on node -1
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.5
[    0.000000] percpu: Embedded 33 pages/cpu s95704 r8192 d31272 u135168
[    0.000000] pcpu-alloc: s95704 r8192 d31272 u135168 alloc=33*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 [0] 6 [0] 7 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced OFF by mitigations=off
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=2M console=tty1 console=ttyS2,1500000 earlycon=uart8250,mmio32,0x2ad40000 root=PARTUUID=5452574f-02 rw rootwait mitigations=off
[    0.000000] Dentry cache hash table entries: 2097152 (order: 12, 16777216 bytes, linear)
[    0.000000] Inode-cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 4193792
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 8.
[    0.000000] software IO TLB: mapped [mem 0x00000000f8e96000-0x00000000fce96000] (64MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=8, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=8.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=8
[    0.000000] RCU Tasks Trace: Setting shift to 3 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=8.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001572] Console: colour dummy device 80x25
[    0.001989] printk: legacy console [tty1] enabled
[    0.002958] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.003912] pid_max: default: 32768 minimum: 301
[    0.004661] Mount-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.005412] Mountpoint-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.007158] spectre-v4 mitigation disabled by command-line option
[    0.008330] cacheinfo: Unable to detect cache hierarchy for CPU 0
[    0.010049] rcu: Hierarchical SRCU implementation.
[    0.010494] rcu: 	Max phase no-delay instances is 1000.
[    0.011200] Timer migration: 1 hierarchy levels; 8 children per group; 1 crossnode level
[    0.012736] smp: Bringing up secondary CPUs ...
[    0.013685] Detected VIPT I-cache on CPU1
[    0.013795] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[... 441 lines cut ...]
[   15.198380] Adding 4089852k swap on /dev/zram0.  Priority:100 extents:1 across:4089852k SSDsc
[   17.712800] eth1: 0xffff8000833c0000, 96:87:1b:69:61:0c, IRQ 93
[   17.725545] br-lan: port 1(eth1) entered blocking state
[   17.725572] br-lan: port 1(eth1) entered disabled state
[   17.725630] r8125 0000:01:00.0 eth1: entered allmulticast mode
[   17.725887] r8125 0000:01:00.0 eth1: entered promiscuous mode
[   17.734409] eth0: 0xffff800083310000, 96:87:1b:69:61:0b, IRQ 94
[   21.683705] br-lan: port 2(phy0-ap0) entered blocking state
[   21.683722] br-lan: port 2(phy0-ap0) entered disabled state
[   21.683761] rtw_8822cs mmc1:0001:1 phy0-ap0: entered allmulticast mode
[   21.683920] rtw_8822cs mmc1:0001:1 phy0-ap0: entered promiscuous mode
[   21.683985] br-lan: port 2(phy0-ap0) entered blocking state
[   21.683994] br-lan: port 2(phy0-ap0) entered forwarding state
[   21.684776] br-lan: port 2(phy0-ap0) entered disabled state
[   22.533255] r8125: eth0: Link is Up - 2.5Gbps/Full
[   22.535102] r8125: eth1: Link is Up - 2.5Gbps/Full
[   22.535277] br-lan: port 1(eth1) entered blocking state
[   22.535288] br-lan: port 1(eth1) entered forwarding state
[   23.963611] br-lan: port 2(phy0-ap0) entered blocking state
[   23.963638] br-lan: port 2(phy0-ap0) entered forwarding state
[   31.183839] vdd_npu_s0: disabling

```

</details>

qingtian110, 2025-07-27:
iperf3 test

```txt
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 24.10.2, r28739-d9340319c6
 -----------------------------------------------------
[root@OpenWrt ~]# iperf3 -c 10.0.0.1
Connecting to host 10.0.0.1, port 5201
[  5] local 10.0.0.216 port 49032 connected to 10.0.0.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  31.0 MBytes   260 Mbits/sec    0    814 KBytes       
[  5]   1.00-2.00   sec  27.1 MBytes   228 Mbits/sec    0    720 KBytes       
[  5]   2.00-3.00   sec  30.4 MBytes   255 Mbits/sec    0   1.04 MBytes       
[  5]   3.00-4.00   sec  28.0 MBytes   235 Mbits/sec    0   1.05 MBytes       
[  5]   4.00-5.00   sec  32.1 MBytes   269 Mbits/sec    0   1.25 MBytes       
[  5]   5.00-6.00   sec  24.8 MBytes   208 Mbits/sec    0   1.07 MBytes       
[  5]   6.00-7.00   sec  31.1 MBytes   261 Mbits/sec    0   1.20 MBytes       
[  5]   7.00-8.00   sec  29.1 MBytes   244 Mbits/sec    0   1.18 MBytes       
[  5]   8.00-9.00   sec  30.2 MBytes   254 Mbits/sec    0   1.26 MBytes       
[  5]   9.00-10.00  sec  27.5 MBytes   231 Mbits/sec    0   1.20 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   291 MBytes   244 Mbits/sec    0             sender
[  5]   0.00-10.02  sec   290 MBytes   243 Mbits/sec                  receiver

iperf Done.
```

qingtian110, 2025-07-27:
Because the kernel uses the rtw88-8822cs driver, it cannot fully utilize the performance of the network card and the speed is limited to 250Mbps.

According to the developer, to achieve the full SDIO WiFi performance of this device, only the Realtek driver (out of tree) can be used.

warpme, 2025-07-28:
Thx for this!
I already backported patch to 6.16 mainline.
dmesg (on mm1 entries looks identical) to yours but speed i'm getting are 45-50Mbps at max
Still investigating why....

warpme, 2025-07-29:
May you provide me link to your's kernel repo?
I want to check what will be diff between: mine 6.16 mainline and yours 6.12.40.....

warpme, 2025-10-01:
I'm closing this ticket due lack of understanding where root cause is (on mainline kernel).
thx!